MarlinFirmware/Marlin #5117: Compilation Bug in 1.x -win10 Ard. ide 1.6.11

Netoperz, 2016-10-31:
downloaded clean code.
Changed bed shape, enabled sd support and reprap discount full graphic controller, no other changes:
u8glib installed in arduino ide, and no problems when compiling rc branch code.

but in stable branch 1.x as follows:

In file included from sketch\dogm_lcd_implementation.h:38:0,

                 from sketch\ultralcd.cpp:39:

sketch\dogm_font_data_marlin.h:12:25: fatal error: utility/u8g.h: No such file or directory

 #include <utility/u8g.h>

                         ^

compilation terminated.

exit status 1

thinkyhead, 2016-10-31:
Arduino IDE can't find your U8glib folder. You might try re-installing it.

Blue-Marlin, 2016-10-31:
The directory structure ib u8glib changed. Try a u8glib from this time ~1.17. The one provided with Marlin 1.0.x (https://github.com/MarlinFirmware/Marlin/tree/1.0.x/ArduinoAddons/Arduino_1.0.x/libraries/U8glib v1.11) should compile fine.

Blue-Marlin, 2016-10-31:
If you want to use a newer u8glib replace all `#include <utility/u8g.h>` with `#include <U8glib.h>`

thinkyhead, 2016-10-31:
Should we "certify" the newer U8glib @Blue-Marlin or are there compatibility issues?

Blue-Marlin, 2016-10-31:
Current Marlin is working flawlessly with current u8glibs. We made the changes on 19 Aug 2015 what corresponds to the introduction of u8glib v1.18.

Netoperz, 2016-10-31:
i think we should change code to correspond with the current u8glib.

there is no problem with new version, and that problem i have seen here is annoing.

Suggestion: switch to new u8glib the same as in RC if there is no known compatibility  problems.

I'll change the lines as the Blue said and i'll check if there are any problems with new setup if everything will be fine i'll post here results.

thinkyhead, 2016-10-31:
Great! Looking forward to your test results.

thinkyhead, 2016-11-04:
@Netoperz I was just scanning code and see we're already using the newer `#include <U8glib.h>`.

Blue-Marlin, 2016-11-04:
> but in stable branch 1.x as follows:

thinkyhead, 2016-11-04:
Yes, I know. So should I do this patch and release 1.0.2-3?

Blue-Marlin, 2016-11-04:
No. Every bug there helps to prevent using it. ;-)

closed 2017-10-31 by thinkyhead