MarlinFirmware/Marlin #5119: E3D PT100 clarification

crncrussell, 2016-10-31:
Historically, I've used a type 20 thermistor and wired the E3d PT100 board to a spare analog input pin on an Audio header.  I've noticed in RC7 there is a now a type 147 thermistor for a PT100 with a 4.7 pull-up resister.  Hoping for clarification: is this intended to calculate the values with the E3D on the "stock" thermistor pins without using an aux analog, or am I misunderstand it's purpose?

Thanks!

Blue-Marlin, 2016-10-31:
I have no clue but maybe the original author (@josefpavlik)

josefpavlik, 2016-10-31:
The type 147 is the (calculated) table for Pt100 with 4k7 pullup without any electronic, So, Pt100 between ground and the analog input, 4k7 as pullup between analog input and Vcc. Nothing else. The Pt100 is connected to T0, T1 or T2 input of the RAMPS board.

However, I recommend to use Pt1000 instead of Pt100 and type 1047. The Pt100 has too small voltage change between min and max temperature, so the precision is not good. Pt1000 has better precision (in this circuit). I'm using Pt1000 in my printer.

crncrussell, 2016-10-31:
Thank you for clarifying @josefpavlik .  I have the E3d amplifier and Pt100 so will stick with that for now on a spare analog pin, but it is good to know that we have the option of good no direct to a ramps without the amplifier board on one of the built in pull-ups.

closed 2016-10-31 by thinkyhead