MarlinFirmware/Marlin #5120: [GCode] "Execute now" feature ?

Salamandar, 2016-10-31:
I just don't understand how there cannot be a "execute RIGHT NOW" GCode message. Or I just didn't find it.
It's just fundamental to bypass the printing GCode "buffer" for emergency stops, pause, or just to NOT wait the end of a move (Z moves can be very long) to start heating.

Blue-Marlin, 2016-10-31:
"execute RIGHT NOW" is the wrong concept. How to decide?
First "Stop", than do what ever you want.

Salamandar, 2016-10-31:
> How to decide?
You decide… to execute right now what you've been told.
You may not want to stop. Temp changes,…

Salamandar, 2016-10-31:
Example : My slicer sets the bed temp to 70°C for the first layer. But right now I want to print some parts without the heating bed.
1. I start the printing
2. Marlin waits for the bed to heat
3. In Pronterface, I stop the bed heating
4. Marlin waits the bed to be hot to… not heat the bed ?!

thinkyhead, 2016-10-31:
Activate `EMERGENCY_PARSER` and then you can interrupt heating with `M108`.

closed 2016-10-31 by thinkyhead