MarlinFirmware/Marlin #5121: Roxy3D example config - Z axis speed

ngarafol, 2016-10-31:
So just wondering if there is some advantage setting Z axis slow as is in example config - Roxy3D folder. It is ridiculously slow, especially during 10x10 G29 P1 probe. Takes literally forever. I set Z axis default steps to 4 or 5 and Z axis acceleration to 1000, it is warp speed compared to example. Am I doing it wrong, getting unreliable readings by speeding axis up (using Z probe)? 
Are there some recomended values for Z probes? 
Now I remember, example is slow due to Roxy using servo for Z probing, thats it?

Blue-Marlin, 2016-10-31:
Don't forget - Roxy is developing that stuff.
In my experience I need some time to find the reset button.
As long as your probes results make sense you can speed up the probing Just try it.

ngarafol, 2016-10-31:
Half way during writing issue, I remembered there were settings for servo in example config, so there is no inductive probe but servo is being used. Maybe 2 is ok for servo but it is taking too long when measuring 10x10 mesh. Like I did not measure how long but seems like forever :D I may have to switch to 15x15 mesh since some spots are not great. So I had to speed up the process just to cut time. 
But then again thought appeared - what if slower means more precise :) But then again its inductive probe. Will have to check if Z axis speed has any influence or not on mesh values.

Blue-Marlin, 2016-10-31:
It works that way:
Check if probe is triggered. If triggered immediately stop.
Make next step.
Check if probe is triggered. If triggered immediately stop.
Make next step.
...
Check if probe is triggered. If triggered immediately stop.
Make next step.

If being too fast 2 things can happen:
If you are in dual or quad stepping mode the resolution is halved/quartered.
The immediate stop is not deaccelerated. You may lose steps here if too fast.

Blue-Marlin, 2016-10-31:
The other way to speed things up is to level your bed as far as possible. Then you can shrink the way the z-axis has to go up and own.

Roxy-3D, 2016-10-31:
> So just wondering if there is some advantage setting Z axis slow as is in example config - Roxy3D folder. It is ridiculously slow, especially during 10x10 G29 P1 probe. Takes literally forever. 

There is a big advantage for me!  My printer's Z-Axis is super slow.   I need it that slow to work and not miss steps.   You can change your Z-Feedrates to what ever make sense on your printer.

> I may have to switch to 15x15 mesh since some spots are not great. So I had to speed up the process just to cut time. 

Get a cup of coffee and go play on the internet for 20 minutes!   That is what I do!

 > But then again thought appeared - what if slower means more precise :)

Slower is more precise.   But the initial probing is just so the Mesh is well enough defined you can do a G26 C P O3.0 and see where the problems are.   After the initial probing, you should be able to fine tune and correct any areas with bad adhesion with a G29 P4 R.

ngarafol, 2016-10-31:
Roxy, what is the cause of Z axis being slow and missing steps on your printer? Z driver overheating? I had that issue and removed ramps board from prusa frame and put it in front of atx psu so it is chilled :)

Nocturnal42, 2016-10-31:
I set my HOMING_FEEDRATE to 400 for the Z axis, it's probably way too fast for accurate reads, but I haven't actually got all the way through the mess creation process yet, and having to wait 1/2 hour each time I was probing the bed wasn't acceptable. So now it only takes a few minutes. 

I've run a few tests, and at the speed, comparing repeat meshes, I getting <0.1mm difference.

Nocturnal42, 2016-10-31:
>You only have to probe the bed once. From then on, it is printing a Mesh Validation Pattern and then editing the problem areas. And in fact, @WOOKIE111 has brought the system up with no probing at all. You can just set the entire Mesh to 0.00000 (with G29 P0 ) and go print a Mesh Validation Pattern

In a perfect world I may only have to probe the bed once, but since I am running into issues finalizing the mesh, I have had to do it many time. The one time I did actually complete a mesh, I failed to save it before resetting my printer (autosave would be a great addition). Attempting to print with the mesh zeroed would at the very least, permanently damage my print surface and at worst break something. My bed is 1-2mm above Z0.

ngarafol, 2016-10-31:
Yes wanted to write the same. With beds that are uneven/bent P0 is bad idea
unless user knows how to abort process.

Roxy-3D, 2016-10-31:
> Yes wanted to write the same. With beds that are uneven/bent P0 is bad idea
unless user knows how to abort process.

Yes!   I do understand that and it should be pointed out just in case somebody pulls up this thread in the future.   But still...   If your printer homes the Z-axis in a reasonable place.   You can start the Mesh Validation Pattern and let it continue so long as the nozzle has enough clearance.   And just stop it if things get too hairy.    In that case, you would just want to position the nozzle close to where you homed it and not give G26 the C option.   In that case it will try to stay as close to the starting position as possible, but the nozzle will be crossing back and forth across the area that has already been probed.

closed 2017-10-31 by thinkyhead