MarlinFirmware/Marlin #5122: Would it be possible to leave settings specific to devs printers commented out?

tofifee, 2016-10-31:
In the latest iterations there is a tendency to have settings enabled that are not meant to be enabled by default according to the comments.

e.g.

```
#define PIDTEMPBED
```

Fine combing through the code to catch these is prone to overlooking some of these and distracts from the other changes.

Hope that does not step on anyones toes.

thinkyhead, 2016-10-31:
`PIDTEMPBED` is disabled by default in `RCBugFix` (i.e., 1.1.0).

Blue-Marlin, 2016-10-31:
????
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L333
????

tofifee, 2016-10-31:
Mea Culpa. Was supposed to be in `devel-ubl`.

https://github.com/MarlinFirmware/Marlin/blob/devel-ubl/Marlin/Configuration.h#L313

closed 2017-10-31 by thinkyhead