MarlinFirmware/Marlin #5123: UBL G29 P1 hitting end of travel

Nocturnal42, 2016-10-31:
Once it has finished probing all the points, It drives back to what I assume is supposed to be X0 Y0, but is instead trying to drive the Y axis past the end of travel (probably doing to same with the X axis, but I can't hear that over the Y axis). 

The LCD shows negative positions for the X and Y, the same amount as my probe is offset.

Roxy-3D, 2016-10-31:
In general, the UBL System tries to move the nozzle back to where it started.   But...  This might be the same issue that was reported earlier.   Do you have a Negative X & Y PROBE_OFFSET_FROM_EXTRUDER?   That was what the other reporters of this issue had.   

I believe you can work around the problem by moving the nozzle to the center of the bed prior to doing the command.   And you can also specify the reference point by doing things like G29 P1 X75 Y75 O      Please try those work arounds and report back if they help.     I do have  a machine with a negative X & Y Offset, but it will be a few more days until I can start trying to bring UBL up on it.

Nocturnal42, 2016-10-31:
Logic suggests it's this 

```probe_entire_mesh( X_Pos+X_PROBE_OFFSET_FROM_EXTRUDER, Y_Pos+Y_PROBE_OFFSET_FROM_EXTRUDER, code_seen('O') || code_seen('M') );```

Combined with this in the LEAVE part of probe_entire_mesh()

'''do_blocking_move_to_xy(X_Pos, Y_Pos);'''

Which is not actually the starting point.

Roxy-3D, 2016-10-31:
Maybe...  I do know there is an issue that has to be fixed.   But the probe_entire_mesh() is being given the coordinates of where to probe, and the do_blocking_move() is being told where to move the nozzle.

Nocturnal42, 2016-10-31:
Unless I am reading it wrong (and it's nearly 5am here, so that's very possible), the blocking move is using the coordinates passed to the function. Which in my case is 0-30, 0-60.

Roxy-3D, 2016-10-31:
> Unless I am reading it wrong (and it's nearly 5am here, so that's very possible), the blocking move is using the coordinates passed to the function. Which in my case is 0-30, 0-60.

YOU ARE CORRECT!   Thanks!   I'll get that updated.

Nocturnal42, 2016-11-01:
That's fixed it for me.

closed 2016-11-01 by Nocturnal42