MarlinFirmware/Marlin #5126: UBL G29 P2 Steppers not holding position.

Nocturnal42, 2016-10-31:
So after a while, it seems the X/Y stepper motors disabled, allowing the head to move freely. The head still moves from position to position, but the steppers no longer hold position between moves.

Would this be the stepper timeout setting is coming into effect because the automated moves are not resetting the timer?

Roxy-3D, 2016-10-31:
I don't think so...  None of that code has changed.    It might be useful to bump the timeout number way up for the stepper motors and see if that helps.    But the other thing it could be is you might have the current turned up too high on your Step-Sticks and the chips might be over heating.   We have seen that happen cause a loss of position.

Nocturnal42, 2016-10-31:
It's not over heating. The X/Y/Z labels start flashing on the screen, which I believe indicates the steppers have been powered down and positional accuracy is not guaranteed.

Nocturnal42, 2016-11-01:
With DISABLE_INACTIVE_X set to false, X holds but Y/Z stop holding position. So it's almost certainly related to the inactivity timer. I haven't tried increasing the timout yet, but I can tell you, I am not taking 120 seconds to probe a single point.

Roxy-3D, 2016-11-01:
I certainly believe you are seeing something.   When we figure out what is at the 'root cause' we probably should check RC-8 for it also.   It is very possible this issue is more places than just UBL.

closed 2017-10-31 by thinkyhead