MarlinFirmware/Marlin #5129: speed ramp up

ruggb, 2016-10-31:
I currently have ca7275a6 loaded and am trying to figure out if I missed something.
CoreXY
When the motion starts the speed ramps up for 0.5 to 1 sec instead of just starting full speed.
Is this a new "feature" or did I miss something after configuring at least 100 different builds?
Never saw this before.

Sebastianv650, 2016-10-31:
Wrong acceleration setting?

ruggb, 2016-11-01:
my accel are all the same as my previous builds:

#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,12000}  
define DEFAULT_ACCELERATION          5000    
#define DEFAULT_RETRACT_ACCELERATION  12000    
#define DEFAULT_TRAVEL_ACCELERATION   9000    

I'll try fa6bf12
seems to be working now - another ghost
thx

closed 2016-11-01 by ruggb