MarlinFirmware/Marlin #5131: UBL Mesh probe point X/Y display

Nocturnal42, 2016-11-01:
Is there a way to get it to show me the X/Y coordinates for the probe points in the mesh? I don't see anything in the README. I ask because manually jogging the head around to get it close to the mesh points to manually edit them is rather tedious and time consuming, but I see that I can manually specify an X/Y coordinate.

At some point, allowing users to specific the point they want to edit as either a point number, or a row/column type of reference system might be a good addition to speed up the mesh editing process.

Roxy-3D, 2016-11-01:
There are several approaches.   You can use the one you like.   Also, you don't have to be exactly at the Mesh Point.   The UBL System will figure out which Mesh Point is the closest to the nozzle and move the nozzle to that position.

-  When you display a Topography map with the O parameter, you will see one of the numbers enclosed in a square bracket pair.    ie.   [-.0157]    That is the closest Mesh Point to the nozzle.   
-  When editing, invalidating, or filling Mesh Points you can specify a R-epeat count.  R with no number will go until you terminate the operation.   Or you can specify a count such as R 5 to do the 5 closest Mesh Points to the current location.    If you have an area with several Mesh Points that need adjustment, you can move to the middle of them and just give the R option.   You just click and ignore the ones that are good.   It takes almost no time to 'ignore' the good points.
- You Know the MESH_MIN_X & Y number.  You also know the distance between Mesh Points.  (At worst case, if you don't you can use the G29 W to get that information.)      The X coordinate of the 6th Mesh Point from the edge will be MESH_MIN_X + 6 * MESH_X_DIST.   You can calculate the number precisely and move the nozzle to where it needs to be.   But I think you will agree the first two options are easier.

Nocturnal42, 2016-11-01:
So, the answer to my question would be no. 

I don't want to be moving the head around, which largely rules out the first two options (If I don't want to move the head, and if the points I want to modify are all at the bottom two rows of the bed, using the repeat function is a lot of ignoring to get there).

Asides from having to manually calculate the X/Y positions, no I wouldn't agree the first two are the easier options, at least not for me. A lot of the time I am not using a front end, I am entering gcode directly. If I have to write out a G1 X### Y### (several times if I am trying to guesstimate an arbitrary point in the mesh) I may as well just be writing G29 P4 X### Y###.

Roxy-3D, 2016-11-01:
If it is important...   It should not be too difficult too difficult to do.    The problem is more of trying to get the columns to line up nice because PronterFace and Repetier host use proportionally spaced fonts.   And then there is the extra problem that the lines can't be that long.

The most natural way to do it would be to print the may similar to how it prints now but put an (X,Y) coordinate there.   But that would double the amount of characters on the line.    However, for what you want to do, 1 decimal point of precision should be enough because that will get you right on top of the mesh point.  

Another method would be to just print the X numbers on one axis of the graph and the Y numbers on the other axis.    (And in fact, in this case it may not be necessary to even make a graph. )     This would be almost trivial to add to the G29 W command.  

I've added the last method to the G29 W command for you to try.   The W-hat command is not pretty, but it should give you the information you need:
```cpp

    SERIAL_PROTOCOLPGM("X-Axis Mesh Points at: ");
    for(i=0; i<MESH_NUM_X_POINTS; i++ ) {			
    	SERIAL_PROTOCOL_F( blm.map_x_index_to_bed_location(i),1 );
    	SERIAL_PROTOCOLPGM("  ");
    }
    SERIAL_PROTOCOLPGM("\n");

    SERIAL_PROTOCOLPGM("Y-Axis Mesh Points at: ");
    for(i=0; i<MESH_NUM_Y_POINTS; i++ ) {			
    	SERIAL_PROTOCOL_F( blm.map_y_index_to_bed_location(i),1 );
    	SERIAL_PROTOCOLPGM("  ");
    }
    SERIAL_PROTOCOLPGM("\n");
```

Nocturnal42, 2016-11-01:
Thanks, and perfect timing. I just printed a validation mesh, so I will give it a go. The terminal I am using is mono-spaced, so the grid display has never lined up properly.

Roxy-3D, 2016-11-01:
> The terminal I am using is mono-spaced, so the grid display has never lined up properly.

Nocturnal42, 2016-11-01:
Probably not. It does make the in progress grid messy and sometimes hard to fully read, but a fully populated mesh aligns with itself fine (with a slight offset from the current position markers). Just the X/Y labeling for the corners is way out on the right. Which while not visually perfect, is still easily understandable.

Octoprint appears to use monospacing for its terminal, but it's also fixed width (I haven't check, but I'm going to go out on a limb and say 80 characters wide). So unless its a small grid, or you want to drop the number of displayed decimal places, it's never going to fit.

closed 2016-11-01 by Roxy-3D