MarlinFirmware/Marlin #5135: Auto bed leveling  seems not compensate on Y axis

murdock62, 2016-11-01:
In first sorry for my bad english !

On my prusa i3, the auto bed leveling work great on X axis but never on y axis.

Is it normal or i made an error somewhere ?

Many thanks.

thinkyhead, 2016-11-01:
How does it not work? The probing fails? The leveling compensation is wrong?

What version of Marlin are you using?

What form of bed leveling are you using?

murdock62, 2016-11-01:
Hello and thanks for your answer.

I use 1.1.0 RC7 and 3 points variant.

When i'm printing, the z axis move when the head move along x axis, but not when the head move along y axis.

Many thanks again for your help !

thinkyhead, 2016-11-01:
Please test with `RCBugFix` and see if you still get the same results. `RCBugFix` is `RC7` plus hundreds of patches.

nikki-reprap, 2016-11-01:
I have sort of the same problem but it is not compensating on any axis. I am using the latest rcbugfix release. this is my config file.    http://pastebin.com/F6m3f0LD

murdock62, 2016-11-06:
Sorry for this late answer.

Try with rcbugfix but with this version, the print start at 2mm of the bed on the z axis.

I dont understand ...

thinkyhead, 2016-11-07:
@murdock62 Just to be thorough, re-calibrate your `Z_PROBE_OFFSET_FROM_EXTRUDER` (`M851`) and reset your EEPROM with `M502` and `M500`. Then test again.

murdock62, 2016-11-07:
It's work !

The EEPROM was not reset ! (Beginer mistake)

The  compensation seem good.

Many thanks for your help !

closed 2016-11-16 by thinkyhead