MarlinFirmware/Marlin #5137: UBL print insanity

Nocturnal42, 2016-11-01:
See this https://youtu.be/2It5vxwVnbQ

To answer the two most likely questions. No, the gcode does not contain instructions to go for infinity and beyond a handful of lines into the print. Yes, the X axis should be moving, but since the first time this happened (and I wasn't filming) it also ripped out the wiring from my X axis and reconnecting them does not seem to have fixed it.

What I suspect is happening here, is that the print is going outside the bounds of the mesh, leading to an unreal offset. Repeating the print with the UBL disabled does not involve the print head launching into orbit.

Roxy-3D, 2016-11-01:
> What I suspect is happening here, is that the print is going outside the bounds of the mesh, leading to an unreal offset. Repeating the print with the UBL disabled does not involve the print head launching into orbit.

I do know of an issue where moving the nozzle off the bed causes the height to be wrong.    Can you position your part such that it is within the Mesh ?    I have code to gracefully handle the case where the nozzle leaves the defined bed space, but it isn't doing the right thing yet.

Nocturnal42, 2016-11-01:
The part fills most of the bed, so there is not a lot of room for maneuvering it. I can turn off the skirt, which reduces the footprint by 5mm, which may bring it back inside the mesh. I'll just have to use a different method of priming the extruder.

Roxy-3D, 2016-11-01:
G1 E 5.0    ????

There are a couple of ideas on what the best way to fix the 'off the bed' issue.   Some people think we should just use the Z-Height of the closest Mesh point, and extend that outwards off of the Mesh.

Would that work for your bed?

Nocturnal42, 2016-11-01:
That generally leaves a blob or strand that mars the print. Usually I would just lay down a line off the buildtak, but still on the the print bed. I'm pretty sure I can do that here, just need to make sure it's within the mesh.

That seems like the simplest approach. It would certainly work with my bed. I'm curious in what cases it wouldn't work. It seems like it would either have to be a relatively small mesh compared to the bed, or hugely warped.

Roxy-3D, 2016-11-01:
> That seems like the simplest approach. It would certainly work with my bed. I'm curious in what cases it wouldn't work. It seems like it would either have to be a relatively small mesh compared to the bed, or hugely warped.

Well...   I think the biggest concern is Delta's where the bed becomes bowl shapped.  If you go off the bed probably a 0.000 correction is wrong and going to cause the nozzle to jump.   But I'm concerned that even extending the outer most correction won't be enough.    I don't know.  I don't have the code up and running on Delta's yet.

Nocturnal42, 2016-11-01:
I think extending the closest point, is better than correcting to zero (which I am pretty sure would plow my  head into the bed before it reached the edge). At least the nearest point is a known value, anything else seems like it would just be a guess.

thinkyhead, 2016-11-01:
> I think extending the closest point, is better than correcting to zero

It's true. The function `bilinear_z_offset` in the `RCBugFix` branch demonstrates how to do this.

closed 2017-10-31 by thinkyhead