MarlinFirmware/Marlin #5142: not really an issue more like a question

locutus40, 2016-11-02:
i just purchased a bltouch probe.. i was wondering in marlin how to set it up...

i see this 
// The BLTouch probe emulates a servo probe.
//#define BLTOUCH

// Z Servo Probe, such as an endstop switch on a rotating arm.
//#define Z_ENDSTOP_SERVO_NR 0
//#define Z_SERVO_ANGLES {70,0} // Z Servo Deploy and Stow angles

i know to uncomment define bltouch.. i got the angles from the website 90 and 10 i put 90 where 70 should be?

so that takes care of the angles but what is servo_nr 0??? what does that mean..????  heres where i bought it..
the info scroll down http://www.ebay.com/itm/201548081668?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

do i have to add the mcodes they show to my print every time? say i want to print something do g28 then g29... marlin knowing the angles should stick out the probe and retract it... on its own so i dont see why they would show you the m codes maybe to manually do it to extend the arm???? for testing...

anyway thanks for any help...

Roxy-3D, 2016-11-02:
You should not need to use any of the MCodes.   Once you turn on the BLTOUCH option, it might be wise to raise the nozzle by 50mm and tell the printer to home.   When it starts trying to home the Z axis you should see the BLTouch deploy and you can reach under it and trigger it with your finger safely away from the bed.  (And if it doesn't trigger...  pull the power plug!)

If it does trigger...  You are set to do G28 and G29 for real.

locutus40, 2016-11-02:
yea because it pops out first i get that part.... i useually do that to heat up the hot end anyway and wait for the ooze to stop before homing etc.. so thats covered..

so i pretty much have to redo my calibration numbers mine now is 29 -21 and 0.90 for z hight in the z probe section so id have to re do the z hight?

hows that gunna work  ill figure it out..

i guess the m codes are to extend it to do manual testing and to just do calibration.. 

but what is this ???  //#define Z_ENDSTOP_SERVO_NR 0  its next to this //#define Z_SERVO_ANGLES {70,0} // Z Servo Deploy and Stow angles

i know its the deploy and retract i can figure out where to put the numbers.. but my main concern now is that NR 0 what the heck is it??

Bob-the-Kuhn, 2016-11-02:
Here's the lines from my config.h file for BLTouch.

```
//#define FIX_MOUNTED_PROBE

// The BLTouch probe emulates a servo probe.
// The default connector is SERVO 0. Set Z_ENDSTOP_SERVO_NR below to override.
#define BLTOUCH
// Z Servo Probe, such as an endstop switch on a rotating arm.
//#define Z_ENDSTOP_SERVO_NR 0
#define Z_SERVO_ANGLES {10,90} // Z Servo Deploy and Stow angles
....
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command
```
In addition I changed the pin definition to
`#define SERVO0_PIN         57`
Which just means any available digital pin is acceptable.

Roxy-3D, 2016-11-02:
> #define Z_ENDSTOP_SERVO_NR 0 
> i know its the deploy and retract i can figure out where to put the numbers.. but my main concern now is that NR 0 what the heck is it??

Don't worry about those.  If you turn on the #define BLTOUCH option, that stuff automatically gets setup.  SERVO_NR_0 is says the Z_ENDSTOP is on servo number 0.  (and the servo angles for deploy and stow get automatically setup too.)

closed 2017-10-31 by thinkyhead

athulsnair, 2018-06-05:
Do I have to change end stop pull up settings for BL touch??
I several videos I've seen that, but those were old FE without #define BLTOUCH

Sidans, 2019-01-28:
And what is NUM_Servo and how do I specify what number should be set for it?
thanks