MarlinFirmware/Marlin #5143: Servo does not move

felice64, 2016-11-02:
Hi there,
I have Marlin 1.1.0-RC6 and a Rumba board.
I am trying to implement the ABL but the servo to move.
 
I have connected the servo (SG90) on exp.3 pins:
2 (+5V)
4 (-)
and then I tested the servo on both pin 6 and 5 for the PWM
 
In Marlin I have activated the auto bed leveling options and the R/C servo support, in fact if I send a G29 the printer moves and does the mapping of the bed on 4 points. I have manually pressed the Z end stop to cause the routine to complete and return the mapping in Pronerface. I have attached my Configuration file.
 
I have tried to send the command manually to move the servo M280 P0 S90, I tried the same command on P1 and P2 on both pins 4 and 5, but nothing....I cannot move the servo. I have checked the servo functionality before and after the test on rumba with an Arduino board and the servo is functional.
 
I checked in pins_RUMBA.h and the servo 0 is assigned as #define SERVO0_PIN 5
 
Please help.

Roxy-3D, 2016-11-02:
> I have tried to send the command manually to move the servo M280 P0 S90, 

I may be mistaken, but isn't the power up and reset condition for the servo 90 degrees?  If so, you are not telling it to move anywhere even if it is connected up properly.

felice64, 2016-11-02:
@Roxy-3D I am not sure about the meaning of your comment. Isn't the M280 P0 S90 the right command to move it manually?
However, the servo does not move even when I send a G29. The probing sequence starts but the servo does not move.

Roxy-3D, 2016-11-02:
> Isn't the M280 P0 S90 the right command to move it manually 90°? 

That is the right command to move it to 90°.   But if it starts at 90°, you won't see movement by doing that.   That was my point.      Do you have these items setup in your Configuration.h file?
```cpp
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command
#define DEACTIVATE_SERVOS_AFTER_MOVE
#define SERVO_DELAY 500
#define Z_ENDSTOP_SERVO_NR 0
#define Z_SERVO_ANGLES {40,85} // Z Servo Deploy and Stow angles
```

felice64, 2016-11-02:
@Roxy-3D the only line that was not uncommented is the
`#define DEACTIVATE_SERVOS_AFTER_MOVE`
and then I changed the angles as per your example. I tried again but it is not moving with the M280, neither on pin 5 or 6. 
However I noted something strange, when I switch on the board with the servo connected it moves few degrees on both pins without any command, just switching it on..

Roxy-3D, 2016-11-02:
Those numbers are out of my Configuration.h file.   They weren't meant to be useful for you.   Just that those things should be factored into your setup.  

Does the RUMBA board have enough power on the +5v to power the servo?  I know that is an issue with RAMPS where you need to provide an external power source for the servo.

felice64, 2016-11-02:
When I use the micro server the extruder and all the motors are not in use so I suppose this is not the issue at present. In my original question above, I have attached my config and pin files. Have you noted anything that may cause the issue?

Roxy-3D, 2016-11-02:
You have 3 servos declared.  Do you have 3 angle pairs available for the three servo's ?   Because I wouldn't be surprised if it is getting something outside that structure.

Roxy-3D, 2016-11-02:
> I have Marlin 1.1.0-RC6 and a Rumba board.

Really...   You should load up RCBugFix.   Some of the servo stuff has changed there.   It will be easier to help you if we are debugging RCBugFix.

felice64, 2016-11-02:
I have never done something like that but I will have a look for sure. Thanks.

Bob-the-Kuhn, 2016-11-02:
Maybe something else is assigned to the PWM pins. The latest RCBugFix has a command that'll tell you what pins are assigned to PWMs.  

In config adv enable pins debugging (at the very end of the file).  Then enter "M43 I"  for a list of all the currently defined pins.  Sometimes the PWM label doesn't appear until you've actually tried to use that PWM.

felice64, 2016-11-02:
Could you please elaborate a bit more on this RCBugFix. I do not know what is about. Please point me to some tutorial or procedure to download and use it.
Thanks a lot for the help. As soon as I will be able to debug I will share the results

Bob-the-Kuhn, 2016-11-02:
Pins 4 & 5 can't be used as a hardware PWM signals.  They're in use by the system.

Try pin 10.   I don't see it defined in your pins file.

Bob-the-Kuhn, 2016-11-02:
RC7 and RCBugFix are both from GitHub https://github.com/MarlinFirmware/Marlin .

To get RCBugFix instead of RC7 click the "6 branches" and select RCBugFx.  Then download the ZIP file.

You may have to use the config fles that downloaded and you may need to update some pn names.

Bob-the-Kuhn, 2016-11-02:
I for mention - check the version.h file to make sure you got the right branch.

felice64, 2016-11-02:
@Bob-the-Kuhn   pin 5 is defined for pwm1 in pins_RUMBA file above. Should I change it to pin 10?

Bob-the-Kuhn, 2016-11-02:
yes

Roxy-3D, 2016-11-02:
From: http://www.reprap.org/wiki/RUMBA

> Correction On some rumba boards (e.g. board blue LKDR20150602) the silk screen labeling on EXP1 and EXP2 is wrong (- or the Smart Controller silk screen labeling is wrong?) - and turn each connector by 180°.

Could that be the source of your error?

felice64, 2016-11-02:
@Roxy-3D the 5 V and the servi-in are on Exp. 3 which is not objector such error.

felice64, 2016-11-02:
@Bob-the-Kuhn I will give a try tomorrow morning when I am sober....

Roxy-3D, 2016-11-02:
> Try pin 10. I don't see it defined in your pins file.

Don't forget to change the definition for the servo in the pins.h file.   As well as physically moving the wire.

Bob-the-Kuhn, 2016-11-02:
I was wrong.  The SERVO0_PIN does not need a hardware PWM.  Any available digital pin will do, including pins 4 & 5.

felice64, 2016-11-03:
@Roxy-3D I changed the pin in pin_RUMBA for server0 to pin 10 but no results. The servo does not move.
I downloaded the RCBugFix from branch 6 and made the configuration. I tested on pin 5 and 6, no success. The servo does not move.
I have attached the configuration I used with the RCBugFix.
The frustration is terrible..
[Configuration_RCBugFix.pdf](https://github.com/MarlinFirmware/Marlin/files/568515/Configuration_RCBugFix.pdf)

thinkyhead, 2016-11-03:
Does `M280 P0` read the servo angle when connected?
Does it read differently when the servo is disconnected?

felice64, 2016-11-03:
@thinkyhead 
Yes it does.
servo0 85

felice64, 2016-11-03:
I was considering a work around. What about using one of the fan pin?
I am using fan1 for the extruder cooling but fan0 is available.
Would it be possible to use such pin for the SG90Servo (5V)?

thinkyhead, 2016-11-03:
Can you get the servo to work (on pin 5) if you disconnect the stepper motors? Also check the amperage on the 5V line. It might be under-powered.

felice64, 2016-11-04:
@thinkyhead 
I tried with all the stepper drivers out but still does not move.
I measured the voltage between pin 4(-) and PWM1. Quite strange I measure 12 V. Between pin 2 and 4 I have 5V

thinkyhead, 2016-11-04:
> between pin 4(-) and PWM1

Which pin is `PWM1`?

felice64, 2016-11-04:
@thinkyhead 
I tried both 4 and 5. PWM1 is 5

thinkyhead, 2016-11-04:
Well, I'm stumped. I suggest hitting the RepRap forums for extra help.

felice64, 2016-11-04:
@thinkyhead I start thinking that I may have a hardware issue with the board. But it is new and I never messed it up. Not sure if I should buy another one to give it a try or change brand.
I will seek support as per your suggestion on the RepRap forums.....lets see if someone else has something to say. Thanks a lot for now. I will leave this issue open just in case.

Roxy-3D, 2016-11-04:
Seriously....  Check the voltage on the Servo power wires.    Please confirm you have a strong and steady 5 volts there.    A lot of times the problem is the servo takes more current than the board can supply and then it doesn't work right.

felice64, 2016-11-04:
@Roxy-3D 
Actually I did it.
In standby with the servo connected it is 6,568 V between pin 2 and 4
When I send the command M280 P0 S80, it is exactly the same, so there is no current drawn.

I noted that if I uncomment this line //#define DEACTIVATE_SERVOS_AFTER_MOVE, when I switch the board on/off the servo moves few degrees on its own.

felice64, 2016-11-05:
I have a doubt. I have compared the exp. 3 pins diagram with the physical connector and something is different.
This is the diagram with 14 pins:
[image]

This is the physical connector, it has 16 pins!:
[image]

I start thinking that the first 2 are not belong to the Exp. 3 connector.

thinkyhead, 2016-11-05:
Correct. Those other 2 pins are separate.
[image]

felice64, 2016-11-05:
But then the servo shall be connected like this?
[image]

And Marlin has nothing to do with this issue!
Sorry for that to the people that supported me but I had the wrong information and the connector layout drove me to make a mistakes.

thinkyhead, 2016-11-05:
Congrats! Hope it all works well from here…

closed 2016-11-05 by thinkyhead

felice64, 2016-11-05:
Thanks to all of you guys.
I will test on Monday.
I am closing this post because the issue is for sure not relevant Marlin but the hardware and wiring.

Thanks again