MarlinFirmware/Marlin #5146: Autolevel Test - Last Marlin-RCBugFix 02/11/2016 19.43

genzo76, 2016-11-02:
With the latest release I have seen a lot of progress, congratulations.

But it is not yet perfect, I did a test over the entire length of the plate.

Here are the photos of varei layer:

Object printed: [https://goo.gl/photos/LYgfWu7YrYVKrCna8](https://goo.gl/photos/LYgfWu7YrYVKrCna8)
First layer: [https://goo.gl/photos/g3pEfhsYWnJTasd19](https://goo.gl/photos/g3pEfhsYWnJTasd19)
Second layer: [https://goo.gl/photos/U1WmTTsuoMcyJe1a9](https://goo.gl/photos/U1WmTTsuoMcyJe1a9)
Thirt Layer: [https://goo.gl/photos/cCifXLzUn38jNb2U9](https://goo.gl/photos/cCifXLzUn38jNb2U9)

As you can see the corners are not perfect yet and the priomo layer is not uniform.

thinkyhead, 2016-11-03:
<details><summary>This is my configuration</summary>
```cpp
//===========================================================================
//============================ Auto Bed Leveling ============================
//===========================================================================
// @section bedlevel

/**
 * Select one form of Auto Bed Leveling below.
 *
 *  If you're also using the Probe for Z Homing, it's
 *  highly recommended to enable Z_SAFE_HOMING also!
 *
 * - 3POINT
 *   Probe 3 arbitrary points on the bed (that aren't collinear)
 *   You specify the XY coordinates of all 3 points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - LINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - BILINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a mesh, best for large or uneven beds.
 */
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
//#define DEBUG_LEVELING_FEATURE

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define ABL_GRID_POINTS_X 3
  #define ABL_GRID_POINTS_Y ABL_GRID_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 0
  #define RIGHT_PROBE_BED_POSITION 185
  #define FRONT_PROBE_BED_POSITION 34
  #define BACK_PROBE_BED_POSITION 170

  // The Z probe minimum outer margin (to validate G29 parameters).
  #define MIN_PROBE_EDGE 10

  // Probe along the Y axis, advancing X after each column
  //#define PROBE_Y_FIRST

#elif ENABLED(AUTO_BED_LEVELING_3POINT)

  // 3 arbitrary points to probe.
  // A simple cross-product is used to estimate the plane of the bed.
  #define ABL_PROBE_PT_1_X 15
  #define ABL_PROBE_PT_1_Y 180
  #define ABL_PROBE_PT_2_X 15
  #define ABL_PROBE_PT_2_Y 20
  #define ABL_PROBE_PT_3_X 170
  #define ABL_PROBE_PT_3_Y 20

#endif
```
</details>

thinkyhead, 2016-11-03:
Are you using `RCBugFix`? Always test with this branch before reporting an issue.

If you did test with `RCBugFix` then test again, as bilinear leveling was patched earlier today.

oysteinkrog, 2016-11-03:
@genzo76: Quick tip, next time you test you can make note of the current commit hash, that way you can tell us you tested that version, and we can tell if you have all the latest fixes etc.
E.g. the latest commit on RCBugFix as of right now is 28a039c0e1fea8a5e0c429f12d6d14c772c2b50f

genzo76, 2016-11-04:
thinkyhead, yes I have used the RCBugFix del 02/11/2016 at 20:00.

oysteinkrog, ok I right, for the future I insert the last commit in the ticket.

thinkyhead, 2016-11-04:
@genzo76 Do you get better/worse/same results using an even versus an odd number of probe points?

genzo76, 2016-11-05:
I have used 3 point in the setting, as you  see my configuration attached at the ticket.

My personal impression is a inversion of the X and Y.

thinkyhead, 2016-11-05:
@genzo76 Try this, reverting the code to its 1.0.2-2 version:
```diff
-        vector_3 planeNormal = vector_3::cross(points[0] - points[1], points[2] - points[1]).get_normal();
-        if (planeNormal.z < 0) {
-          planeNormal.x *= -1;
-          planeNormal.y *= -1;
-          planeNormal.z *= -1;
-        }
+        vector_3 planeNormal = vector_3::cross((points[0] - points[1]).get_normal(), (points[2] - points[1]).get_normal()).get_normal();
+        if (planeNormal.z < 0) planeNormal.z *= -1;
```

genzo76, 2016-11-05:
I'm sorry, I had tried this solution proposed of you, but not  seen the difference from the two solution.

The problem not is this.

thinkyhead, 2016-11-06:
@genzo76 The implication is that planar bed leveling has always been broken. I doubt this is the case. Do you get a similar result if you use `AUTO_BED_LEVELING_PLANAR` instead of `AUTO_BED_LEVELING_3POINT`? Because it uses the same technique of creating a planar matrix.

Fenixin, 2016-11-28:
Hello!

I'm having similar issue with bilinear auto bed leveling. I have a very uneven bed so after reading about bilinear auto bed leveling I got excited and tried the new firmware. Right now I'm running the very last RCBugFix (commit 20b0889602b9b67deaeaddad8425c52c4becf70c) and though, after G29, I can see the z motors moving up and down at different speeds as I move the extruder across de bed It doesn't seems to compesate for my bed very well. I haven't printed anything big with this new configuration, but testing the position of the extruder with a piece of paper shows that the extruder sometimes is too far from the bed and in other places scratching the surface.

Is there anything I can provide/test to help solve this issue? I expect to have some free time this week.

Some info: For auto bed leveing I'm using a capacitive probe mounted in the extruder. The repeatibility test seems to give me good results:
```
08:42:23.557 : M48 Z-Probe Repeatability Test
08:43:08.677 : Finished!
08:43:08.677 : Mean: 3.310125 Min: 3.303 Max: 3.314 Range: 0.011
08:43:08.677 : Standard Deviation: 0.003077
08:43:08.677 : X:112.00 Y:107.00 Z:8.35 E:0.00 Count X: 8960 Y:8560 Z:33400
```

Using bilinear with ```ABL_GRID_POINTS_X 3``` I get a bilinear leveling grid:
```
    08:35:01.717 : G29 Auto Bed Leveling
    08:35:44.575 : Bilinear Leveling Grid:
    08:35:44.575 : 0     1     2
    08:35:44.576 : 0 +0.17 +0.24 +0.45
    08:35:44.576 : 1 -0.14 -0.06 +0.23
    08:35:44.576 : 2 -0.10 +0.05 +0.35
    08:35:44.577 : X:195.00 Y:187.00 Z:8.00 E:0.00 Count X: 15600 Y:14960 Z:33400

```
Usingn ```ABL_GRID_POINTS_X 4```:

```
    08:26:42.374 : G29 Auto Bed Leveling
    08:27:56.392 : Bilinear Leveling Grid:
    08:27:56.392 : 0     1     2     3
    08:27:56.392 : 0 +0.16 +0.21 +0.27 +0.44
    08:27:56.393 : 1 -0.08 -0.06 +0.05 +0.26
    08:27:56.393 : 2 -0.16 -0.12 +0.01 +0.23
    08:27:56.394 : 3 -0.13 -0.02 +0.11 +0.34
    08:27:56.394 : X:194.00 Y:187.00 Z:8.01 E:0.00 Count X: 15520 Y:14960 Z:33400
```

I also tried linear (with no better results) using  ```ABL_GRID_POINTS_X 3```, and the matrix seems to be really different. Is that normal?
```
    08:21:17.962 : Bed Level Correction Matrix:
    08:21:17.962 : +0.999996 +0.000000 +0.002851
    08:21:17.963 : +0.000005 +0.999999 -0.001618
    08:21:17.963 : -0.002851 +0.001618 +0.999995
```

thinkyhead, 2016-11-28:
> the matrix seems to be really different. Is that normal?

That's the [transformation matrix](https://en.wikipedia.org/wiki/Matrix_(mathematics)), which is always 3x3 no matter how many points you probe. If you want to see the actual probed values, you should use the `V4` flag with `G29`.

thinkyhead, 2016-11-28:
For your Bilinear issue, see if probing the bed with even more points helps. You can go as high as 16x16 with the current `RCBugFix`.

Fenixin, 2016-11-28:
Thanks, knowing that is a transformation matrix makes a lot more of sense.

I have tried with 4x4, 5x5 and 6x6 measures in the bed and the results are suspiciously the same.

**The procedure is:** G28, G29 V4 and then lower the extrusor to z=0. Then I use the paper to see how much the extrusor grabs the paper. 

**The results:** In the center the extrusor grabs the paper just in the right amount, the farther I get from the center of the bed (in any direction, by the way is a cartesian printer) the farther the extrusor gets from the center of the bed and the less grab there is. With a distance from the center of the bed of 50mm the paper feels loose.

With the data printed by G29 V4 I have plot a 3d mesh diagram of the bed just in case it helps:

[image]

The difference between max and min is about 0.6mm

In some time I will try with more measures (7x7 and 8x8). Any other tests that I can do? Any other finde tuning needed?

Fenixin, 2016-11-28:
Just tested bilinear with 7x7 measures. Plotting the resulting mesh is consistent with the one sowhed above, the bed has the same form and the difference between the max and the min is pretty much the same. When I move the extruder along the bed I can see that how the extruder moves up and down in the right direction following the surface plotted abve. But the result is the same, once you get too far from the center of the bed the extruder goes too far from the bed.

With no idea of how the inside works, to me it looks like that is overcompensating the defects of the surface of the bed. Is there anthing else I can test? Any information I can provide to help fine tune this?

Thanks for your answers!

thinkyhead, 2016-11-30:
@Fenixin The bed compensation is surely based on the matrix, applying bilinear compensation. I'm busy with a client right now but perhaps I can help troubleshoot this tomorrow.

One thing you can do is issue `M114` at various points and look at the Z stepper count compared to the current Z position. With leveling disabled the stepper count matches up with the current Z by the simple formula `z_in_mm = z_stepper_count / steps_per_mm`. When leveling is enabled the stepper count value will be higher or lower than that formula by some amount.

So move the nozzle around the bed after `G29` and do `M114` at various points. Post the output of `M114` here along with the mesh that `G29` reports, and we can compare the values to see if they match up.

Fenixin, 2016-12-01:
Thanks for the answer.

Here are the values you asked for:

```
08:30:01.092 : G29 Auto Bed Leveling
08:30:09.567 : Bed X: 30.000 Y: 30.000 Z: 3.391
08:30:14.014 : Bed X: 65.000 Y: 30.000 Z: 3.448
08:30:18.475 : Bed X: 100.000 Y: 30.000 Z: 3.435
08:30:22.936 : Bed X: 135.000 Y: 30.000 Z: 3.469
08:30:27.351 : Bed X: 170.000 Y: 30.000 Z: 3.514
08:30:31.796 : Bed X: 170.000 Y: 62.000 Z: 3.404
08:30:36.337 : Bed X: 135.000 Y: 62.000 Z: 3.345
08:30:40.911 : Bed X: 100.000 Y: 62.000 Z: 3.308
08:30:45.469 : Bed X: 65.000 Y: 62.000 Z: 3.307
08:30:50.074 : Bed X: 30.000 Y: 62.000 Z: 3.264
08:30:54.696 : Bed X: 30.000 Y: 94.000 Z: 3.201
08:30:59.334 : Bed X: 65.000 Y: 94.000 Z: 3.226
08:31:03.955 : Bed X: 100.000 Y: 94.000 Z: 3.228
08:31:08.544 : Bed X: 135.000 Y: 94.000 Z: 3.283
08:31:13.086 : Bed X: 170.000 Y: 94.000 Z: 3.337
08:31:17.596 : Bed X: 170.000 Y: 126.000 Z: 3.344
08:31:22.170 : Bed X: 135.000 Y: 126.000 Z: 3.283
08:31:26.807 : Bed X: 100.000 Y: 126.000 Z: 3.231
08:31:31.445 : Bed X: 65.000 Y: 126.000 Z: 3.225
08:31:36.099 : Bed X: 30.000 Y: 126.000 Z: 3.192
08:31:40.704 : Bed X: 30.000 Y: 158.000 Z: 3.234
08:31:45.278 : Bed X: 65.000 Y: 158.000 Z: 3.287
08:31:49.851 : Bed X: 100.000 Y: 158.000 Z: 3.306
08:31:54.377 : Bed X: 135.000 Y: 158.000 Z: 3.345
08:31:58.936 : Bed X: 170.000 Y: 158.000 Z: 3.394
08:31:58.936 : Bilinear Leveling Grid:
08:31:58.937 : 0     1     2     3     4
08:31:58.937 : 0 +0.06 +0.12 +0.10 +0.14 +0.18
08:31:58.937 : 1 -0.07 -0.02 -0.02 +0.01 +0.07
08:31:58.937 : 2 -0.13 -0.10 -0.10 -0.05 +0.01
08:31:58.938 : 3 -0.14 -0.11 -0.10 -0.05 +0.01
08:31:58.938 : 4 -0.10 -0.04 -0.02 +0.01 +0.06

```

And here are the values of M114 issued in three points of the matrix
```
(point 0,0)
08:38:32.045 : X:30.00 Y:30.00 Z:0.00 E:0.00 Count X: 2400 Y:2400 Z:242
point(0,4)
08:41:23.739 : X:170.00 Y:30.00 Z:0.00 E:0.00 Count X: 13600 Y:2400 Z:735
point(2,2)
08:47:39.027 : X:100.00 Y:94.00 Z:0.00 E:0.00 Count X: 8000 Y:7520 Z:-410
```

My steps per unit as in ``` configuration.h```:
```
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4000, 95.2 }
```

Using the formula you provide (and suppossing a row in the matrix moves along the x coordinate in the printer):
point (0,0)-> (x,y)=(30,30,0) -> 242/4000=0.0605 (expected +0.06)
point (0,4) -> (x,y)=(170,30,0) -> 735/4000=0.18375 (expected +0.18)
point (2,2) -> (x,y)=(100,94,0) -> -410/4000=-0.1025 (expected -0.10)

So it looks ok.

Yesterday I changed the z probe speed to 1/4, 1/6 and 1/8 of the homing speed but it didn't make any difference in accuracy.

Fenixin, 2016-12-01:
Printing this thing: 
http://www.thingiverse.com/thing:1306709

Gives this result:
https://dl.dropboxusercontent.com/u/4351869/Frame%20printed.jpg
You can see how the plastic lines get thinner in the boundary of the bed.

Thinking about hardware problems that can cause this, the only thing that comes to my mind right now is that the capacitive sensor is more sensitive as it gets to the boundaries of the bed (so the printer thinks the bed is closer). That would explain everything. But I don't really know if that makes sense.

thinkyhead, 2016-12-02:
@Fenixin Something is definitely amiss with the measurement. The `M48` command allows some experimentation to see how well the probe is reading. You can also try doing some manual probing with `G30` and see if the distance measured with a caliper from the nozzle to the bed is the same after probing at different points. The `G30` command now takes X and Y arguments so you can probe selected points more easily.

Fenixin, 2016-12-05:
@thinkyhead, thanks for your answers. I won't have time to play around this week but I will definitely play with those g-codes when I get some time.

thinkyhead, 2016-12-05:
In your own time. Looking forward to your results.

Fenixin, 2016-12-13:
So after some test (in RC8) it seems that the problem is the capacitive probe or the way is mounted in the 3d printer.

Measures made by hand don't seem to completely agree with those made with the probe but, somehow, the probe seems to be repeatable in its measures. Also the deviation from probe measures and by hand measures seems to be pretty small and repeatable.

I will try to move the probe to other position, right now it's near the extruder, I think I will place it in the X carriage. Any help/suggestion is welcome.

I don't think this is a problem with the firmware anymore, so feel free to close the issue. Thanks a lot for your answers

closed 2017-02-11 by thinkyhead