MarlinFirmware/Marlin #5147: Extruder fan not starting as the extruder heats up

mperdue, 2016-11-02:
I'm having trouble getting the extruder fan to run when the extruder heats up.

I have the following entries in configuration_adv.h

#define E0_AUTO_FAN_PIN -1
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   255  // == full speed

which would seem to indicate that the fan should start at full speed when the temperature hits 50c but it does not.

in the file pins_AZTEEG_X3_Pro.h I have

// Fans/Water Pump to cool the hotend cool side.
#define ORIG_E0_AUTO_FAN_PIN 5
#define ORIG_E1_AUTO_FAN_PIN 5
#define ORIG_E2_AUTO_FAN_PIN 5
#define ORIG_E3_AUTO_FAN_PIN 5

Which seems to match the settings in previous versions

Anyone have any ideas?

mperdue, 2016-11-02:
OK, I'll give that a shot. But then what is the purpose of the ORIG_E#_AUTO_FAN_PIN defines?

thinkyhead, 2016-11-03:
@mperdue The defines in `Configuration_adv.h` are _overrides_ of the values from the pins files. Overriding the value and setting it to -1 disables it. To use the default setting (i.e., as set by `ORIG_E0_AUTO_FAN_PIN`) just comment out the definition in `Configuration_adv.h`.

mperdue, 2016-11-03:
The defines in Configuration_adv.h are overrides of the values from the pins files. Overriding the value and setting it to -1 disables it. To use the default setting (i.e., as set by ORIG_E0_AUTO_FAN_PIN) just comment out the definition in Configuration_adv.h.

@thinkyhead Certainly that works, it just seems to me that it would be better to have the lines commented out and add them back if you want to disable them or override them. It's no big deal, it's just different than I do it when I code. I can learn to do things a different way. :-) Thanks.

thinkyhead, 2016-11-03:
> it would be better to have the lines commented out

Indeed. In fact, my original thought was to replace the pin overrides with option flags, but step 1 was to simply rename the pins and supply defaults.

mperdue, 2016-11-03:
@thinkyhead BTW, the current rcbugfix is working pretty well with my Core-XY machine.

closed 2016-11-03 by thinkyhead