MarlinFirmware/Marlin #5149: UBL Idiot proofing G26 P

Nocturnal42, 2016-11-03:
Just a thought, but perhaps this should not extrude an infinite amount of filament. At some point, Marlin should say enough is enough. Some idoit (ie Me), has walked away and gotten distracted, and I should probably stop creating interesting swirls of filament on the desk.

thinkyhead, 2016-11-03:
Is this not working for you?
```cpp
// This option prevents a single extrusion longer than EXTRUDE_MAXLENGTH.
// Note that for Bowden Extruders a too-small value here may prevent loading.
#define PREVENT_LENGTHY_EXTRUDE
#define EXTRUDE_MAXLENGTH 200
```

Nocturnal42, 2016-11-03:
```               while( !G29_lcd_clicked() ) {
                        //chirp_at_user();
                        destination[E_AXIS] += 0.25;
                        mesh_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS],
//                                              planner.max_feedrate[E_AXIS]/(15.0), 0, 0xffff, 0xffff );
                                                planner.max_feedrate[E_AXIS]/(15.0), 0 );

                        stepper.synchronize();
                        set_destination_to_current();
                        idle();
                }
```
I could be wrong, but since this is just continually extruding 0.25mm, it doesn't count as a single extrude.

Roxy-3D, 2016-11-03:
I can certainly add code to limit the amount of the extrusion to prime the nozzle.  But at some point, we do have to trust the user to tell us what to do also.     There are some people complaining that G26 feels too much like a UNIX utility with way too many options and capabilities.   And then we have other people saying "If I'm careless and tell G26 to prime the nozzle and I walk away to get coffee, I want it to be smart and do the right thing."

It will only take  a couple of lines of code to limit how much filament is extruded to prime the nozzle.   But do we really want to do that?

thinkyhead, 2016-11-03:
"chirp_at_user" 😀

Nocturnal42, 2016-11-03:
>at some point, we do have to trust the user to tell us what to do also

Endstops, soft endstops, max extrude length, thermal overrun, cold extrusion protection, the entirety of SanityCheck.h etc. These all exist because sometimes, the firmware (or compiler for SanityCheck.h) needs to make the decisions to NOT trust what the user has told it to do. Sometimes the user makes mistakes, poor choices, or fails to notice important details and ignoring the user is the better / safer choice. I think an infinite extrude falls within those bounds.

I commented chirp_at_user() out when it was missing and preventing it from compiling. I also undid the change when you fixed that. I must have had a copy stashed and copied it back in subsequently by mistake. I didn't know the prime operation was active because there were several walls, doors, a few talking people and a television between me and the printer at the time. It could have chirped twice as loud, and it still wouldn't have been noticed. As to people all wanting different things

>“You can please some of the people all of the time, you can please all of the people some of the time, but you can’t please all of the people all of the time”

Though I suspect the people complaining it's like a unix command and has to many options (have they even looked at GCODE?), are looking for a simple option that doesn't require them to have to think about or understand how it works. ie a command that essentially embodies ```G29 P1; G29 P2 H 5; G26 P 10; G29 P4 R; G29 S 0; G29 A``` in one carefully guided hit, is probably what they are looking for.

Roxy-3D, 2016-11-03:
I'll add the code to limit the amount of filament that can be extruded during the 'prime' function.   It really isn't that much code.    I'll keep my fingers crossed that people don't want me to limit how much filament gets extruded during the actual printing of the test pattern.    :)

And really...    I do appreciate the suggestions on how to improve things.

Nocturnal42, 2016-11-03:
That's good, because I'm really struggling to constrain my urge to complain..... argh.... can't... hold... it.... in.... G29 HOMES ON EVERYTHING! *pants* Sorry. That one drives me up the wall. I try telling myself it's a feature, it's more efficient this way, that I will probably want to do something that requires things to be homed very soon. I know its not important, but the wait for homing on something that doesn't need it gets me every time.

Roxy-3D, 2016-11-03:
> That's good, because I'm really struggling to constrain my urge to complain..... argh.... can't... hold... it.... in.... G29 HOMES ON EVERYTHING! 

Well....    a couple of comments on that!    G29 does need to have the X & Y homed to do its work.   And UBL's G29 is not unique on that.   Starting with the very first G29 for 3-Point Leveling, it insisted the machine was homed first.

But G29 will automatically home the machine for you if it hasn't already been done.   (Which is new behaviour and simplifies things!)   And...   If you *KNOW* you don't need the machine homed you can specify the 'N' option which over rides the G29 insistence that the machine be homed.   That is of limited real world use.   But that function is in there because I know how you feel.   If I bring up the printer and just want to see the Topographical Map, or do a 'W'hat option,  I get really tired of waiting for the home operation.   Especially on my printer that has a very very slow Z-Axis!

Nocturnal42, 2016-11-03:
You had me at 'N'. Your entire post could have just been 'N'.

Looks like the README is slightly out of date compared to the header in G29_Unified_Bed_Leveling.cpp.

Roxy-3D, 2016-11-03:
> Looks like the README is slightly out of date compared to the header in G29_Unified_Bed_Leveling.cpp.

ReadMe is back up to date.   I wish the Mark-Up language was easier to use to format the text.  I can't get all the columns to line up right.

closed 2016-11-03 by Nocturnal42

thinkyhead, 2016-11-03:
> I can't get all the columns to line up right.

I can't | get all
------------ | -------------
the | columns to
line up | right.

Roxy-3D, 2016-11-03:
@thinkyhead    I think I can use the Pipe   (vertical bar) to do columns.   I just haven't gotten to it yet.  I'm busy in the lcd code today.

thinkyhead, 2016-11-03:
I saw you were probably referring to pre-formatted text, so I deleted the comment about table formatting. I'm sure you can get things aligned in your plaintext editor.