MarlinFirmware/Marlin #5151: UBL G29 Z # Question

Nocturnal42, 2016-11-03:
How exactly does this work? I mean I can see with ```G29 W``` that a it's setting a Z offset, but I can't seem to store that offset. M500 doesn't do it, nor does saving the mesh, and there doesn't seem to be any way of applying it directly to the mesh itself. I'm assuming the offset is doing something, I don't see a difference when printing, but that may be because my offset is too small.

Roxy-3D, 2016-11-03:
Yikes!!!   Sorry about that!    

The history of the Z-Offset is why it is included (currently) in the UBL System.   But I'm working to remove it.   It adds a lot of run time floating point calculations for not much benefit.    There are a lot of other ways to shift the nozzle's height.   For example M851 and Z-BabyStepping.    The Z-Offset is just adding complexity to all of the calculations.

But what I'm going to do is allow the user to shift the entire Mesh up or down much like the Mesh Editor or Z-BabyStepping operates.   That will replace most of the functionality that is lost by taking the Z-Offset away.    I'm still working on the code, but my recommendation is you set the Z-Offset to 0.000 and just leave it there.

Nocturnal42, 2016-11-03:
Does M851 affect the mesh after it has been created, or only during probing? Also, is there a way to auto probe the bed that doesn't invalidate the entire mesh?

Roxy-3D, 2016-11-03:
> Does M851 affect the mesh after it has been created, or only during probing? 

M851 only's function is to change the Z_PROBE_OFFSET_FROM_EXTRUDER number.   And that updated number can be saved in the EEPROM for automatic, future use.    It has no direct interaction with the Mesh.   But it does directly change the nozzle's height from a Z-Home operation.

> Also, is there a way to auto probe the bed that doesn't invalidate the entire mesh?

I'm not sure I understand what you are trying to accomplish.   But if for some reason you have a partially defined Mesh, you can continue the automated probing with G29 P1 C     If for some reason you have totally wrong values in some area of the Mesh, you can move the nozzle there, and do a G29 I 10   (to clear out 10 Mesh Points in that area).     You could then do a G29 P1 C to repopulate those Mesh Points with freshly acquired probes.

Nocturnal42, 2016-11-03:
>I'm not sure I understand what you are trying to accomplish

It's called "Oh look, I just finished my mesh, printed off a test pattern and realized that my Z_PROBE_OFFSET is wrong", so basically, I want (or wanted at the time) to be able to automatically reprobe all those points, without having to manually reprobe all the manual points.

> It has no direct interaction with the Mesh. But it does directly change the nozzle's height from a Z-Home operation.

Only if you use the probe for homing. My Z axis homes to the dual endstops at the top (which also levels my X axis relative to my frame).

Roxy-3D, 2016-11-03:
So...  You could try this:     Move the nozzle to the corner of the bed such that the probe is furthest from the center of the bed.   You can then do G29 I 10 O   commands and watch the area of the Mesh where the probe can reach be cleared.   As you repeat the G29 I 10 O command, pay attention that the invalidated region is not getting too close to the manually probed (or filled) areas.   If it is...  Move the nozzle such that the points you want to invalidate are the closest ones to the nozzle.   (And bump the repeat count down!!!)

closed 2017-10-31 by thinkyhead