MarlinFirmware/Marlin #5153: The head is not stopping on the endstops after every homing

RajaaKahel, 2016-11-03:
Hello, Please I need help to change the Z hight after every homing. It's moving down after every homing about like 10cm. I don't want that to happen. Where I can change it please?
I tried to put // before "do_blocking_move_to_z(delta_clip_start_height);" and it was ok, but when the head is moving down to do auto-leveling; it did crazy leveling in the air near the bed and the first point only touched the bed in the center (not on the side as usual).
Thank you!
Best regards

Blue-Marlin, 2016-11-03:
Probing begins with a sideways move to the first probing point. That is impossible with a DEATA. For that the nozzle is driven down after homing by exactly the move you removed. Put it in again an all is well.

Bob-the-Kuhn, 2016-11-03:
I don't know if this applies, but on a Cartesian machine, the following #defines (in configuration.h) determine what X, Y & Z are set to when doing a G28 command.  Definitely had me pulling my hair out for a while.

```
// Travel limits after homing (units are in mm)
#define X_MIN_POS 0        // G28 sets X position to this value after homing
#define Y_MIN_POS 0        // G28 sets Y position to this value after homing
#define Z_MIN_POS 0        // G28 sets Z position to this value after homing
```

thinkyhead, 2016-11-04:
On a DELTA the XY are 0,0 after homing, and the Z is `MANUAL_Z_HOME_POS`. Note that the min/max positions and the home positions are not the same thing.

RajaaKahel, 2016-11-05:
Thank you all. I'll try to find the best method, because I don't want the nozzle to crash right on my tall print anymore :).

Blue-Marlin, 2016-11-05:
Is there any special reason for homing your machine with a part on the bed?
If you just want to move the nozzle away from your part when finished use 'G0 X0 Y0 Ztop_of_buildroom'.
G28 is not a 'go to home position' but a 'find where you are'.

RajaaKahel, 2016-11-05:
Thank You Blue-Marlin I got your point, I tied G0 X0 Y0 Ztop_of_buildroom in my G-code, it went down to the center of my deltas bed. What g-code should I write in the ending-script to move the arms up to the endstops and stay there after a print is done?

Blue-Marlin, 2016-11-05:
Replace 'top_of_buildroom' with your printers number for it.

RajaaKahel, 2016-11-05:
Great! thank you!

closed 2016-11-06 by thinkyhead

Barrocas, 2016-11-13:
I also had problems with this.

I'm running a Mini Kossel (Delta), and it had Marlin 1.1.0 RC6, when it did G28 it would go up to homing, hit endstops, lower a bit and slowly go up again, hit endstops again, and stop there.

When I updated to RC7 (and also BugFix), it does the same procedure as in RC6, but instead of stopping at the top (with endstops triggered), it then lowers some 58mm. 
Acceptable when homing at the start, but since I also had a G28 in the end gcode (as default in slicing programs), printing tall parts can be dangerous, because it would lower the hotend on top of the print.

I understood what Blue-Marlin commented (it would just require to modify the end gcode), but I'm also curious to know where this setting is modified, since I would also like to have the RC6 behaviour in another delta that I'm building and need to do some testing.
Is there a way to set G28 in BugFix as it was in RC6, to hit the endstops and just stay there?

thinkyhead, 2016-11-15:
Obviously a G28 at the end of a print is a terrible idea. Be sure to smack whoever is adding that to their slicers as a standard thing.

Barrocas, 2016-11-15:
Lol, it was fine when using RC6 on delta machines, since it would just home
to the top and rest there :)

Can you point in what file can whe modify that setting, if possible?

-- 
Cumprimentos,
D.er Hugo Barrocas

thinkyhead, 2016-11-16:
> Can you point in what file can we modify that setting, if possible?

At this time there's no option to disable the lowering of the carriage to the "safe zone" at the end of `G28`. The best option is to use `G0 Z300` (or whatever your top-most Z is) at the end of the print.

Barrocas, 2016-11-16:
Ok, thanks.

-- 
Cumprimentos,
D.er Hugo Barrocas