MarlinFirmware/Marlin #5158: newbee help

stigjoergensen, 2016-11-04:
Hi Gang

I have had a printer for 6 month now - and i have manged to get marlin working on my printer - all is good - but now im a developer - and i wish to contribute... and i have figured out how to get my own copy.. but that dont include all the latest stuff - in particular im missing some code that was merged a few days ago and labeled something like rc8 - its about lights - gcode M355 - where can i find this? - all i can see is the bugfix branch and the RC branch...

and further its seems things are going in circle - at some wiki pages its listed that this is the main marlin code - but other places its listed there is a dev-master brach, but that seems to be obsolete?

so im a bit confused here.

Roxy-3D, 2016-11-04:
RC-7 (and RC-8) are versions when enough chnages to the code have happened that it made sense to mark that point in time.  (Also, any main revision of the Release Candidate such as RC-7 or RC-8 is fairly stable and going to be very usable.)

RCBugFix it the base version (such as RC-7 or RC-8) with bug fixes applied to it.   In general, you should be using RCBugFix unless you have a good reason to backup to the start of a particular Release Candidate.

stigjoergensen, 2016-11-04:
Thanks for writing it more clearly..

but where do i find the code for PR #5133 - when viewing code in the RCBugFix i cannot see the implementation of the above... which i tought would be the place to find it after it have been merged?

/Stig

thinkyhead, 2016-11-04:
> but where do i find the code for PR #5133

It's merged into `RCBugFix` now.

stigjoergensen, 2016-11-05:
Dont know what happened the first time i downloaded the RCBugFix - but when downloading it when i got home the code was there as expected.

Thanks for your help..

closed 2016-11-05 by stigjoergensen