MarlinFirmware/Marlin #5159: Suggestion for PROBE_BED_POSITION with ABL in RCBugFix

jlarsenwy, 2016-11-04:
This is really just a suggestion for the current state of RCBugFix and not an actual issue.

I might be a little slower than most, but I found the `PROBE_BED_POSITION` definitions under Auto Bed Levling to be somewhat confusing.  Marlin assumes the hotend is in the "front right" position when homed (as specified by the little diagram in the Z Probe Options section). This can be very confusing depending on your printer and its orientation.  For example, on my printer the hotend is in the "back right" when looking at the front of it. 

Instead of specifying the boundaries as left/right/front/back relative to the hotend, I think it would be simpler to specify them in terms of X/Y Min and Max values.  It looks like in the UBL branch you've moved towards this with the mesh boundaries, which I do like.   This might be irrelevant if UBL is eventually merged, so maybe another easier fix would just be to add additional comments (or that little diagram) to the Auto Bed Leveling section mentioning the relativity of the movements.  

I know this is somewhat trivial, but it might help out some other people like me.  Thanks for your time.

chuck787, 2016-11-04:
Same here. I have placed my probe front right so the hot end is to the left of the capacitive probe.  This works well in version what I think is version 1.0.2 but I cannot get the Auto Bed Leveling to work in 1.1.0-RCBugFix because I don’t understand the constants required.   Initially it failed the Sanity check then I changed the parameters and both the software and I are totally confused so I reverted and back to 1.0.2 which is a pain because they need different versions of Arduino and neither will compile with the current Arduino 1.6.12

Roxy-3D, 2016-11-04:
If you post a picture of your nozzle/probe, and  Configuration.h along with the Sanity Check error messages...   Somebody here will give you guidance.

thinkyhead, 2016-11-04:
> Marlin assumes the hotend is in the "front right" position when homed

Front-left, actually.

I suggest rotating the machine so that 0,0 is oriented front-left. Then make your settings pretending this is the normal "front" of the machine. Then you never have to touch them again.

chuck787, 2016-11-05:
Thanks Thinkyhead.    I will exchange the extruder and sensor and see if that works.  This started as a RepRapPro with melzi controllers and tri-color.   I converted to Ramps changed all the bearings and everything else except the frame.

[image]

[image]
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/572889/configs.zip)

jlarsenwy, 2016-11-05:
> Front-left, actually, is assumed to be X0 Y0 on most cartesians.

My bad, a typo on my part.  But perhaps this further exemplifies that the assumptions currently in place are not as obvious as one might have thought.  Even so, I shouldn't have to physically rotate my printer to understand the coordinates that are being asked.  This is all probably trivial in the long run, but I thought it was worthwhile to express my opinion on the language that's being used.

Roxy-3D, 2016-11-05:
> Thanks Thinkyhead. I will exchange the extruder and sensor and see if that works. This started as a RepRapPro with melzi controllers and tri-color. I converted to Ramps changed all the bearings and everything else except the frame.

My suggestion would be to only change one thing at a time.   It it a lot easier to make things work if everything doesn't change at the same time.

Those pictures are very helpful.   Let's just get your printer going without changing anything.   After it is going, you can decide if you want to re-orientate it.   What exactly does the Sanity Check complain about?   (I'll look at your Configuration.h file after I know what error message you are fighting.)

thinkyhead, 2016-11-05:
> Even so, I shouldn't have to physically rotate my printer to understand the coordinates that are being asked.

Why not configure the machine so front-left is X0 Y0? Is there some special reason why the back-right must be X0 Y0 in your setup? If it's just a preference, how come?

chuck787, 2016-11-05:
> > Roxy-3D My suggestion would be to only change one thing at a time. 

I agree.  The printer works with the current stable release.  I updated only the firmware to rcbugfix and that does not work because sensor is math positive to hot-end instead of the on the left.  0,0 is front left.  The printer was purchased years ago with hardware for 3 hot-ends so could run 3 colors but I was only using one so it was simple to just put the sensor onto the empty position on the right, no hardware mods.  Swapping sensor and hot end is just four screws and firmware.

jlarsenwy, 2016-11-05:
> Why not configure the machine so front-left is X0 Y0?

That's.... actually a really interesting idea.  I hadn't considered that for some reason.  My first thought is that would put my finished prints behind the X-axis rails of my printer (Makerfarm Pegasus) which would make them harder to get to without moving the axis, but I don't see why I couldn't change my end g-code to do that for me.  I'll give that some more thought.  Thanks!

thinkyhead, 2016-11-06:
> I don't see why I couldn't change my end g-code to do that for me

True. I use my end GCode to move the bed all the way forward, which on my i3 is `G0 Y190`.

chuck787, 2016-11-06:
Swapping sensor and hot end doesn’t work.  Tried it and remembered that the reason put the sensor on right of hot end was the sensor is always over the bed and correctly detected Z home and stopped.  With Sensor on left nothing but air.  Cannot put extension on bed for sensor because it would interfere with frame.  Imho the best solution is to correct the formulas to correctly handle offset like it did in prior versions as I may not be the only one now or in the future..

thinkyhead, 2016-11-07:
> Imho the best solution

How about `Z_SAFE_HOMING`?

chuck787, 2016-11-12:
Tried that last week with unsatisfactory results.  I am in process of trying latest RCBugFix with sensor on right as it corrected a display problem.  In my case it appears the SanityCheck is more problem then working code.

thinkyhead, 2016-11-16:
@chuck787 Please describe the unsatisfactoriness of `Z_SAFE_HOMING` so that we may improve it.

closed 2017-10-31 by thinkyhead