MarlinFirmware/Marlin #5160: Marlin Calibrate Scara offset negative error

multisystem, 2016-11-05:
Hello,

I can not run a calibration with Marlin-1.1.0 RC7 (1.0.xxx OK) and Scara kinematic
I made a G28
then M360 to move Theta 0 ° position
a look back M114 (SCARA Cal - Theta: -17.82)
a M206 X-17.82
if the value is negative, the offset if i remade a M360 is not considered?
If the offset is a positive example M206 X17.82 it works
Do you have any idea why it is not possible to put a negative offset?

Sorry for my broken English (i'm French) google translate use ...
Regards
François

thinkyhead, 2016-11-05:
RC7 doesn't have the best SCARA support. I've done a lot of patching of SCARA over the last few months, as I've been helping to develop a SCARA machine with MakerArm. So I recommend you test `RCBugFix` and start there. Then we can work on it further if there are still configuration issues.

thinkyhead, 2016-11-05:
I just checked more closely, and, indeed, I did patch `M360`, etc. so they add the `home_offset` (as set by `M206`) onto the destination. However, `M206` doesn't apply to the SCARA angles (A and B). Instead it applies to the cartesian space. This makes `M206` more consistent, so it works in the same manner on all types of machines.

So, the question now is, how are we supposed to configure the "home position" for a SCARA?

I believe the key is to figure out where on the XY plane the nozzle ends up when `G28` is done, then set this position as your `MANUAL_X_HOME_POS` and `MANUAL_Y_HOME_POS`.

multisystem, 2016-11-05:
Hello,
Thank you for your reply :)

Currently I am working on a Scara robot fablab Montpellier LABSud.
for now it's just a test box quickly built.
http://forum.labsud.org/viewtopic.php?f=51&t=248
https://www.youtube.com/watch?v=H5ieboKMeLQ&feature=youtu.be

if it works well I will do later versions 2,
a small robot to do the writing, and a 3D printer scara single-arm open-source hardware will
ongoing reflection for future version: https://3dwarehouse.sketchup.com/model.html?id=cc442b8e-7abc-49f7-93c9-3c832633214f

RCBugFix I test, I made you a test back later.

Sorry for my English ... Brooken
Regards
François

multisystem, 2016-11-05:
Hello,
I just tested with Marlin-RC BugFix, I had nothing happens ...
move the engines do not work
G28 key just 1 time 1 arm and the end position and stops and everything is blocked ...

SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing

move does nothing and returns:
SCARA Theta:-126322568.00   Psi+Theta:-126322568.00

the test configuration:

`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**
 *
 *  ***********************************
 *  **  ATTENTION TO ALL DEVELOPERS  **
 *  ***********************************
 *
 * You must increment this version number for every significant change such as,
 * but not limited to: ADD, DELETE RENAME OR REPURPOSE any directive/option.
 *
 * Note: Update also Version.h !
 */
#define CONFIGURATION_H_VERSION 010100

[... 1345 lines cut ...]
 * 34 - RAMPS1.4 - uses Analog input 5 on the AUX2 connector
 * 81 - Printrboard - Uses Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 - Rambo  - uses Analog input 3
 * Note may require analog pins to be defined for different motherboards
 **********************************************************************/
// Uncomment below to enable
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00  //Enter the diameter (in mm) of the filament generally used (3.0 mm or 1.75 mm) - this is then used in the slicer software.  Used for sensor reading validation

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0   //The number of the extruder that has the filament sensor (0,1,2)
  #define MEASUREMENT_DELAY_CM        14   //measurement delay in cm.  This is the distance from filament sensor to middle of barrel

  #define MEASURED_UPPER_LIMIT         3.30  //upper limit factor used for sensor reading validation in mm
  #define MEASURED_LOWER_LIMIT         1.90  //lower limit factor for sensor reading validation in mm
  #define MAX_MEASUREMENT_DELAY       20     //delay buffer size in bytes (1 byte = 1cm)- limits maximum measurement delay allowable (must be larger than MEASUREMENT_DELAY_CM  and lower number saves RAM)

  #define DEFAULT_MEASURED_FILAMENT_DIA  DEFAULT_NOMINAL_FILAMENT_DIA  //set measured to nominal initially

  //When using an LCD, uncomment the line below to display the Filament sensor data on the last line instead of status.  Status will appear for 5 sec.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H`

thinkyhead, 2016-11-07:
Sorry about that. The code for `MAKERARM_SCARA` is not published yet, and I'm not free to release the code at this time, as I'm still under contract with MakerArm.

If your SCARA is similar, then I can offer guidance. At the moment SCARA is not very well supported in Marlin. The code I've implemented for MakerArm is much more complete and deals with more SCARA-related issues.

thinkyhead, 2016-11-07:
I wasn't able to tell from your video, but does your SCARA have end-stops? And how are they arranged?

multisystem, 2016-11-07:
Hello
Detail of my setup
http://www.openhardware.eu/tmp/Scara.png

Grean2007, 2016-11-21:
**multisystem**     Hello,
I just tested with Marlin-RC BugFix, I had nothing happens ...
move the engines do not work
G28 key just 1 time 1 arm and the end position and stops and everything is blocked ...
SENDING:G28
move does nothing and returns:
SCARA Theta:-126322568.00   
 Psi+Theta:-126322568.00

**I have the same! It is a pity that it is not running the new firmware!**

thinkyhead, 2016-11-26:
I wish I had a SCARA to test with. If I try to address this stuff I'm afraid I will just be shooting in the dark. Do you have anyone local to your area who could get into the code and figure out the issue?

Grean2007, 2016-11-26:
Marlin-RC6 Scara works good. Just does not work  #define REPRAP_DISCOUNT_SMART_CONTROLLER (not compiled).
 Marlin-RC7, BugFix: G28 Home all- XY trailer to go together and then Z. Then X and Y axes are not moving. Only works Z. 
SCARA Theta:-126322568.00
Psi+Theta:-126322568.00

multisystem, 2016-11-26:
Hi
@Grean2007
Yes it is a small bug
If you comment in the file ... / ultralcd.cpp on **all version 1.1.xxx,** after that compiles
```
    #if ENABLED(SCARA)
      //MENU_ITEM_EDIT(float74, MSG_XSCALE, &axis_scaling[X_AXIS], 0.5, 2);
      //MENU_ITEM_EDIT(float74, MSG_YSCALE, &axis_scaling[Y_AXIS], 0.5, 2);
    #endif
```

@thinkyhead
I have 2 types of scara robots for testing,
1 Scara Morgan type  http://www.openhardware.eu/tmp/Scara.png
1 scara 1 arm (kinematics invert) http://www.openhardware.eu/tmp/Scara/DSCN0637.JPG

I can help debugger, the hardest for me is English...

Grean2007, 2016-11-26:
Thank you! Working REPRAP_DISCOUNT_SMART_CONTROLLER  (Marlin-RC6 Scara)
My 3D Printers: 
SCARA
https://www.youtube.com/watch?v=V-ltmE66adw
https://www.youtube.com/watch?v=QEI8J-BrP04
H-Bot
https://www.youtube.com/watch?v=V5eOtMyqQfA
https://www.youtube.com/watch?v=jQFdguE94N4
Delta
https://www.youtube.com/watch?v=3RIY7qQ7zt4

**Help** run on SCARA Marlin-RC7, BugFix.
Help run on SCARA Marlin-RC7, BugFix.

thinkyhead, 2016-12-08:
I hope to work on SCARA problems soon. A lot of bugs and glitches to work on now that RC8 is getting public testing…

Grean2007, 2016-12-09:
Marlin-RC8. На Scara не работает.  То же самое  G28 Home all- оси X и Y идут к концевикам вместе и доходят до первого (X или Y). Затем идет Z. После этого оси X  и Y не двигаются. Работает только     ось Z. 
SCARA Theta: -13134457.00
Psi+Theta: -13134457.00
[image]

invent2main, 2016-12-09:
Hello,
I have been working on a SCARA project for several months and I can not solve the problems of 'homing'; The coordinates after the homing do not match those expected and parameterized in MANUAL_X / Y_HOME_POS
The new version of Marlin-RC seems to fix some problems, but homing does not start on X and Y (Z works)...
Thank you again for the outstanding work
Pat

http://invent2main.org/armoni.html
https://vimeo.com/183643279
http://forums.reprap.org/read.php?110,521712,page=3

boelle, 2019-02-18:
has SCARA not been removed from marlin? is so we could close this one

boelle, 2019-03-12:
@multisystem have you tried with the latest bugfix 2.0?

closed 2019-05-16 by thinkyhead