MarlinFirmware/Marlin #5161: Lin_Advance issues

apballard, 2016-11-05:
I've been trying to keep up with the PR's but I am struggling to monitor changes.

My last few downloads of RCBugFix this week have introduced something new with regards to lin_advance.

In the past I calibrated my k factor to 47.  Then I saw the formula in the config file and after using it I found I was spot on.

However this week my lin_advance k factor is sitting around 10.  Reverting to old RCBugFixes fixes the problem.  Leaving it at 47 at the moment results in gaps at end of long extrusions and the first few layers being horrible.

In addition, in the last few weeks of RCBugFix, the very first extrusion of a print results in a blob at the beginning, but after a brim.  So the brim is printed, then it starts with the inner perimeter and at that point, where it starts, it blobs.  The rest is 100%.  This does not happen with earlier RCBugFixes.

Let me know what additional information you need.  I'm always willing to test.  If you would like pictures let me know.

thinkyhead, 2016-11-05:
@Sebastianv650 Thoughts?

Sebastianv650, 2016-11-05:
I guess it's the rework of LIN_ADVANCE #5124 that changed the behaviour. Basicaly it's only a bugfix, but up to now I'm not 100% sure if this changes the K factor. On my printer, it seems K is still around 75, therefore I decided to leave the formula as it is.
@apballard if you ignore the formula and try to calibrate the K factor from scratch, are you able to find a working value that solves the blobs? Your description sounds as your old K of 47 is now way too high. If you feel your K should be in the range of, say, 9-10 but 9 is too low and 10 seems to be too high be aware that since this PR you can also enter fractions, 9.5 for example.

I will grab the latest RCBugFix and do some test prints to ensure there are no other changes that broke LIN_ADVANCE. If you want to give me your stl, I will also test this if there are some specialities that I might not get in my models.

Sebastianv650, 2016-11-05:
@apballard What kind of printer do you have: Cartesian, Delta, Core? Direct drive, extruder steps/mm?

Sebastianv650, 2016-11-05:
I checked the current RCBugFix code, works flawless on my printer.
To come back to the impact of the PR to the K factor, I guess after it applied K should be smaller. K=75 still works for me, but maybe 65-70 would be even better now. The difference is small, to be sure I would have to an extended tests with hughe magnifications. But there might be no valid comparison between "old" and "new" K as the behaviour before the patch was not clean.

apballard, 2016-11-05:
@Sebastianv650 Thanks for the feedback.  I have a E3d BigBox Pro V1, basically Cartesian, direct drive, doing about 324 steps/mm.

I'll pull the latest absolute RCBugFix and test it out and then just double check a RC7 BugFix from a few weeks ago and test.  At least we can do fractions, so will give that a shot.  I did start low and increment, but by about K20 I was experiencing gaps at end of lines and infill was not meeting the perimiters.

At first I thought it was my own modifications for my lcd screen, but I went back to "vanilla" RCBugFix and the problem was still there.

I'll run some tests tomorrow and let you know.  I'm then away for 5 days next week.

Sebastianv650, 2016-11-06:
OK, looking forward to your results. Do you use PLA?
Aditional to my documentation [here](https://www.marlinfw.org/docs/features/lin_advance.html) you may have also a look at the picture from the JKN documentation [here](http://www.sailfishfirmware.com/doc/tuning-jkn-advance.html). The example on the right with the gap has a too high K value.
So maybe your printer realy wants to have K values <20.

You could also print flat cubes with K=5, 10, 15 and 20, put them on a scanner up-side down do a scan with some as high as possible resolution. If you upload this image, it could also help a lot.

apballard, 2016-11-12:
Ok, i'm still testing some things.  I'm comparing the latest RCBugFix vs the copy I have from October 15th.

Using 25mm cubes to calibrate, it would seem my K factor jumps from 47 to about 6 between the two versions.  I've kept all possible settings the same.

apballard, 2016-11-12:
I'm just going to test K4.7 to see if its been changed by a factor of 10.

apballard, 2016-11-12:
I'm having some issues with FWRetract, but will open another issue for that.

apballard, 2016-11-12:
Its working now.  I can't explain it.  I was messing around with FWRetract settings and next thing K6 was too little.  Now its back to 47.

closed 2016-11-12 by apballard