MarlinFirmware/Marlin #5162: M605 S0 M605 S1 Problem in dual carriage mode

gsart84, 2016-11-05:
I got an problem.

Marlin adv is configured for auto park mode (S1)
if i move an axis after homing, the toolheads are moving incredible slow to x-45 y-280 z+10

if i want to move an axis i have to write: 
M605 S0
M605 S1
and then i can move.

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 280
#define Y_MAX_POS 280
#define Z_MAX_POS 280

#define DUAL_X_CARRIAGE
#if ENABLED(DUAL_X_CARRIAGE)
#define X2_MIN_POS 46     
  #define X2_MAX_POS 326    
  #define X2_HOME_DIR 1   
  #define X2_HOME_POS X2_MAX_POS 
 #define DEFAULT_DUAL_X_CARRIAGE_MODE 1

thinkyhead, 2016-11-05:
What version of Marlin are you using?

gsart84, 2016-11-05:
marlin rc7.

10minutes ago i installed rc7 bugfix release. same problem

thinkyhead, 2016-11-05:
Odd, because `M605 S0` and `M605 S1` don't really do anything. All they do is set:
```cpp
    active_extruder_parked = false;
    extruder_duplication_enabled = false;
    delayed_move_time = 0;
```
But these are already the defaults. Thus nothing is changed by `M605 S0` or `M605 S1`.

gsart84, 2016-11-05:
ok. i never have this error with slow moving axis to the endstops before dual x carriage.
with m605 s0 can move the printer withoud problems after homing. (example g1z1)
with m605 s1 first it moves slowly to all endstop, then it moves fast to x0 y0 z10 and then it does the g1z1

i dont know where the error is

thinkyhead, 2016-11-05:
Dig.

gsart84, 2016-11-05:
its now abouth a month im searching the internet about this error and i cant figure out

thinkyhead, 2016-11-05:
> i don't know where the error is

If I had a dual X printer to play with all this would have been solved ages ago. Since I don't, I have to run the code in my head. So give me some time to look into it. I may ask you to help by doing some testing. But I'm going to bed now. It will have to be later on.

Meanwhile, here's what `G28` is doing:
```cpp
        int tmp_extruder = active_extruder;
        active_extruder = !active_extruder;
        HOMEAXIS(X);
        inactive_extruder_x_pos = RAW_X_POSITION(current_position[X_AXIS]);
        active_extruder = tmp_extruder;
        HOMEAXIS(X);
        // reset state used by the different modes
        memcpy(raised_parked_position, current_position, sizeof(raised_parked_position));
        delayed_move_time = 0;
        active_extruder_parked = true;
```

thinkyhead, 2016-11-05:
I suspect that `active_extruder_parked = true` is involved.

gsart84, 2016-11-05:
ok, thank you. this error gives me headaches.
i downt know if this error can be caused by x and y offset.

in marlinmain, i find the value  active_extruder_parked = false

thinkyhead, 2016-11-05:
@gsart84 Let me see if I understand what the correct behavior should be in Mode 1 (`DXC_AUTO_PARK_MODE`)…

- On `G28` the carriages should home to opposite ends of the X axis.
- The second carriage (i.e., `T1`) should be considered parked.
- The first carriage should really be un-parked, since it's free to move.
- The `active_extruder` (and active carriage) should always be `T0`.
- The `current_position` should of course be the position of `T0`.

Anything else I'm forgetting? I'll post what's *actually* happening in `G28` next…

thinkyhead, 2016-11-06:
Actual current behavior of `G28` in `RCBugFix` (with `DXC_AUTO_PARK_MODE`)…

- Remember which is the currently selected tool.
- Do a "tool_change" to select tool 0 (with no auto-move)
  - If already `T0` then nothing is done.
  - If it was `T1`…
    - No `G28` before? Print "No move on toolchange"
    - If the `delayed_move_time` is set or T0 isn't parked then park T0
    - "apply Y & Z extruder offset" (usually zero for a Dual X setup)
    - Set the new `active_extruder` to 1
    - Set `raised_parked_position` to the `current_position`
    - Add `TOOLCHANGE_PARK_ZLIFT` to `current_position[Z_AXIS]`
    - (Make sure this doesn't exceed the Z max software endstop)
    - Set `active_extruder_parked` true and `delayed_move_time` to zero
    - Tell the planner the new position
- Set `extruder_duplication_enabled` to `false`
- Home Z to max? Or raise Z to prepare for XY homing
- Quick home XY, if `QUICK_HOME` is set
- Home Y if `HOME_Y_BEFORE_X` is set
- Do homing of X in dual carriage style (code I pasted above)
  - Home the inactive extruder to its end
  - Store the position of that carriage as `inactive_extruder_x_pos`
  - Home the active extruder to its end
  - Set `raised_parked_position` to the `current_position` (again)
  - Set `active_extruder_parked` true and `delayed_move_time` to zero
- Do homing of Y and Z if not already done ("safe homing" may apply)
- Tell the planner the new position
- Do a tool-change back to the previously-active extruder (no auto-move)
  - If already `T0` then nothing is done.
  - If it was `T1`…
    - If T0 isn't parked then park T0
    - "apply Y & Z extruder offset" (usually zero for a Dual X setup)
    - Set the new `active_extruder` to 1
    - Set `raised_parked_position` to the `current_position`
    - Add `TOOLCHANGE_PARK_ZLIFT` to `current_position[Z_AXIS]`
    - (Make sure this doesn't exceed the Z max software endstop)
    - Set `active_extruder_parked` true and `delayed_move_time` to zero
    - Tell the planner the new position

It's quite a bit of spaghetti. Not easy to follow.

thinkyhead, 2016-11-06:
So, here's something you can do to help diagnose this in more detail. First, enable `DEBUG_LEVELING_FEATURE` and re-flash. Before homing do `M111 S247` to enable all logging except "DRYRUN." Now when you do `G28` there should be some extra logging of the whole procedure. Post the log here for review.

If we can't figure it out from that log, I'll keep adding more logging messages until it becomes clear what's happening.

gsart84, 2016-11-06:
thank you! 
ok i i put the code m111 s247 in and then g28:

Connecting...
Zàqjk½KìFñ
6X;V·H.l‹p|]ˆxñJ„J€YJ„J€Xñ
€\F]{L$<ø~5
Zƒ&ƒøÃ&ƒø%€5
ÿ=þ=
':63Vl6N&^FñJ„{‚JM¤{‚(JJ‚Xñ
jo·f±J¤Zàqjk½KìFñ
6X;V·H.l‹p|]ˆxñJ„J€YJ„J€Xñ
€\F]{L$<ø~5
Zƒ&ƒøÃ&ƒø%€5
ÿ=þ=
':63Vl6N&^FñJ„{‚JM¤{‚(JJ‚Xñ
jo·f±J¤Zàqjk½KìFñ
6X;V·H.l‹p|]ˆxñJ„J€YJ„J€Xñ
€\F]{L$<ø~5
Zƒ&ƒøÃ&ƒø
start
Printer is now online.
echo: External Reset
Marlin 1.1.0-RCBugFix
echo: Last Updated: 2016-07-26 12:00 | Author: (none, default config)
Compiled: Nov  6 2016
echo: Free Memory: 2844  PlannerBufferBytes: 1264
Using tool 0.
echo:V27 stored settings retrieved (410 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y100.00 Z4000.00 E102.00
echo:Maximum feedrates (mm/s):
echo:  M203 X150.00 Y150.00 Z2.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X700 Y700 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P4500.00 R3000.00 T2000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X5.00 Y5.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X-46.00 Y-14.00 Z0.00
echo:Hotend offsets (mm)
echo:  M218 T1 X1496 Y1504 Z1512
echo:Material heatup parameters:
echo:  M145 S0 H200 B50 F0
  M145 S1 H220 B80 F0
echo:PID settings:
echo:  M301 P38.74 I6.16 D60.93
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 T1 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.06
echo:SD card ok
echo:Active Extruder: 0
echo:SD card ok
>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
>>> g28
SENDING:G28
echo:G28
>>> gcode_G28
[... 131 lines cut ...]
 soft_endstop_max = 280.00
*** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
> zprobe_zoffset = -1.06
> home_offset[Z] = 0.00
  current_position=(159.00, 116.00, 1.06) :
<<< set_axis_is_at_home(Z)
  current_position=(159.00, 116.00, 1.06) : sync_plan_position
  current_position=(159.00, 116.00, 1.06) : > AFTER set_axis_is_at_home
  current_position=(159.00, 116.00, 1.06) : set_probe_deployed
deploy: 0
do_probe_raise(10.00)
>>> do_blocking_move_to(159.00, 116.00, 11.06)
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_blocking_move_to
>>> do_blocking_move_to(159.00, 116.00, 11.06)
<<< do_blocking_move_to
<<< homeaxis(Z)
<<< Z_SAFE_HOMING
  current_position=(159.00, 116.00, 11.06) : > (home_all_axis || homeZ) > final
  current_position=(159.00, 116.00, 11.06) : sync_plan_position
  current_position=(159.00, 116.00, 11.06) : clean_up_after_endstop_or_probe_move
<<< gcode_G28
echo:Active Extruder: 0

gsart84, 2016-11-06:
after this, when i type g1 z1, no debug code happens, only m105 and it moves all axis to xo yo z10

gsart84, 2016-11-06:
here i have made a video for you : [https://youtu.be/Y8UuKQfCeCE](url)
and in my github you can find my uploaded rc bugfix

thinkyhead, 2016-11-07:
I'm slowly scanning through the log to see what's going on. I'll let you know if I reach any conclusions, or if I need to add more logging.

One thing I note (unrelated to the issue) is that you are using a `home_offset[X_AXIS]` of -46. In fact, with the latest Marlin, you should instead set your actual `X_MIN_POS` and `X_HOME_POS` to -46. The `home_offset` is meant to be used more as an on-the-fly adjustment rather than as a permanent part of your setup.

thinkyhead, 2016-11-07:
So… I'm certain now that this issue is caused by the Dual X carriage "`active_extruder_parked`" flag being left on after Z homing. With `Z_SAFE_HOMING` the carriage should obviously not be considered parked. The super-slow move is because Marlin "thinks" it's doing a pure Z move to unpark the carriage.

Anyway, I'm patching now, and I'll let you know when it's merged.

gsart84, 2016-11-07:
ooooh you find the error??? you my hero :-)
let me know if i can test anything to night!!

Thanks!!! :-)

thinkyhead, 2016-11-07:
I hope so! Try this branch and all should be better: https://github.com/thinkyhead/Marlin/tree/rc_dualx_homing

Before you test, be sure to alter your configured `X_HOME_POS` and `X_MIN_POS` – setting them to -46, then do `M206 X0` followed by `M500`. This is more important with this update because both carriages may now be affected by the `home_offset` and any shifting caused by `G92`.

gsart84, 2016-11-07:
ok, sorry for my bad english skills, understand it right?
i have to do from:
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 280
#define Y_MAX_POS 280
#define Z_MAX_POS 280

to:
#define X_MIN_POS -46
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 280
#define Y_MAX_POS 280
#define Z_MAX_POS 280

right?

and set the offset to x0 with m206...

thinkyhead, 2016-11-07:
Yes, correct settings. Then `M206 X0` and `M500` after flashing.

gsart84, 2016-11-07:
ok thank you, the main problem seems to be away! you my heroe!!!! :-)

now i got the problem that the x2 crashes in to the endstop while moving to x140

gsart84, 2016-11-07:
ok, thats strange. i do g28, all works fine. i can move t0 to table middle (y140 x140)
now when i do toolcange to t1, it crashes to de xmax endstop.
ant then it moves the z axis upside withoud stoping

gsart84, 2016-11-07:
here is the new errorlog.
i do:
g28 
move t0 to y140 x140
tool change

`Connecting...
à1jk½KìFñ*6X;V·H.,‹p|]ˆxñJ„J€YJ„J€X±
€|F]{L$|ø~5
ñ
MmaAL'
„%€(5
ÿ5þ5
':63Vl6N&^FñJ„{‚JM¤{‚(JJ‚xñJJo6—FñJÄZà1jk½KìFñ*6X;V·H.,‹p|]ˆxñJ„J€YJ„J€X±
€|F]{L$|ø~5
ñ
MmaAL'
„%€(5
ÿ5þ5
':63Vl6N&^FñJ„{‚JM¤{‚(JJ‚xñJJo6—FñJÄZà1jk½KìFñ*6X;V·H.,‹p|]ˆxñJ„J€YJ„J€X±
€|F]{L$|ø~5
start
Printer is now online.
echo: External Reset
Marlin 1.1.0-RCBugFix
echo: Last Updated: 2016-07-26 12:00 | Author: (none, default config)
Compiled: Nov  7 2016
echo: Free Memory: 4065  PlannerBufferBytes: 1264
echo:V27 stored settings retrieved (410 bytes)
echo:Steps per unit:
echo:  M92 X80.00 Y100.00 Z4000.00 E102.00
echo:Maximum feedrates (mm/s):
echo:  M203 X150.00 Y150.00 Z2.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X700 Y700 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
Using tool 0.
echo:  M204 P4500.00 R3000.00 T2000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X5.00 Y5.00 Z0.40 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y-14.00 Z0.00
echo:Hotend offsets (mm)
echo:  M218 T1 X1466 Y1474 Z1482
echo:Material heatup parameters:
echo:  M145 S0 H200 B50 F0
  M145 S1 H220 B80 F0
echo:PID settings:
echo:  M301 P38.74 I6.16 D60.93
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 T1 D3.00
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.06
echo:Active Extruder: 0
>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
echo:M105
>>> g28
SENDING:G28
echo:G28
>>> gcode_G28
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X:35 Y:10 Z:-1.06 (Right-Back & Below Nozzle)
Auto Bed Leveling: 3POINT
[... 163 lines cut ...]
echo:M105
echo:M105
Using tool 1.
echo:T1
>>> gcode_T(1)
  current_position=(140.00, 140.00, 11.06) : BEFORE
Dual X Carriage Mode DXC_AUTO_PARK_MODE
Raise to 11.26
MoveX to -46.00
Lower to 11.06
  current_position=(-46.00, 140.00, 11.46) : New Extruder
Active extruder parked: yes
  current_position=(-46.00, 140.00, 11.46) : New extruder (parked)
  current_position=(-46.00, 140.00, 11.46) : Sync After Toolchange
  current_position=(-46.00, 140.00, 11.46) : sync_plan_position
  destination=(140.00, 140.00, 11.06) : Move back
echo:Active Extruder: 1
  current_position=(140.00, 140.00, 11.06) : AFTER
<<< gcode_T
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
Disconnected.`

thinkyhead, 2016-11-07:
Still working out the tool-change behavior. Please join the topic #4694.

closed 2016-11-07 by thinkyhead