MarlinFirmware/Marlin #5166: Enhancement: Show Turning direction for knob on LCD when moving axis

berryelectronics, 2016-11-05:
I think it would be really great to have a text that Shows like a small turn direction know to the left and right and a +/- next to them to know in which direction the axis will go positive or negative. same could be implemented to the temperature Change.

With the Minimum LCD being 2x16, it could easily fit on the second row while the first one is reserved for the current value.

thinkyhead, 2016-11-06:
Where?

Roxy-3D, 2016-11-06:
I would presume it should be on all screens.   (And obviously...  Sensitive to what type of screen it is.   Menu screens can be reversed just as value screens can be.  So to my way of thinking, the +'s and -'s might flip depending on what kind of screen is up and running.)

thinkyhead, 2016-11-06:
> I would presume it should be on all screens.

Don't forget: When you PRESUME you make a PRES out of U and ME.

berryelectronics, 2016-11-06:
@thinkyhead i initially thought about the value edit screens.
For LCDs with Buttons, basic arrows like the ones in the HD44780 Charset can be used to indicate which directional button increases/decreases the value, while on knob-based LCDs there has to be like a left/right turn arrow to indicate the increase/decrease. i couldnt find such a char in the aformentioned charset, so it would be a custom character.

thinkyhead, 2016-11-07:
@berryelectronics Isn't it the case that with arrow buttons both UP and RIGHT increase the value, while DOWN and LEFT decrease it?

berryelectronics, 2016-11-07:
@thinkyhead could also be, its long ago since i used a button-based lcd. 

But yeah, you're right. Just bothered me a bit when i switched from RC3->RC7 where suddenly the direction of my know changed and i accidentally crashed the hotend into the bed a few times

closed 2016-11-07 by berryelectronics