MarlinFirmware/Marlin #5170: Stuttering

chinzw, 2016-11-06:
Ever since i updated from 1.1.0-RC3 my printer stutters all the time, and i have no idea whats going on. Its making my prints come out looking really bad.

Here's a video of whats happening:
[https://www.youtube.com/watch?v=r6KwHx0x_AQ](https://www.youtube.com/watch?v=r6KwHx0x_AQ)

You can see the stuttering after it homes and tries to go back to the origin.

Sebastianv650, 2016-11-06:
For me that doesn't look like classic stuttering (which I define as sudden stop/start). Is there a mesh bed leveling active? It looks like it's breaking with normal acceleration values on junction points between lines..

chinzw, 2016-11-06:
UBL is active. Thats one line of code. For example, if i do G1 X0Y0Z0 and then G1X100Y100Z0 i get the same "stuttering". This doesnt happen if i roll back to 1.1.0-RC3

Sebastianv650, 2016-11-06:
> UBL is active. Thats one line of code. For example, if i do G1 X0Y0Z0 and then G1X100Y100Z0 i get the same "stuttering".

One line of gcode will be splitted at the boundaries of a mesh region into "sub sections". I'm not realy in the UBL thing, but as far as I know they changed a lot since RC3 so it's quite possible it's now diferent. The only strange thing is, why it's breaking that much. There should be only a very small amount of jerk in this case, so the junction speed should be nearly crusing speed. Let's wait for the UBL profesionals ;-)

chinzw, 2016-11-06:
I just tried with UBL disabled and still the same issue. Even something as simple as G0 X0 or G1 X0 does 2 or 3 "motions"

Roxy-3D, 2016-11-06:
What is the size of the Mesh?    There is always a slight pause when the printer shifts from being totally idle to making a long move.   The problem is it has to break that move up into smaller moves and that first move has nothing in the Planner's queue to connect to.    But while the first move is being performed, more moves (further splits of the original, longer line) are being generated and those will be smooth.

You should only see that pause if the Planner's buffer queue runs dry.   And then only if a very long move is being feed to the Planner.

chinzw, 2016-11-06:
> What is the size of the Mesh? There is always a slight pause when the printer shifts from being totally idle to making a long move. The problem is it has to break that move up into smaller moves and [...]

Well, even running this makes it stutter:
G1 X0
G1 X100

Not only once, it does it a number of times like in my video. This does not happen with RC3

Roxy-3D, 2016-11-06:
What kind of LCD Display do you have?   I'm guessing it is a Graphics one?   Those take extra processing power.    Right now...  There are some extra calculations in the Mesh_Buffer_Line.cpp file that do not need to be there any more.

If you pull up Mesh_Buffer_Line.cpp, you can edit out the 5 occurrences of this code block.   This code block is calculating the old way of doing the Z-Correction and comparing it to the results from the new and faster way of doing it.

You will get back some CPU cycles by deleting this.   But also, assuming you have a Graphics LCD Display, you should consider increasing your  #define LCD_UPDATE_INTERVAL 250 in  Ultralcd.h.   That will free up some CPU Cycles also.

```cpp
// debug code to use non-optimized get_z_correction() and to do a sanity check
// that the correct value is being passed to planner.buffer_line()
//
z_optimized = z0;
z0 = blm.get_z_correction( x_end, y_end);
if ( fabs(z_optimized - z0) > .01 || isnan(z0) || isnan(z_optimized) )  {
debug_current_and_destination("!FINAL_MOVE: z_correction()");
if ( isnan(z0) ) SERIAL_ECHO(" z0==NAN  ");
if ( isnan(z_optimized) ) SERIAL_ECHO(" z_optimized==NAN  ");
SERIAL_ECHOPAIRPGM("  x_end=",x_end);
SERIAL_ECHOPAIRPGM("  y_end=",y_end);
SERIAL_ECHOPAIRPGM("  z0=",z0);
SERIAL_ECHOPAIRPGM("  z_optimized=",z_optimized);
SERIAL_ECHOPAIRPGM("  err=",fabs(z_optimized-z0));
SERIAL_ECHO("\n");
}
```

chinzw, 2016-11-06:
> What kind of LCD Display do you have?
I have a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
I tried both those changes, still the same issue.

Roxy-3D, 2016-11-06:
> What is the size of the Mesh?

chinzw, 2016-11-06:
> What is the size of the Mesh?

MESH_NUM_X_POINTS  10
MESH_NUM_Y_POINTS  10
MESH_MIN_X         0
MESH_MIN_Y         0
MESH_MAX_X         190
MESH_MAX_Y         180
MESH_X_DIST        21.111110
MESH_Y_DIST        20.000000

Roxy-3D, 2016-11-06:
If this is happening during the print once the skirt is drawn, I think you better move back to RCBugFix.

chinzw, 2016-11-06:
It happens all the time indeed. I just disabled UNIFIED_BED_LEVELING_FEATURE (had to make some changes to the code since there's quite a few #if ENABLED(UNIFIED_BED_LEVELING_FEATURE) missing.
Once i got it fully disabled, the stuttering went away.

chinzw, 2016-11-06:
Also, i just rebuilt the mesh with 3x3 points and only get a "pause" when the nozzle crosses the center point.

Roxy-3D, 2016-11-06:
My new development machine has a Graphics LCD Panel.   Up until now, I've never had one.   It's taking me a little bit of time to get setup to to continue the code development on it.   I'll be able to see these issues first hand very soon and I should be able to get them resolved.   But it might take a couple of weeks to get resolution.

Roxy-3D, 2016-11-06:
Can you post your Configuration.h file for me?   And the Configuration_Adv.h file also?  

Thanks!

chinzw, 2016-11-06:
Here you go!
[confg.zip](https://github.com/MarlinFirmware/Marlin/files/574194/confg.zip)

closed 2017-10-31 by thinkyhead