MarlinFirmware/Marlin #5172: LCD Blinking X,Y,Z (axis letters)

AndKe, 2016-11-06:
Why is the LCD (44780 based) continuously blinking those three letters ? (1.1.0 , RC7)

AndKe, 2016-11-06:
Ok, I found it has something todo with homing https://github.com/MarlinFirmware/Marlin/pull/3782

closed 2016-11-06 by AndKe

thinkyhead, 2016-11-07:
If not homed they blink. And if the steppers power down (as they do after a period of inactivity) they lose their precise position and may be moved easily by hand. The labels blink to indicate that a new `G28` will be required for positional certitude.

AndKe, 2016-11-07:
May I suggest less annoying way ?  -, displaying  "X ?  Y ? Z ?"  instead of blinking all the time it's idle.

thinkyhead, 2016-11-07:
It was once like that, but it was requested by many to keep the coordinates on-screen.

AndKe, 2016-11-07:
yes, that's strange . why would we prefer to see position numbers - especially when notifying user that they are not reliable ?

thinkyhead, 2016-11-07:
I prefer it because I don't like to fly blind. I can't speak for others.