MarlinFirmware/Marlin #5173: Filament change....Did it work?

jaysonkelly, 2016-11-07:
With filament change enabled, there is a menu option available in the Tune menu while printing to change the filament. It works, but nothing happens when you click the menu item. Thinking that nothing happened I selected the option again and didn't know until i turned on the serial that it was Enqueueing the M600 command. I think it would be great if upon selecting the change filament command the menu returned to the status screen and a message indicating that filament change has been Enqueued shown on the screen. I imagine this is an easy fix...Thoughts?

Roxy-3D, 2016-11-07:
> I think it would be great if upon selecting the change filament command the menu returned to the status screen and a message indicating that filament change has been Enqueued shown on the screen. I imagine this is an easy fix...Thoughts?

My thinking is the LCD Screen should stay in a filament change control type screen until the filament change is completed.    That would make more sense to me.

jaysonkelly, 2016-11-07:
That would be great as long as it went to that screen as soon as the filament change menu were selected. As it is now, nothing happens.

thinkyhead, 2016-11-07:
> returned to the status screen and a message

That's a good point. Obviously we need to wait for the current move to complete, so it's not possible to go to filament change immediately, but a message could be shown right away. I'll make this patch soon.

thinkyhead, 2016-11-07:
It so happens that the first screen for filament change says:
```
Wait for start
of the filament
    change
```
…so the above patch jumps straight to that screen.

Barrocas, 2016-11-07:
Could it be shorter, like "Waiting for filament change"?

thinkyhead, 2016-11-07:
@Barrocas It is what it is, and has already been translated into all the supported languages.

closed 2016-11-07 by thinkyhead