MarlinFirmware/Marlin #5178: Melzi temp fail when sd enabled

billius92, 2016-11-07:
Hi , i am new here and this is going to be my first topic. 

I have a printer with Melzi board with atmel 1284p and recently flashed Marlin RC. I run through configure.h and made all required changes except enable sd support.I uploaded and everything run fine , stepper moves, bed and extruder thermistor readings etc in host mode.When i uncomment sdsupport and uploaded i faced this problem: in Repetier , when printer is just connected , i can see both temp reading but after a moment extruder reading goes to 0 C .SD can be mounted and works fine .If i reupload firmware with disableδ

 sdsupport , extruder reading works fine again.

My solution is to print direct from Repetier host but only for short time prints unfortunately.
[image]

I tried both latest version of Marlin RC and RCbugfix

English isn't my native language so i apologize for any pain in your eyes.

i attach video of bug and photo of board

billius92, 2016-11-07:
i can't upload zip with video

billius92, 2016-11-09:
this is the link to video...at 0.08 you can see that both tempartures have the correct readings and at 0.09 that extruder temp reading is 0 c. This is what happens when i enable sdsupport from marlin. So i can't really print in sd mode.

https://www.youtube.com/watch?v=GUWPxS9YB5g&feature=youtu.be

thinkyhead, 2016-11-15:
It sounds like potentially a pin conflict. Do you have a spare thermistor input? Try connecting the thermistor to a different input and change `TEMP_0_PIN` to the new input. If that helps then we should see where the conflict might be.

billius92, 2016-11-16:
Thanks for your reply
If you mean spare psysical input for thermistor on board , the answer is no.It has only two for thernistors.I found the TEMP_0_PIN that you said and switch pin number with TEMP_BED_PIN , the result was to have the problem with bed temp this time and extruder temp work fine when sd enabled, as soon as i switced the cables with bed thermistor on board too.So that means board's tracks are good , right?
So must be somewhere in the firmware?

thinkyhead, 2016-11-17:
I don't think the firmware is implicated here. Since the problem shifted to the bed when you switched pin numbers, the problem is clearly related to the SD detection interfering somehow with the thermistor input. I would look very closely at the traces on the board and use a multimeter to test for shorting.

billius92, 2016-11-17:
Hi, i just tested for short cicuit with a multimeter and i didn't find anything wrong.Particularly i tested pin of extruder temp on 1284p with all other pins ,and none of them made multimeter to beep.Only when i tested with extruder thermistor board connectore heard a beep , which is normal and also  means that trace isn't interrupted .The same with bed thermistor pin,I also tested pins od sd socket with thermistor pins and didn;t hear beep.What i noticed however is that sd mode may not work fine as i thought.When i mount sd , filenames aren't correct.For example a rover.gcode file is shown as ROVER-1.GCO.See in attached photo what i mean.How can i check that code in marlin is correct as far as sd is concerned?
[image]

Blue-Marlin, 2016-11-17:
8.3 filenames. That's normal.

billius92, 2016-11-17:
Sorry for being igonorant, but what is 8.3? some kind of version?

Blue-Marlin, 2016-11-17:
https://en.wikipedia.org/wiki/8.3_filename

billius92, 2016-11-17:
ok thanks for the info, i thought that wasn't normal becasuse it sowing the filename as in pc before....I have good news though , i made it work both temps and sd.The fix was to choose bobuino pinout instead of standard in arduino tools.I will make a printing test with sd tomorrow. I also found an alternative setting for sd which i am not sure what i am supposed to do with libraries(see photo), but i hope that i will not need this.

thanks for the help anyway

[image]

Blue-Marlin, 2016-11-17:
What did you use before bobuino? Just to have a warning.

thinkyhead, 2016-11-18:
Why does Bobuino differ from Arduino, I wonder…?

Blue-Marlin, 2016-11-18:
There is no official Arduino board using a 1284p processor. 
So ether he has used an other addition than 'bobduino' like 'sanguino' 
or he tried a random Arduino board - what can not work.

billius92, 2016-11-18:
When i first tried to upload marlin via arduino , i had to find first and install a proper board to communicate with my melzi. So after several trial and errors i finally made it work with MightyCore-master which contains several atmega processors like 1284, 644 , 164 , 8535...I have to mention that i had to change upload speed for 1284p from a text named boards.txt in program files.In arduino MightyCore-master has also other options like clock speed BOD etch, one of this options is to choose pinout (see photo) I suspected that something was going wrong with pin numbering of processor so i downloaded 1284p datashet, where i realise that there are two types of 1284p , one square (like mine) and one that has one dimension longest than the other and they have different pin numbering.So i found out that bobuino pinout is for my square.
[image]
[image]

thinkyhead, 2016-11-19:
Good to know. Has anyone posted a page about this topic on the RepRap wiki? That would be a good addition.

closed 2016-11-19 by thinkyhead