MarlinFirmware/Marlin #5181: New G30 code does not check if probe position is possible

manianac, 2016-11-08:
Commit dcfd7c69e8f6eaa286e6f0d6f924608c390f5f31 does not constrain the desired position.

My probe is pretty far off from my nozzle (-22,37) and if I ask it to do what should be an invalid position Marlin then disregards software endstops and leaves me with
`Send: G30 X20 Y20`
`Recv: Bed X: 20.00 Y: 20.00 Z: 0.98`
`Recv: X:42.00 Y:-17.00 Z:9.15 E:0.00 Count X: 3360 Y:-1360 Z:36600`

We could limit the position before `probe_pt` is called, or limit it inside there.

Roxy-3D, 2016-11-08:
G30 probably should not have been assigned to a single point probe function.   But be that as it may...   It needs a G28 before it can operate correctly.   Did you do a G28 before invoking G30?

manianac, 2016-11-09:
Sorry for the late response.  

Yes this is confirmed to happen right after G28.  It is still trying to move outside the build area.
```
[...]
Send: G28
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:0
[...]
Send: G30 X10 Y10
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: Bed X: 10.00 Y: 10.00 Z: -15.95
Recv: X:32.00 Y:-27.00 Z:9.15 E:0.00 Count X: 2560 Y:-2160 Z:36600
[...]
```
Issuing this on my setup causes it to try to move to -27 on Y and slam the endstop.

thinkyhead, 2016-11-15:
I have a function for general use, `position_is_reachable`. I should just add a call to that in `G30`, methinks.

closed 2016-11-15 by thinkyhead