MarlinFirmware/Marlin #5182: Servo0 still doesn't move

felice64, 2016-11-08:
Hi there,

- Rumba board brand new
- Using RCBugFix
- Wired correctly on Exp. 3, +5V(pin 2), -(pin4), PWM(pin5)
- Measured voltage on pins 2-4, it is exactly 5V
- Command M280 P0 S70, servo does not move
- Command M280 P0, reads the last given angle with pin 5 wire connected or disconnected
- Servo moves few degrees(10?) when I switch on/off the board.

Printed out pin assignment (full file attached):

PIN:  5        SERVO0_PIN                  Output = 0    TIMER3A    WGM: 0    TIMSK3: 2   **Can't be used as a PWM because of counter mode**

Help please!
[rUMBApINS.txt](https://github.com/MarlinFirmware/Marlin/files/577776/rUMBApINS.txt)

Blue-Marlin, 2016-11-08:
> PIN: 5 SERVO0_PIN Output = 0 TIMER3A WGM: 0 TIMSK3: 2 Can't be used as a PWM because of counter mode

That should not be the problem because our servo library does not use hardware PWM.

felice64, 2016-11-08:
I have no idea about what is going on with these pins. However, despite the fact that pin 5 is defined for Servo0, the servo works when physically connected to pin 6.

Fulg, 2016-11-08:
Looking at the [RUMBA schematic](http://reprap.org/mediawiki/images/2/24/RRD-RUMBA_SCHEMATICS.png), Arduino Pin 5 (PE3, PWM2) is connected to EXP3 Pin 6 (not 5)...

Roxy-3D, 2016-11-08:
I wonder if you have the right pin connected to the servo?  Is it possible the board is mis-labeled? 

It might be useful to add this code to Marlin_main.cpp and to connect an LED up to your supposed Pin-5.

At least then, we would have an independent confirmation that Pin-5 is responding to the micro-processor.

```cpp

/**
 * M43: Scan for a pin to help the user locate unused pins on their controller board
 *      Either an LED or volt meter can be placed on the desired pin.
 *      This command will scan the non-sensitive pins and put a voltage on the pin for
 *      a short duration.   
 *
 * S	Start Pin number.   If not given, will default to 0
 *
 * E	End Pin number.   If not given, will default to 127 
 *
 * N	No Sensitive Pin Checks.   Use with caution!!!!
 *
 * R    Repeat pulses on each pin this number of times before continueing to next pin
 *
 * W    Wait time (in miliseconds) between pulses.  If not given will default to 500
 *
 */

//
// sensitive_pin() is used by both M43 and M44 to avoid messing with pins that should not be touched.
// It is cleaner to have it as a function call than as in-line logic.
//
static bool sensitive_pin(int p) {
int i;
    for (uint8_t i = 0; i < COUNT(sensitive_pins); i++)  {
	if (p == sensitive_pins[i] || p==68 || p==69 || p==70 || p==71 || p==72 || p==73 || p==74  //causes KILL on my printer
												)  {
	  return true;
      }
    }
    return false;	
}

inline void gcode_M43() {
int p, j, s=0, n_flag=0, e=127, w=500, r=1; 

  if (code_seen('R')) 
    r = code_value_int();

  if (code_seen('S')) 
    s = code_value_int();

  if (code_seen('E')) 
    e = code_value_int();
  
  if (code_seen('N') )
    n_flag++;

  if (code_seen('W')) 
    w = code_value_int();

  for(p=s; p<=e; p++) {
      if ( n_flag==0 && sensitive_pin(p) ) {
        SERIAL_ECHOPAIRPGM("Sensitive Pin: ", p);
        SERIAL_ECHOPGM(" untouched.\n");
      } else {
    	SERIAL_ECHOPAIRPGM("Pulsing Pin: ", p);
        pinMode(p, OUTPUT);
        for(j=0; j<r; j++) {
           digitalWrite(p, 0);
           idle();
           delay(w);
           digitalWrite(p, 1);
           idle();
           delay(w);
           digitalWrite(p, 0);
           idle();
           delay(w);
        }
      }
    SERIAL_ECHOPGM("\n");
  } 
  SERIAL_ECHOPGM("Done\n");
}
```

felice64, 2016-11-08:
@Roxy-3D 
Pin 5 is defined for servo0 but then the PWM output is  physically on pin 6.
I have checked the output on pin5 with an oschilloscope but there is nothing there and the servo does not work.

Fulg, 2016-11-08:
@felice64 There is no relation between Arduino "pins" and the EXP3 connector pins. As I said above (you probably missed it), Arduino pin 5 is physically connected to EXP3 pin 6, you can see this in the [RUMBA schematic](http://reprap.org/mediawiki/images/2/24/RRD-RUMBA_SCHEMATICS.png).

felice64, 2016-11-08:
Got it. Quite confusing but finally I got it. Thanks.

closed 2016-11-08 by felice64

Fulg, 2016-11-08:
No worries. It is especially confusing because Arduino pin 5 is also physically on the [Mega2560 pin 5](http://www.arduino.org/learning/tutorials/hackings/atmega-2560-arduino-pin-mapping). Also, on the RUMBA  EXP3 pin 5 is also a SERVO pin, but the SECOND servo. :)

Roxy-3D, 2016-11-08:
> No worries. It is especially confusing because Arduino logical pin 5 is also physically exposed on the Mega2560 as pin 5. And, on the RUMBA, EXP3 pin 5 is also a SERVO pin, but the SECOND servo. :)

And that is the reason for that code pasted in up above.   When I need a pin to control something, it is confusing as can be trying to map the AVR's physical pin to a logical pin and then finding where it is wired on a connector.   And to make things worse, the pin numbers change depending upon whether you are using FastIO or the other library.

So for me...   I just connect up a LED to a free pin and let the code scan for it.   And then I know for sure what pin number to use and I'm not pulling my hair out trying to make things work.

thinkyhead, 2016-11-15:
> the pin numbers change depending upon whether you are using FastIO or the other library

Fortunately we've wrangled that now. Analog inputs are numbered 0-16. The rest of the pins are numbered according to their pin number on the AVR chip itself, and `fastio.h` maps the pin numbers to their busses and interrupts.