MarlinFirmware/Marlin #5183: Plotter mode

stigjoergensen, 2016-11-08:
Hi gang

I have this idea that I want to run by you to get your input.
Putting 2 pens on the printer and putting it into plotter mode…
The pens are lowered via servo. 
I think it should be relatively straight forward – when “extruding” lower the pen, otherwise keep the pen raised
The only thing needed as far as I can see is;
1)	Pen offset table – just like the dual head there is today
2)	A GCode to put the printer into plotter mode – mode is left when steppers are put into sleep
One could argue why the servo – I just think its faster to move a servo a few degrees rather than lowering/raising the entire bed.
What do you think? Doable? To me it seems like a relatively small change or am I missing something here?

Roxy-3D, 2016-11-08:
Yes...  This is all very straight forward and do-able.   You should be able to modify your branch of Marlin to do that very easily.     If it was me doing it...   I think I would change the idea slightly such that G1's E parameter controlled how much the pen held by the servo is lowered (or pressed) into the bed.  A bigger E number might mean the pen is pressed harder.   A very small E number might mean the pen is raised off of the bed (paper).

But you can certainly do it your way also.

stigjoergensen, 2016-11-08:
Thats a good idea - perhaps both ways can work with a #define plottter_penpressure to switch between the "just drop" and the "apply pressure" method.

Need to investigate how the E parameter is working now in more details.

Roxy-3D, 2016-11-08:
I can give you a high level view...    The G1 X Y Z E tells Marlin where those axis need to move to.    You would probably cut into the front end of the Planner code  (planner.buffer_line() ) and not generate positioning information for the E stepper motor.   Instead...  Down in the ISR  where new commands are pulled off the queue to control the stepper motors, you would translate the E movement specified into a servo position.

Of course it will be a little more difficult than that...   But in an hour or two, you should have it plotting things.

stigjoergensen, 2016-11-08:
Hi Roxy

Thanks for the hints, something to dig into here... very nice and extremely help full of you.

Thanks again..

stigjoergensen, 2016-11-14:
Thanks again Roxy

Ill close this now so it dosnt fill up in the issue list...

hope you woundnt mind if i contact you again, if i run into issue...

again - thanks for all your help so far.. really appreciate  it

closed 2016-11-14 by stigjoergensen