MarlinFirmware/Marlin #5185: Extruder -> Feeder

landodragon141, 2016-11-08:
So this isn't a huge deal but it could be valuable to re-identify "extruder" as "feeder" since this is much more indicative of its role. In my mind the "extruder" should be a combination of feeder and hot end. Obviously this goes against the current convention but hey look how many time they've redefined the name for filament printers. Maybe we could be trend setters haha!

adamfilip, 2016-11-08:
Extruder and hotend is my preference

esenapaj, 2016-11-08:
My image:

Extruder | Feeder | Hotend
--- | --- | ---
[image] | [image] | [image]

landodragon141, 2016-11-08:
That's exactly my thought.

adamfilip, 2016-11-08:
Extruder and Hotend are the common words used currently, checkout E3D website.. its Hotend and Extruder, not feeder

trying to change it to feeder just will end up confusing people..
there isnt even a need to call the entire assembly anything specific. unless you want to call it a gantry i suppose.

landodragon141, 2016-11-09:
I disagree. There are a number of people that have been thinking along the same lines already. Tom's3D is a notable example. Besides one of the biggest issues we are fighting in Marlin right now (from a user standpoint) is how hard it is to understand. To this end I have been working on the documentation side of Marlin to try and flesh it out and make it more easy for the layman to understand.

Roxy-3D, 2016-11-09:
@landodragon141 

> To this end I have been working on the documentation side of Marlin to try and flesh it out and make it more easy for the layman to understand.

Would you like to do a preliminary set of documentation for the UBL code?  I'll edit and expand what ever you can get in place.    Probably the most important thing to do initially is get good directions for generating an accurate Mesh.

landodragon141, 2016-11-09:
@Roxy-3D Sure I'll see what I can do. Is there anything started or are we working from a blank page?

thinkyhead, 2016-11-15:
> There are a number of people that have been thinking along the same lines already

For now I'd suggest including a note in the documentation that an extruder consists of one or more feeders plus a hotend. Sometimes a single hotend has multiple steppers, and sometimes a single stepper has more than one hotend. For legacy reasons we still have the `EXTRUDERS` setting in Marlin. That gets further broken down into `E_STEPPERS`, `HOTENDS`, and so on, depending on the other settings, such as `SINGLENOZZLE`, `MIXING_EXTRUDER`, etc. In future —for the Marlin 1.2.x branch— we'll revisit the verbiage.

closed 2018-02-26 by thinkyhead