MarlinFirmware/Marlin #5187: Auto bed level Z offset

alan707, 2016-11-08:
I am very happy with how easy it was to upgrade my MakerFarm Prusa i3V to add the Auto Bed Level servo using RCBugFix.  Command G28 works great, Command G29 works great, however there are a few issues I've encountered:

- I have set `define Z_PROBE_OFFSET_FROM_EXTRUDER -14.2` 14.2 mm is the measured distance difference from the depressed switch to the tip of the nozzle.
- if I run the printer with a test file it starts printing above and looks like the offset is not taken into consideration
- If I instead change the Z Offset from the LCD Menu > Control > Motion > Z Offset and then store with M500, I run into the problem that when doing bed leveling again, it does not lift off by 5mm as it did before to go from point to point.  It just drags the switch around the bed.
- When I was running a test print file I did not see the independent z-stepper motors moving to adapt to the bed level.  

Any help is greatly welcome!

My configuration file is here: [https://github.com/alan707/Marlin/blob/Makerfarm_Prusa_i3v_10in/Marlin/Configuration.h](https://github.com/alan707/Marlin/blob/Makerfarm_Prusa_i3v_10in/Marlin/Configuration.h)

Roxy-3D, 2016-11-08:
My recomendation would be to try setting the Z-Offset back to 0.000mm and then use the M851 command to adjust the Z_PROBE_OFFSET_FROM_EXTRUDER number.    And when you get it where you want it...  Do an M500.

alan707, 2016-11-08:
Thanks!  I tried this and it worked.  What about the last issue I encountered?  It doesn't seem like the z stepper motors are adjusting the height as the extruder travel across the bed.  Any way I can check that the auto bed leveling feature is in fact working?

Roxy-3D, 2016-11-09:
What type of Bed Leveling do you have active and configured?  Different things need to be done depending upon the answer.   But if it is Grid or 3-Point, give the  G29 command a V4 so you get all the output.   What you want to see is a Bed Level Correction matrix that is close to 'diagonal' but has some smaller components off the diagonal.    If it is Grid or 3-Point, post that matrix and we will have a better guess what is going on.   (and it wouldn't hurt to post the entire G29 output so we can see the probed points too.)

alan707, 2016-11-09:
It is a grid (9 points).  And here is the output of the entire G29 command: https://gist.github.com/alan707/41940ce496acb04931900cf62b19f3f6

alan707, 2016-11-09:
I had manual mesh bed leveling enabled before and that worked great by adjusing the Z-stepper motors independently.  I wonder if there is something I need to turn on with my configuration: [https://github.com/alan707/Marlin/blob/Makerfarm_Prusa_i3v_10in/Marlin/Configuration.h](https://github.com/alan707/Marlin/blob/Makerfarm_Prusa_i3v_10in/Marlin/Configuration.h)

alan707, 2016-11-11:
Should I try the latest RCBugFix to see if the z steppers adjust the gantry height as it moves across the bed?

Roxy-3D, 2016-11-11:
Yes.   Unless there is a good reason to be using another code base, RCBugFix is the most checked out and it is also where the most focus is.   It is easier to get help and easier to get changes made in that branch.

alan707, 2016-11-15:
I tried out the latest version of RCBugFix and updated the `Configuration.h` file. I am using Linear Bed Leveling (Auto) and the z stepper motors don't seem to be rotating at all as the gantry travels across the bed.  Is there anything that I have configured incorrectly?  

I've updated my Github Repo with the [latest config file](https://github.com/alan707/Marlin/blob/Makerfarm_Prusa_i3v_10in/Marlin/Configuration.h) I'm currently running.  Any thoughts are welcome!

Blue-Marlin, 2016-11-15:
If you are using Linear Bed Levelling the most likely reasons for not seeing z corrected are:
Your bed is level.
You deactivated levelling by homing (G28) after you levelled (G29).

alan707, 2016-11-15:
> You deactivated levelling by homing (G28) after you levelled (G29).
That was it!  Works great now, thanks!

closed 2016-11-15 by alan707