MarlinFirmware/Marlin #5189: Rebuilding my printer

boelle, 2016-11-09:
I finally came arround to gather bits for rebuilding my Mendel90

a few Q's:

1) I have gotten the full graphics display and looked at the schematics. The SD detect line is not connected to anything. I remember to have seen that the fix is just a Jumper wire. But is my mind just playing me a trick here?

2) I have allready build most of the extruder and X carriage. It will feature an servo so i can get bed "correction" going. 

It made me go hmm.... How hard would it be to use the touch probe to drive the 2 Z steppers up and down so X is parallel with the bed? I have gotten a Stepper expander that connects to AUX3 and allows 2 extra steppers on top of what the Ramps board allows. I got this since i will at some point use a Diamond hotend to get colour mixing going. 

I will have 1 spare stepper driver and well why not use that for automatic correction. 

Currently i use a dial gauge to level the X axis.

3) Another thought is if it would be possible to use the spare thermistor input to display either Cold end temp (to check if cold end fan does its job right) or to display electronics temp?

it will still be 2-3 weeks before i get there.

boelle, 2016-11-09:
@Roxy-3D What do you think about point 2? i ask you since I remember that you do most work on bed correction these days.

Roxy-3D, 2016-11-10:
> What do you think about point 2? i ask you since I remember that you do most work on bed correction these days.

There is a need to take what Prusa did.  It it really good.   It needs to be pulled into the main branches of Marlin.   That is what is being discussed here:  https://github.com/MarlinFirmware/Marlin/issues/5116

There are multiple problems.  One is doing his skew correction without special additions to the heat bed.   Another is we need one axis pair at a solid and stable 90 degrees.   This could be part of the answer to that bigger problem.  This could put the X axis at a perfect 90 degree angle to the Z-Axis.

boelle, 2016-11-10:
Hmm... X/Y Skew will be hard to do without any special bed. Prusa's has spots at known X/Y cords where the inductive probe does not trigger. So it will just search for these and go over them very slowly to get the exact center of them.

To get the X axis parallel to the bed would also require special setup... ie individual Z stepper drivers. Dont know if that and touch probe would be too special?

Anyways I plan to get it wired so... How "hard" would the programming part of it be?

Roxy-3D, 2016-11-10:
> How "hard" would the programming part of it be?

It wouldn't be too hard.   But the person doing the programming would need to have a machine wired (and configured) with the two independent Z-Motors.   Without that, there is no way to debug the code.

boelle, 2016-11-10:
bummer.... and i dont assume that you have the time to do some testing over hangouts etc if planned well ahead?

boelle, 2016-11-10:
for point 1 i guess i have to solder in a jumper wire from  pin 8 on AUX 3 to pin 2 also on AUX 

http://reprapproject.nl/images/20110826153710!Arduinomega1-4connectors.png

that says that Pin 2 on AUX 3 is D49 which is the pin number set in the pins file

Roxy-3D, 2016-11-10:
> bummer.... and i dont assume that you have the time to do some testing over hangouts etc if planned well ahead?

It is a very interesting idea.   But I'm trying to get the fancier Mesh Bed Leveling all tidied up.  And after I get that done, I need to re-merge (and debug) with RC-8.   It will be several months before I have any time to do anything else.

boelle, 2016-11-11:
No problemo, my dial gauge will have to make do until then

Do we have an expert for these full graphics displays? I want a confirm that the SD card detect should be connected/fixed as i figured. Just so i dont blow it up :-D

Blue-Marlin, 2016-11-12:
About `RepRapDiscount Full Graphic Smart Controller`, the adapter and SD-detect.

There is no problem! SD-detect is normally connected to D49 on RAMPS - regardless of the somewhat wrong/confusing schematics drawing.

On the display board sd-detect is connected to EXT2-7
On the adapter board EXT2-7 is named B7. B7 arrives at AUX3-8. AUX3-8 is in the same column as the not connected pin (AUX3-2) at the same connector - but at the other end.  5V is at the same end but in the other column.
Now find the not connected pin on RAMPS AUX3. Now realise the numbering seems to be different. On RAMPS the NC-pin is AUX3-8. The other end of the connector (same column) is __D49__ - and in the other column we find 5V.

Grab your multimeter and measure. 
Or much easier - compile in M43 and see if D49 changes when you plug in the sd-card.

The real error in the schematics of the adapter is only a typo. D49 is named D43

boelle, 2016-11-12:
oki time to get the meter out :-D

boelle, 2016-11-12:
@Roxy-3D 

just a bit ahead of time my config and pin file. Had to mod the pin file as i have an stepper expander board and fan extenders also

Pin file (ramps): http://pastebin.com/858uwaJ6
Config: http://pastebin.com/ACBD0JTe

will be sorting all my screws etc to get them organized and doing final shopping before tear down and rebuild during the week we enter.

boelle, 2016-11-12:
@Blue-Marlin "compile in M43"

how do i do that? i figure as you said that it would be much more easy and i dont have to connect anything

Blue-Marlin, 2016-11-12:
Search for M43 in Configuration_adv.h

boelle, 2016-11-12:
ahh

so //#define PINS_DEBUGGING changes to #define PINS_DEBUGGING

and i do M43 P49 to get the state of D49 ?

Blue-Marlin, 2016-11-12:
I'm tired to do your homework.
>  /**
   * M43: Pin report and debug
   *
   *      E<bool> Enable / disable background endstop monitoring
   *               - Machine continues to operate
   *               - Reports changes to endstops
   *               - Toggles LED when an endstop changes
   *
   *   or
   *
   *      P<pin>  Pin to read or watch. If omitted, read/watch all pins.
   *      W<bool> Watch pins -reporting changes- until reset, click, or M108.
   *      I<bool> Flag to ignore Marlin's pin protection.
   *
   */

boelle, 2016-11-12:
excuse me:

/**
 * Add M43 command for pins info and testing
 */
//#define PINS_DEBUGGING

that was all i had to go for

boelle, 2016-11-12:
but oh well sd detect works... pin state is 0 with a card

thinkyhead, 2016-11-16:
I recommend using a proper text editor that lists all the files in a given folder in a sidebar. Then you can globally search all the source code. You'll find all GCode commands documented in `Marlin_main.cpp`, most with pretty good detail. I've been trying to improve that. Also, we're making sure to post all new or changed codes on the RepRap wiki on the GCodes page.

boelle, 2016-11-17:
Got most things wired up.

My pins file: http://pastebin.com/cesmRTc7

I have put a few pins from configuration_adv in here so i have them all in one place. Does the pin file overrule configuration or will it be the other way round?

as for individual drivers for Z i read that defining Z2 will just use the next free extruder for that. Something inside me twists badly about the thought. Also what if all extruder steppers are used up as they will be when i upgrade to a mixing hotend? for now i have just copied in the step/dir etc pins and added - before the pin number to disable them

tomorrow will be spent adding connector to the last stepper and Z limit switch just so i can get to a working order

thinkyhead, 2016-11-18:
> Does the pin file overrule configuration or will it be the other way round?

The configurations will override pins, but only because they get processed first. The compiler will complain if you override a pin that isn't set up specifically to be overridden, but will allow it to compile.

By convention, the recommended pins for each function are predefined in the pins files, and those should be preferred unless there's compelling reason to use different ones.

boelle, 2016-11-18:
my reasons are that this fan extender are using the servo connector so pretty limited in choices of what pin numbers i can use.

The same for the stepper extender, its connecting to AUX2

What in the case of Individual Z steppers ? is there a way to force it to use a specific pin setup?

boelle, 2016-11-18:
anyways... got everything ready to do a test print

still going with the standard Mendel90 X carriage as the RC shops did not have all connectors needed, So auto bed correction is going to wait a few weeks more.

thinkyhead, 2016-11-19:
> What in the case of Individual Z steppers ? is there a way to force it to use a specific pin setup?

You can override the Z2 steppers in your `Configuration.h` file if you don't want to use the next empty E driver. You should also be able to override other stepper pins there, but the compiler will give warnings about the pins being redefined.

boelle, 2016-11-19:
thanks.. will have a look in to that later this weekend

so far the only real thing that is bugging me is that it does a reset on connect over serial.

mostly as i dont have a touch probe yet i will have to set Z height over bed each time. i know the Z endstop should do it but still.... 

are there a way i can avoid the reset on connect thing?

boelle, 2016-11-19:
looked in Configuration.h and Configuration_adv.h

i'm a noop because it did not find an option to do it. also looked for where it defines to use next empty e driver and also came up empty.

no rush on this one thou.. the reset on connect is more annoying

Blue-Marlin, 2016-11-19:
First do your homework (https://github.com/MarlinFirmware/Marlin/issues/5241#issuecomment-261717855) - than come back - not the other way around.

boelle, 2016-11-19:
figured the reset thing... just had to little fantasy that it was a hardware issue... 

http://forums.reprap.org/read.php?1,442013,442013#msg-442013

has the fix...  cut the track and bring them out to an external switch or a jumper header or whatever works

boelle, 2016-11-20:
just tried out manual mesh bed cirrection

Does this look total nuts or can it be improved? i used a 0.1 steel feeler blade instead of paper

SENDING:G29
State: On
Num X,Y: 7,7
Z search height: 4
Z offset: 0.00000
Measured points:
  3.80000  3.80000  3.80000  3.80000  3.80000  3.70000  3.70000
  3.80000  3.80000  3.80000  3.80000  3.70000  3.70000  3.70000
  3.80000  3.80000  3.80000  3.80000  3.80000  3.70000  3.70000
  3.80000  3.80000  3.80000  3.80000  3.80000  3.80000  3.80000
  3.90000  3.90000  3.90000  3.90000  3.80000  3.80000  3.80000
  3.90000  3.90000  3.90000  3.90000  3.90000  3.90000  3.80000
  4.00000  4.00000  4.00000  3.90000  3.90000  3.90000  3.80000

Blue-Marlin, 2016-11-20:
Looks like reasonable data for a relative flat and only slightly tilt bed.
With a reasonable nozzle diameter -> first layer height you should be able to print without  any auto bed levelling.
So just try to print.
If you want to test something, try to compare if your manually measured height differences are about the same with the other types of levelling and if you can print with all of them.

(Should keep us busy until Christmas)

boelle, 2016-11-20:
will do and i will get back sometime just before xmas

yes its slightly tilted... i guess its the X carriage that is not parallel with the bed

closed 2016-12-06 by boelle