MarlinFirmware/Marlin #5190: DOGLCD manual extruder control does not obey EXTRUDE_MINTEMP setting.

burakozhan, 2016-11-09:
It is possible to move extruder from graphic LCD menu although configuration has safeguards against cold hotend movement. These safeguards are obeyed by host software, but not by LCD menu.

I have a graphic LCD of type REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. 
On this LCD I can go to setting: Prepare -> Move axis -> Move 1mm -> Extruder. 
Here I can turn the wheel and forward/reverse as much as I like, altough the hotend is COLD.
This has potential to cause damage to hotend / bowden tube / extruder / filament / whatever.
I have the following setting in my config file :
#define EXTRUDE_MINTEMP 190
When I use software on my PC to command the extruder while the hotend is cold I get an appropriate error message. When I reduce the above setting to 10 (below room temp) I can move the extruder from the PC software.
Therefore this errorous behaviour is limited to command from LCD screen menu.

- The Marlin version: 1.0.1-RC7
- Machine model and manufacturer: Custom Huxley clone
- Electronics board: Ramps 1.4
- Machine components (Hotends, thermistors, thermocouples, relays, etc.): Cheapo chinese with 100k
- Host software (if any): pronterface
- Printing method: Host vs. Stand alone

Configuration.h available upon request.

thinkyhead, 2016-11-15:
It seems that `RCBugFix` is doing the following. In `ultralcd.cpp` we have…
```cpp
inline void manage_manual_move() {
  if (manual_move_axis != (int8_t)NO_AXIS && ELAPSED(millis(), manual_move_start_time) && !planner.is_full()) {
    planner.buffer_line_kinematic(current_position, MMM_TO_MMS(manual_feedrate_mm_m[manual_move_axis]), manual_move_e_index);
    manual_move_axis = (int8_t)NO_AXIS;
  }
}
```
The `buffer_line_kinematic` method calls `_buffer_line` which then does…
```cpp
#if ENABLED(PREVENT_COLD_EXTRUSION)
  if (de) {
    if (thermalManager.tooColdToExtrude(extruder)) {
      position[E_AXIS] = target[E_AXIS]; // Behave as if the move really took place, but ignore E part
      de = 0; // no difference
      SERIAL_ECHO_START;
      SERIAL_ECHOLNPGM(MSG_ERR_COLD_EXTRUDE_STOP);
    }
  }
#endif
```
So as far as I can tell we've already fixed the issue.

boelle, 2016-11-19:
@burakozhan 

Have you tested RCBugFix?

burakozhan, 2016-11-21:
Sorry did not have time till now to specifically test that, since I have been working on other projects. 
But at the time of writing the original post 1.1.0-RC7 was ahead of RCBugFix.

thinkyhead, 2016-11-26:
I probably had just patched the README on RC7. Look forward to your test result.

boelle, 2016-11-26:
@burakozhan had time to check?

boelle, 2016-12-01:
@burakozhan had time to retest?

boelle, 2016-12-11:
@jbrazio @thinkyhead maybe also close this since its inactive. I mean @burakozhan have not made a sound in 20 days

closed 2016-12-13 by thinkyhead