MarlinFirmware/Marlin #5195: Nozzle extruding at HOMING POSITION after reaching model reference temperature...

Shrikant888, 2016-11-10:
Hi all,

I wanted to give print at 205 degrees CUBE model. After nozzle reaching to 202 degree its extruding filament nearly 4mm because EXTRUDER STEPPER is rotating at 202 degrees. 

I wanted that after giving print its should reach to reference temperature and start to print with nearly 1mm extruding at HOMING position.

Please tell me where can I change in MARLIN VERSION 1.0....

Kaibob2, 2016-11-10:
I don't get it

Shrikant888, 2016-11-10:
[image]

Shrikant888, 2016-11-10:
I wanted that there should be 1mm extrusion at ARROW MARKED point. What shall I do?
Using MARLIN V1.0

Blue-Marlin, 2016-11-10:
Just tell Marlin what to do. Alter the start script in your host or slicer.

Kaibob2, 2016-11-10:
This could be a start script which primes the nozzle 15mm above the bed right before the print starts.
```
G28 		;home all axes
G91		;Absolute Mode
G1 Z10 F240		;lift Z 10mm
G90		;Relative Mode
T0		;Select First Extruder
G91		;Relative Mode
G1 E15.0000 F150	;Extrude 15mm to prime the nozzle
G90		;Absolute Mode
G4 P3000		;Wait 5sec o remove the filament blob or string from nozzle
M117 Druck gestartet...	;LCD Info
```

Shrikant888, 2016-11-11:
Thats good. 

Is there any ways to change this problem in MARLIN?

Kaibob2, 2016-11-11:
No

Shrikant888, 2016-11-11:
I am testing 2 printers.. One is default and another is modified with MARLIN but both I am using CURA as default software.

I have print at a time for two printers and found that MODIFIED board is extruding 2-3mm at HOMING position and for Default printer there is no extrusion at HOMING position. So, I have flashed with MARLIN. So there should be some thing which is coming from MARLIN. I guess not from CURA.

Please any one can find it.. It could be grateful for me sir.

Shrikant888, 2016-11-11:
How to solve this "z wobble"?

Kaibob2, 2016-11-11:
MARLIN only does what you tell it to do. As long as there is no GCODE to process there will be no extruder movements which cause this blob at the beginning. Take a close look at your GCODE which gets sent to your printer. 
MARLIN won't extrude a few milimeters at the beginning of each print just for fun and no reason!

Kaibob2, 2016-11-11:
This is no Marlin associated issue.
It will be better if discuss this at http://forums.reprap.org

ruggb, 2016-11-14:
maybe this is enabled
EXTRUDER_RUNOUT_PREVENT

thinkyhead, 2016-11-16:
This shirt is popular with people who are software engineers and web developers.
[image]
We are writing low-level code. We don't do customer support.

closed 2016-11-16 by thinkyhead