MarlinFirmware/Marlin #5196: Can somebody explain this below MARLIN VERSION 1.0 code...

Shrikant888, 2016-11-10:
#define FWRETRACT  //ONLY PARTIALLY TESTED
#ifdef FWRETRACT
  #define MIN_RETRACT 0.1                //minimum extruded mm to accept a automatic gcode retraction attempt
  #ifdef DIRECT_DRIVE_EXTRUDER
    #define RETRACT_LENGTH 2               //default retract length (positive mm)
    #define RETRACT_FEEDRATE 170            //default feedrate for retracting (mm/s)
    #define RETRACT_ZLIFT 0                //default retract Z-lift
    #define RETRACT_RECOVER_LENGTH .2      //default additional recover length (mm, added to retract length when recovering)
    #define RETRACT_RECOVER_FEEDRATE 8     //default feedrate for recovering from retraction (mm/s)
  #else
    #define RETRACT_LENGTH 1.2               //default retract length (positive mm)
    #define RETRACT_FEEDRATE 20            //default feedrate for retracting (mm/s)
    #define RETRACT_ZLIFT 0                //default retract Z-lift
    #define RETRACT_RECOVER_LENGTH 0       //default additional recover length (mm, added to retract length when recovering)
    #define RETRACT_RECOVER_FEEDRATE 2     //default feedrate for recovering from retraction (mm/s)
  #endif
#endif

Kaibob2, 2016-11-10:
This is the part of where you can enable Firmware Retract and set the parameters for it.
`#define FWRETRACT //ONLY PARTIALLY TESTED`
means Firmware Retract enabled
`//#define FWRETRACT //ONLY PARTIALLY TESTED`
means Firmware Retract disabled.

When you enable Firmware retract all retraction movements of the extruder will be handled by the firmware (Marlin) and you should disable Retraction in your Slicer.
Btw.: You already have an answer [here](http://forums.reprap.org/read.php?1,721924,721924#msg-721924)

Shrikant888, 2016-11-11:
[image]

Shrikant888, 2016-11-11:
[image]

Shrikant888, 2016-11-11:
what is 'z wobble'?

Kaibob2, 2016-11-11:
[Here](http://lmgtfy.com/?q=Z-wobble)

Kaibob2, 2016-11-11:
Change
#define FWRETRACT //ONLY PARTIALLY TESTED
to
//#define FWRETRACT //ONLY PARTIALLY TESTED
in `Configuration_adv.h`

This will disable the firmware retract in Marlin. You should use the CURA retract settings.

Shrikant888, 2016-11-11:
So above cutting on model is due to RETRACT or WOBBLE.

Now as I say RETRACT is ENABLED in CURA and MARLIN.

Will it be effecting this problem??
 
or

if WOBBLE then what parameter shall I change? Mechanical Z setting?

Kaibob2, 2016-11-11:
It is presumably due to retract. Do you even read what get's posted here?

Kaibob2, 2016-11-11:
Sorry, i'm out. This is no Marlin associated issue.
It will be better if discuss this at http://forums.reprap.org

landodragon141, 2016-11-11:
The retract in CURA is software retraction. The retract in Marlin is firmware retraction. If you are slicing with CURA disable firmware retraction in marlin. Z wobble is cause by bad z axis. It means that there is a mechanical issue. Google it you will find help.

landodragon141, 2016-11-11:
In regards to your print above you might be printing too fast, too high acceleration, too high temperature, too much flow. I would advise that you stop trying to print pretty things and focus on getting your printer properly calibrated. There are a number of great guides linked in the Marlin documentation which you can find at marlinfw.org.

Shrikant888, 2016-11-14:
Hi All!!!

I have this code RETRACTION...

HOW TO DISABLE IT? Like what line shall I comment to DISABLE RETRACTION?

#define FWRETRACT //ONLY PARTIALLY TESTED
#ifdef FWRETRACT
#define MIN_RETRACT 0.1 //minimum extruded mm to accept a automatic gcode retraction attempt
#ifdef DIRECT_DRIVE_EXTRUDER
#define RETRACT_LENGTH 2 //default retract length (positive mm)
#define RETRACT_FEEDRATE 170 //default feedrate for retracting (mm/s)
#define RETRACT_ZLIFT 0 //default retract Z-lift
#define RETRACT_RECOVER_LENGTH .2 //default additional recover length (mm, added to retract length when recovering)
#define RETRACT_RECOVER_FEEDRATE 8 //default feedrate for recovering from retraction (mm/s)
#else
#define RETRACT_LENGTH 1.2 //default retract length (positive mm)
#define RETRACT_FEEDRATE 20 //default feedrate for retracting (mm/s)
#define RETRACT_ZLIFT 0 //default retract Z-lift
#define RETRACT_RECOVER_LENGTH 0 //default additional recover length (mm, added to retract length when recovering)
#define RETRACT_RECOVER_FEEDRATE 2 //default feedrate for recovering from retraction (mm/s)
#endif
#endif

Please tell...

landodragon141, 2016-11-14:
Comment out this line
#define FWRETRACT -> //#define FWRETRACT

Shrikant888, 2016-11-15:
Ya ok sir

closed 2016-11-16 by thinkyhead