MarlinFirmware/Marlin #5197: Getting cuts in model

Shrikant888, 2016-11-10:
[image]

Blue-Marlin, 2016-11-10:
Does not look like a Marlin specific problem. Search for 'z wobble' and 'z banding'.

Kaibob2, 2016-11-10:
To me it looks like that there is not only a wobble or banding problem. Also consider tweaking temperature and retract settings in your slicer.
I would suggest calibrating your printer with much less complex models at first!

Shrikant888, 2016-11-11:
Thanks for reply..

Sowhat is ' z wobble' and 'z banding'  ?

Where to change in program MARLIN or CURA..?

Shrikant888, 2016-11-18:
[image]

emartinez167, 2016-11-18:
STOP! Do you even read the emails?

Go to the REPRAP forum!!!

Kaibob2, 2016-11-18:
@Shrikant888 Stop bugging around with these kind of issues! I also told you a few times. What's up with you?
THIS IS NO MARLIN RELATED ISSUE. CHECK YOUR MECHANICS!

Shrikant888, 2016-11-18:
Ya ya ok sir. I thought to get some help. I am beginner. 

Alright!!!

emartinez167, 2016-11-18:
Bugger off!!! Or I bet one of the admins will kick you out for good!

emartinez167, 2016-11-18:
And don't say I didn't warn you you would piss some people!!!

Kaibob2, 2016-11-18:
Only to make it absolutely clear:
You will always get help here, but never if you unmistakable convey everybody that you are too lazy to read, learn, try and make progress on your own. Maybe this hobby is just not the right one for you.
Like @thinkyhead stated "This platform is for improving Marlin and fixing issues and not for end user support"
Goodbye and good luck.

thinkyhead, 2016-11-19:
It doesn't mean we don't love you, @Shrikant888. It's just that there are better places to ask for help with these kinds of issues. As when you have a problem with your car, you don't phone up the designer of the car, you find a mechanic or someone who owns the same car.

Or course, if you do find some bug or error in Marlin, please don't hesitate to report it.

For assistance with these kinds of issues, I find the best resources are:
- The [RepRap Forums](http://forums.reprap.org) — Use the "search" function.
- The Facebook groups [RepRap](https://www.facebook.com/groups/5809198019/) and [RepStrap](https://www.facebook.com/groups/240122566171878/)
- And of course, [Google](https://google.com)! Look for articles on Calibration, Tuning, Etc.

closed 2016-11-19 by thinkyhead