MarlinFirmware/Marlin #5198: Development question:  which IDE

stigjoergensen, 2016-11-10:
Hi gang

Which IDE are you using?

im currently working with arduino standard IDE - seems good for small projects - but at the magnitude of Marlin - i think its a bit hard to work with - esp when the file count is higher than tabs abailable, its a hazzle to scroll in the small menu created... 

That the reason behind this question - there must be alternatives out there - im thinking about going to Visual Studio (which im all ready using) but perhaps there is even better options out there

Thanks

Blue-Marlin, 2016-11-10:
Most of use seem to use Arduino IDE from Arduino.cc (Not Adruino.org).
For just compiling and uploading the source, for most, it's fine to just click the arrow button.
But I suppose all serious developers use external editors (like Notepad++ on Windows).

Makefile users seem to be rare.

With PlatformIO you'll have about the same problem as with Arduino IDE if you don't like Atom.

Eclipse based IDEs seem to be much to complex for the average user - especially until things are set up.

Drakelive, 2016-11-10:
Hi 

I use this plugin for eclipse for many months: [Arduino Eclipse ](http://eclipse.baeyens.it)
I finally changed my life.

drk

Sebastianv650, 2016-11-10:
I'm using the Arduino IDE, because normaly the code changes I do are not that big (including LIN_ADVANCE). It's very comfortable to change a little bit, do an upload, see how the code behaves and continue coding. The long list of files to scroll down is anoing, that's true.

For studying new code sections of Marlin, I use Notepad++. I like that you can mark a variable and it will highlite all other sections where it is used. That makes it very easy to see the workflow. And Notepad++ is great for comparisons between versions etc.

Drakelive, 2016-11-10:
With Arduino Eclipse may integrate the Marlin project with git. This way you can receive updates and interact with the community.
Eclipse also has a real environment for debugging.

Blue-Marlin, 2016-11-10:
Ups.
Your linking to (http://eclipse.baeyens.it/) here, made them believe in  a denial of service attack! :-)
The Arduino IDE problem seems to be bigger then I supposed.

I have been on the site right after you sent the link. While reading it got slower and slower and braked together.

gcormier, 2016-11-11:
Visual Studio Community edition with the Arduino plugin

Drakelive, 2016-11-11:
Not work on linux machine  :-(

Drk

landodragon141, 2016-11-14:
When I edit the Documentation I actually do it online in Github lol

stigjoergensen, 2016-11-14:
seems we have most of the tools now - will have a look at them and see what fit me the best..

thanks for sharing...

closed 2016-11-14 by stigjoergensen