MarlinFirmware/Marlin #5201: [Question] Autotemp on M104

FHeilmann, 2016-11-10:
I stumbled across this today. So I've started using autotemp on my prints and I love the feature. While tweaking the min and max temperatures, I found myself wanting to adjust the autotemp values mid print.
This is tricky, since only M109 is currently able to set autotemp parameters. Since M109 is blocking, the current temperature must not be lower than what the autotemp will want to set after M109 is issued, or else the printer will pause, causing blobs.

Is there any specific reason autotemp can not also be set with M104? Otherwise I'd whip up an appropriate pull request or, if it is generally not desired, maintain the whole thing in my private repo.

thinkyhead, 2016-11-15:
I can see no reason why it shouldn't be included in `M104` too. I'll submit a patch soon.

FHeilmann, 2016-11-15:
I've got a pull request at the ready, I'll submit it

closed 2016-11-16 by thinkyhead