MarlinFirmware/Marlin #5202: Printer Killed() Usb Issue...nothing else in log

revmyduckie, 2016-11-11:
Hi,

i recently installed Marlin on my i3 Kit running ramps 1.4 (from BQ), 
unfortunately i am have been running into a usb issue with marlin, that
i have been unable to fix.

Marlin Version: 
 - RC6 currently 

Issue:
 Whenever a machine (raspberry pi, laptop) is connected via USB,
 the Printer randomly "crashes" within a short period of time showing
 the following error message:

 and  stops functioning until both the usb cable is disconnected from 
 the machine and the printer is power cycled.

 Printing from SD Card works flawlessly.

Last few lines from Log:
  Send: M105
  Recv: echo:SD card ok
  Recv: ok T:47.5 /0.0 B:60.2 /0.0 @:0 B@:0
  Send: N0 M110 N0*125
  Recv: ok
  Send: M85 S30
  Recv: ok
  Send: M105
  Recv: ok T:47.3 /0.0 B:59.9 /0.0 @:0 B@:0
  Send: M105
  Recv: ok T:46.5 /0.0 B:59.3 /0.0 @:0 B@:0
  Send: M105
  Recv: ok T:45.8 /0.0 B:58.6 /0.0 @:0 B@:0
  Recv: Error:Printer halted. kill() called!

#yes the values arre correct, the printer was still cooling down 
#from an sd card printjob when i tried the usb serial connection again.

Configuration Files:
 - configuration
   http://pastebin.com/fMe4jK2u

 - configuration_adv
    http://pastebin.com/HGmMEDiu

 - pins_ramps
    http://pastebin.com/8KhsLV1D

Attempted Solutions:
 - Disable Kill pin via pin configuration file for ramps board - error persists
 - Tried a different machine - error persists
 - Move power supply further away from printer
 - try a different Version of Marlin (RC6+RC7 tested)- error persists.
 - Disconnecting the Reprep Discount Controller (display and SD Card reader)

Diagnostic Notes:
 - Switching back to "Stock firmware" provided by bq, lets me print a test benchy 
   just fine. (Heated bed not configured)
- Diabling the Heat bed does not solve the issue.
- Printer works with marlin and current configuration when printing from SD Card
- Configuration Files based on Examples files.

Hardware Deviation from Factory/Stock Kit setup:
 - "MK3"  Heated bed kit (works in SD Card mode)
 - Marlin installed ;)

if any of you have an idea whats going on, i would be very grateful for any information that helps solve my problem. 

regards

Roxy-3D, 2016-11-11:
You should load up RCBugFix.    A lot of improvements have been added to it.

Grogyan, 2016-11-11:
Where is the rest of the log?

You have posted up where the temp drops about 1 DegC
Might be related to
[http://reprap.org/wiki/G-code#M85:_Set_inactivity_shutdown_timer](http://reprap.org/wiki/G-code#M85:_Set_inactivity_shutdown_timer)

revmyduckie, 2016-11-11:
Switched over over to RC-Bugfix,

Configuration
 - http://pastebin.com/G15F3mhh

configuration_adv
 - http://pastebin.com/v5xE9Vyk

pins_ramps (kill pin disabled again)
 - http://pastebin.com/4yKeU0nH

Full Logfile:
http://pastebin.com/Ra5TZTN3

Interesting Observation:
 - error only occurs when the serial communication is in use 
   (usb can be connected but serial inactive => works)
 - Two Menu Items are missing from lcd (pid tune, filament change) but are configured in configuration   
   (unsure if related)
- It happens regardless of weather the heaters are active or not.

thinkyhead, 2016-11-15:
Any message on the LCD screen when kill is called?

revmyduckie, 2016-11-16:
Problem Solved: Hardware Defect

I got a replacement ramps board, uploaded the same config+version+branch i had on the other one.

The issue solved itself.

regardsless of that, thank you for the support

closed 2016-11-16 by revmyduckie