MarlinFirmware/Marlin #5206: Incorrect menu entries when using INDIVIDUAL_AXIS_HOMING_MENU

Pipshag, 2016-11-11:
Using #define INDIVIDUAL_AXIS_HOMING_MENU makes the menu entry for Prepare not work as intended:

- When entering "Prepare" you get the expected menu entries - but at the bottom you get the main menu entries as well.

Tested on commit 06f2282. Haven't had time to revert and test older versions (sorry!).

Changes I've made to Marlin (other than pure configuration);  https://github.com/Pipshag/Marlin/commit/efeb8843c1f7dc91391e51ec0c8c6211a89bfab2  (Version.h), https://github.com/Pipshag/Marlin/commit/8f1cb55d0f5304a4aaa97b6bf6fa5c19f2e24db8 (ultralcd.cpp), https://github.com/Pipshag/Marlin/commit/8f1cb55d0f5304a4aaa97b6bf6fa5c19f2e24db8 (ultralcd.h), https://github.com/Pipshag/Marlin/commit/119eb3f205b21fa665ec95d30fffb14db0a14f68 (pins_RUMBA.h)

Configuration.h https://github.com/Pipshag/Marlin/blob/RCBugFix/Marlin/Configuration.h
Configuration_adv.h https://github.com/Pipshag/Marlin/blob/RCBugFix/Marlin/Configuration_adv.h

thinkyhead, 2016-11-15:
All I see is that when `INDIVIDUAL_AXIS_HOMING_MENU` is enabled, three items (`Home X`, `Home Y`, and `Home Z`) are added to the Prepare menu. These additional items should not affect the rest of the menu.

Pipshag, 2016-11-15:
Thanks for looking into this. I must sincerely apologize for taking up your time, figures I botched it and there is no fault at all with the code.

This issue can be closed.

closed 2016-11-16 by thinkyhead