MarlinFirmware/Marlin #5207: G29 V4 Switch Not Returning Verbose Data in Bugfix RC7 ( 26 Jul 2016)

Dreamloom, 2016-11-11:
I send "G29 V4" command have my printer auto level (bilinear) but it is just returning the grid point displacements and the x,y locations probed - not all the details the V4 switch returned in the RC 5 bugfix.
Just noticed this today when I had a little issue, and wanted a bit more data. Thanks for all you folks do!

boelle, 2016-11-11:
have you tried latest RCBugfix and not RC7 ?

Dreamloom, 2016-11-11:
Oops.. I thought RC7 was the latest release - Not real good navigating here yet ;)

Roxy-3D, 2016-11-11:
The branch RCBugFix is always the most 'corrected' branch.  

You can get it here:   https://github.com/MarlinFirmware/Marlin/tree/RCBugFix

Dreamloom, 2016-11-11:
Thanks! I'll give it a go and report back ASAP.

Blue-Marlin, 2016-11-11:
> (bilinear)

The bilinear levelling is, what used to be the DELTA levelling, now extended to other platforms.
There is no matrix to report and no difference from a corrected to the measured level.
There is not more to report.
If you want the same report use the same kind of levelling. (LINEAR)

Dreamloom, 2016-11-12:
With the latest version is Roxy-3D's link, V4 did not return additional data, using BiLinear method.  My bed is 700mm x 350mm, and is slightly "mounded" - having a high-spot near the center - would the Linear method work for this and provide the full V4 output I am seeking?

Dreamloom, 2016-11-12:
Ok, re-read the description and it seems pretty straight-forward that Bi-Linear is what I need. Are there any new switches to get the Deviation, etc?

Roxy-3D, 2016-11-12:
> Ok, re-read the description and it seems pretty straight-forward that Bi-Linear is what I need. Are there any new switches to get the Deviation, etc?

You probably need some flavor of 'Mesh Leveling'.   But given the size of your bed, you may want to consider bringing up the devel_ubl branch of the code found here:  https://github.com/MarlinFirmware/Marlin/tree/devel-ubl

The reasons are probably two fold.  First, you may need a higher resolution Mesh.  But you also will need the tools to edit, refine and store the Mesh.    Right now, those tools are only in the devel-ubl branch.

Dreamloom, 2016-11-13:
I will check it out, thanks!

closed 2016-11-13 by Dreamloom