MarlinFirmware/Marlin #5208: MarlinFirmware 1.1.0 RC7 missing servo endstop

kaliboe, 2016-11-12:
I am following a tutorial https://www.youtube.com/watch?v=6msLOR_EfKc where they explain to change some setting at the servo endstop. Now on version 1.1.0 RC7 I don't find this option anymore. please help

Blue-Marlin, 2016-11-12:
That video was published 11.11.20 __13__.
It is related to a 3 years old Marlin version .
Please allow us some progress and forget about that video.

boelle, 2016-11-12:
nad dont use RC7.. use latest RCBugfix

kaliboe, 2016-11-12:
of course I allow progress but the option servo endstop is still integrate in RC7, only the option to angle your servo is not in there anymore, that's strange.

Blue-Marlin, 2016-11-12:
Open your eyes.
>// Z Servo Probe, such as an endstop switch on a rotating arm.
//#define Z_ENDSTOP_SERVO_NR 0
//#define Z_SERVO_ANGLES {70,0} // Z Servo Deploy and Stow angles

kaliboe, 2016-11-12:
thanks for opening my Eye's Blue-Marlin

closed 2016-11-16 by thinkyhead