MarlinFirmware/Marlin #5210: Measure Z height with M428

HansGrub3r, 2016-11-12:
Hi!

(Printer is a Mini Kossel without heatbed, nozzle acts as Z-probe)

Last night I did what I've wanted for months, flashed the latest Marlin 1.1.0 onto my Arduino Mega2560 / Ramps 1.4.
Spent some hours configuring it to match my old Marlin the best I could since alot had changed.

When I thought that I had it all set up, flashed it onto the printer. Hooked it up with pronterface -
G28 to home it, jogged down to check my Z0. Then I found out that I couldn't move past 0 at this point I were approximate 20mm above the bed.
In my previous Marlin I know that I could change that in Config.h so that I could move past Z0, searched the whole Marlin but couldn't find it. I recon that IF I set my Z height higher than my printer is tall I would be able to reach the bed and then use M666 H(new Z height). But I then realized that M666 aint doing that kind of thing anymore..

Started to google my problem and found a thread on this GitHub that I would be able to use M428 where I wanted my Z0 to be located. 
I have now been at this m428 for about an hour and everytime I am positioned at the point where I want my Z0 and send m428 my printer gives me a beep and on the LCD it tells me 'Offsets applied'. 
Save it with M500, G28 to home it, G0 F3000 Z0, and it stops at Z0 - but that Z0 isn't at my bed, if I jog manually I am Z-17.6 below Z0 where my bed is located..

Been trying this countless of times now and every time I get a different lenght between my printers Z0 and the actual Z0 on my bed.. been trying to google myself a solution but found nothing so far, hence why I am now posting in here.

Kaibob2, 2016-11-12:
Check this line in configuration_adv.h
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration_adv.h#L239
and these lines in configuration.h
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Configuration.h#L690-L691

HansGrub3r, 2016-11-12:
Thanks for reply, but I'm not really sure what to do with that ?

Barrocas, 2016-11-12:
In configuration.h, change the 
#define min_software_endstops true

To
#define min_software_endstops false
And then you can move the nozzle lower than Z=0, it'll indicate negative values, so you can find your correct Z Max value.

Once you find your correct Z max height, insert it in the configuration.h and reflash.
Check if it's good.
If yes, you can re-apply the 'min software endstops' to 'true'.

HansGrub3r, 2016-11-13:
Thanks! Took me 4 minutes and now I got my Z0 at the right spot!

closed 2016-11-14 by HansGrub3r