MarlinFirmware/Marlin #5211: Question: Mesh Bed Leveling - automatic or no?

infernusdomoso, 2016-11-12:
The mesh bed leveling feature - I'm having a hard time finding a concise answer: Is it possible to do mesh leveling automatically like you can with the auto-leveling? Right now I'm using a touch sensor on the printhead to do auto-leveling (9 point auto probing), but my large bed is somewhat warped, and I'd like to use the mesh leveling - however it appears the mesh leveling is not automatic? Am I missing something? It doesn't make much sense that it can 9 point auto-probe for an auto-level, however not 9-point auto-probe for mesh? Or am I just missing something?

FHeilmann, 2016-11-12:
The difference here is that the auto leveling does not store anything hence why it must be repeated for each print. Mesh leveling has to be performed manually (unless you use the devel ubl branch) and then stores the result in eeprom. Afterwards before a print, the printer merely has to probe one mesh point to synchronize with the mesh.

Roxy-3D, 2016-11-12:
@infernusdomoso    Bring up the devel_ubl branch and give it a try.    Once you get it limping along, below is a cheat sheet of commands to get you up and running:

- G28
- G29 P1     (to automatically generate as much of the mesh as possible)
- G29 P3 R C.5    (to fill in the unprobable regions with a 'safe' value of +.5mm)
- G29 A    (to activate the UBL System)
- G29 S 1  (to save the Mesh in slot 1)
- G26 P C O 3.0    (to generate a Mesh Validation Pattern to find areas to further refine)
- G29 P4 R     (with nozzle placed over less than perfect areas of the mesh to edit it)
- G29 S1    (to resave the updated Mesh)

infernusdomoso, 2016-11-12:
Thanks for the pointers. Is the devel_ubl necessary to run those functions? The reason I ask is I've heavily modified my firmware (automating the heat sink fan based on nozzle temperature, z-probe sled operation, using 2nd extruder axis for a stepper to auto-adjust z-probe height etc) which would be a PITA to port over.

Hmm wonder how hard it would be to put devel_ubl's G28 and G29 commands into my modded 1.1.0RC4?

infernusdomoso, 2016-11-12:
Probably worth being in a separate question, but don't want to spam the issues with simple questions. Has it ever been looked into, for an uneven bed surface, for the firmware to generate a custom raft which takes the uneven bed into account, laying down a bed with more layers where necessary so that the final raft surface is completely flat? Just a thought that popped into my head just now. That way an uneven bed could have objects printed onto it evenly instead of the objects being slightly warped also.

Roxy-3D, 2016-11-12:
Right now somebody is trying to re-merge the UBL code with RC-8.   But that will take 6 weeks of time.  

You can't just move the G28 and G29 functions from the UBL code base.   It has hooks into everything from the EEPROM storage to the LCD Panel.       It would be much easier to move your stuff over to the UBL code.   But with that said, it would seem you could bring up the UBL System and see what you think and then decide if it will help you.

thinkyhead, 2016-11-16:
Hi @infernusdomoso. Yes, there is an automatic version of "mesh" bed leveling now in Marlin `RCBugFix` (soon to be released as `1.1.0-RC8`). The new option is named `AUTO_BED_LEVELING_BILINEAR`.

ProfessionalTrash, 2017-05-15:
I'm pretty new to the whole printing stuff (and github so sorry if this is posted in the wrong place), but I have a Tronxy P802MA with autoleveling. I cant figure out how to update the firmware on it. It is currently running Marlin 1.x? Other posts that I have found don't give a complete walkthrough of how to update firmware and what needs changed in the firmware before actually flashing it to the printer. Really the only thing that I need to change in the firmware is enable mesh bed leveling. It's currently set at 9-point autolevel (which is plane-based leveling I believe?)

Bob-the-Kuhn, 2017-05-15:
You are in the right place.  We're glad to help.

Please open a new issue.  Just go to the top of the page and click the green **New issue** button

----

When you open the new issue please include the log of the powerup/reset messages.  Just copy the log into a .TXT file and drop the file on your post.

Do you have access to the files **Configuration.h** and **Configuration_adv.h**?  If yes then ZIP them up and drop it onto your post.

We'll also want to know what controller you are using.  Is it the MELZI_MAKR3D?

ProfessionalTrash, 2017-09-07:
Sorry for taking so long to reply. Took a break from figuring out the firmware update. However now that I have moved to a PEI bed instead of blue tape, I have no other choice but to update the firmware in order to enable automatic mesh bed leveling. Unfortunately, since I have no way of downloading the firmware from my board, I can not supply you with the config files. I am also not sure how to get the powerup log files (printer is being run from a raspberry pi 3 running octoprint). The board is a melzi2.0_v5 on a Tronxy P802MA. Also what do I choose for creating a new issue? Only three options are available; feature request, bug report, or compile error. Thanks for the help!

Roxy-3D, 2017-09-07:
It seems like we should add a 'Technical Support' type of issue.   In the mean time you can do a feature request and the requested feature is to know how to do something.

closed 2017-11-27 by thinkyhead

jordathan, 2018-01-21:
my mesh does 25 points how do i make it less?

thinkyhead, 2018-01-22:
@jordathan 
```cpp
  #define GRID_MAX_POINTS_X 3
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
```

ypsonar, 2018-04-24:
Dear friends,

I have updated Marlin v1.1.8 , and activated AUTO_BED_LEVELING_BILINEAR with 3*3 point.

To use auto bad leveling feature , I started with G28 and then G29 then Print. I does not found correct bad leveling I am going fail to make perfect fast layer for large print.

Please help with very systematically in two parts.

One is all settings of firmware, and second is command sequence specifically for auto bad leveling very properly please.

There are lot of articles available on bad leveling , such info confuse us  so I request provide updated info for Marlin v1.1.8 only.

Please reply as early as possible.

Thanks

thinkyhead, 2018-04-24:
@ypsonar —

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube