MarlinFirmware/Marlin #5213: need help with autobedleveling

kaliboe, 2016-11-13:
hi, I am a rookie trying to learn and configure the autobedleveling on my printer.
I have started from scratch to configure the marlin firmware because I did not had any information from my printer. Now I connected a servo arm with micro-switch. When I send a G28 all works fine, the arm lowers and stops the extruder from lowering. Because I am a rookie I search the internet for sollutions, I follow this guy's tutorial from youtube https://www.youtube.com/watch?v=YpiOsetkIRg (I know it's 3years old) so I enter my measurements like the guy explant in the video. Now I cannot upload it to me printer, I receive this error messages

`Arduino: 1.6.8 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Marlin.ino.cpp:1:0:

Configuration.h:479: error: floating constant in preprocessor expression

 #define X_PROBE_OFFSET_FROM_EXTRUDER 25.00  // X offset: -left  +right  [of the nozzle]

                                      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:24: note: in definition of macro 'max'

 #define max(a,b) ((a)>(b)?(a):(b))

                        ^

sketch\Conditionals_post.h:158:51: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

   #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

                                                   ^

sketch\SanityCheck.h:403:37: note: in expansion of macro 'MIN_PROBE_X'

       #if LEFT_PROBE_BED_POSITION < MIN_PROBE_X

                                     ^

Configuration.h:479: error: floating constant in preprocessor expression

 #define X_PROBE_OFFSET_FROM_EXTRUDER 25.00  // X offset: -left  +right  [of the nozzle]

                                      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:32: note: in definition of macro 'max'

 #define max(a,b) ((a)>(b)?(a):(b))

                                ^

sketch\Conditionals_post.h:158:51: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

   #define MIN_PROBE_X (max(X_MIN_POS, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

                                                   ^

sketch\SanityCheck.h:403:37: note: in expansion of macro 'MIN_PROBE_X'

       #if LEFT_PROBE_BED_POSITION < MIN_PROBE_X

                                     ^

Configuration.h:479: error: floating constant in preprocessor expression

 #define X_PROBE_OFFSET_FROM_EXTRUDER 25.00  // X offset: -left  +right  [of the nozzle]

                                      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:24: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                        ^

sketch\Conditionals_post.h:159:51: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

   #define MAX_PROBE_X (min(X_MAX_POS, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

                                                   ^

sketch\SanityCheck.h:405:40: note: in expansion of macro 'MAX_PROBE_X'

       #elif RIGHT_PROBE_BED_POSITION > MAX_PROBE_X

                                        ^

Configuration.h:479: error: floating constant in preprocessor expression

 #define X_PROBE_OFFSET_FROM_EXTRUDER 25.00  // X offset: -left  +right  [of the nozzle]

                                      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:32: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                                ^

sketch\Conditionals_post.h:159:51: note: in expansion of macro 'X_PROBE_OFFSET_FROM_EXTRUDER'

   #define MAX_PROBE_X (min(X_MAX_POS, X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))

                                                   ^

sketch\SanityCheck.h:405:40: note: in expansion of macro 'MAX_PROBE_X'

       #elif RIGHT_PROBE_BED_POSITION > MAX_PROBE_X

                                        ^

Configuration.h:480: error: floating constant in preprocessor expression

 #define Y_PROBE_OFFSET_FROM_EXTRUDER 16.00  // Y offset: -front +behind [the nozzle]

                                      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:24: note: in definition of macro 'max'

 #define max(a,b) ((a)>(b)?(a):(b))

                        ^

sketch\Conditionals_post.h:160:51: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

   #define MIN_PROBE_Y (max(Y_MIN_POS, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))

                                                   ^

sketch\SanityCheck.h:407:40: note: in expansion of macro 'MIN_PROBE_Y'

       #elif FRONT_PROBE_BED_POSITION < MIN_PROBE_Y

                                        ^

Configuration.h:480: error: floating constant in preprocessor expression

 #define Y_PROBE_OFFSET_FROM_EXTRUDER 16.00  // Y offset: -front +behind [the nozzle]

                                      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:32: note: in definition of macro 'max'

 #define max(a,b) ((a)>(b)?(a):(b))

                                ^

sketch\Conditionals_post.h:160:51: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

   #define MIN_PROBE_Y (max(Y_MIN_POS, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))

                                                   ^

sketch\SanityCheck.h:407:40: note: in expansion of macro 'MIN_PROBE_Y'

       #elif FRONT_PROBE_BED_POSITION < MIN_PROBE_Y

                                        ^

Configuration.h:480: error: floating constant in preprocessor expression

 #define Y_PROBE_OFFSET_FROM_EXTRUDER 16.00  // Y offset: -front +behind [the nozzle]

                                      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:24: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                        ^

sketch\Conditionals_post.h:161:51: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

   #define MAX_PROBE_Y (min(Y_MAX_POS, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))

                                                   ^

sketch\SanityCheck.h:409:39: note: in expansion of macro 'MAX_PROBE_Y'

       #elif BACK_PROBE_BED_POSITION > MAX_PROBE_Y

                                       ^

Configuration.h:480: error: floating constant in preprocessor expression

 #define Y_PROBE_OFFSET_FROM_EXTRUDER 16.00  // Y offset: -front +behind [the nozzle]

                                      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:32: note: in definition of macro 'min'

 #define min(a,b) ((a)<(b)?(a):(b))

                                ^

sketch\Conditionals_post.h:161:51: note: in expansion of macro 'Y_PROBE_OFFSET_FROM_EXTRUDER'

   #define MAX_PROBE_Y (min(Y_MAX_POS, Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))

                                                   ^

sketch\SanityCheck.h:409:39: note: in expansion of macro 'MAX_PROBE_Y'

       #elif BACK_PROBE_BED_POSITION > MAX_PROBE_Y

                                       ^

exit status 1
floating constant in preprocessor expression

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
can somebody help me further with my problem?
here is the link to my configuration.h file [http://www.kalidesign.be/marlin/Configuration.h](url)

Blue-Marlin, 2016-11-13:
Our advice:
```
// Z Probe to nozzle (X,Y) offset, relative to (0, 0).
// X and Y offsets must be integers.
//
// In the following example the X and Y offsets are both positive:
// #define X_PROBE_OFFSET_FROM_EXTRUDER 10
// #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
```
And the error message:
```
error: floating constant in preprocessor expression
``` 
could not be more clear.

Use
```
X_PROBE_OFFSET_FROM_EXTRUDER 25
```
instead of
```
X_PROBE_OFFSET_FROM_EXTRUDER 25.00
```

kaliboe, 2016-11-13:
Blue-Marlin thanks for your help, I appreciate it
What do you mean with Our advice?

kaliboe, 2016-11-13:
I did a test and I can upload it to the printer :-)
Run the G29 command and it works except that the arm with the micro-switch is not raised between the different points on the bed. So it travels over the bed with the arm down and the micro-switch touching the bed. Normal it need to raise for save travelling to all the calibration points and lowers at the calibration point or am I wrong?

Blue-Marlin, 2016-11-13:
# // X and Y offsets must be integers.

What are your settings for:
```
/**
 * Z probes require clearance when deploying, stowing, and moving between
 * probe points to avoid hitting the bed and other hardware.
 * Servo-mounted probes require extra space for the arm to rotate.
 * Inductive probes need space to keep from triggering early.
 *
 * Use these settings to specify the distance (mm) to raise the probe (or
 * lower the bed). The values set here apply over and above any (negative)
 * probe Z Offset set with Z_PROBE_OFFSET_FROM_EXTRUDER, M851, or the LCD.
 * Only integer values >= 1 are valid here.
 *
 * Example: `M851 Z-5` with a CLEARANCE of 4  =>  9mm from bed to nozzle.
 *     But: `M851 Z+1` with a CLEARANCE of 2  =>  2mm from bed to nozzle.
 */
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```

kaliboe, 2016-11-13:
//===========================================================================
//============================= Z Probe Options =============================
//===========================================================================

//
// Probe Type
// Probes are sensors/switches that are activated / deactivated before/after use.
//
// Allen Key Probes, Servo Probes, Z-Sled Probes, FIX_MOUNTED_PROBE, etc.
// You must activate one of these to use AUTO_BED_LEVELING_FEATURE below.
//
// Use M851 to set the Z probe vertical offset from the nozzle. Store with M500.
//

// A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
// For example an inductive probe, or a setup that uses the nozzle to probe.
// An inductive probe must be deactivated to go below
// its trigger-point if hardware endstops are active.
//#define FIX_MOUNTED_PROBE

// The BLTouch probe emulates a servo probe.
//#define BLTOUCH

// Z Servo Probe, such as an endstop switch on a rotating arm.
#define Z_ENDSTOP_SERVO_NR 0
#define Z_SERVO_ANGLES {0,80} // Z Servo Deploy and Stow angles

// Enable if you have a Z probe mounted on a sled like those designed by Charles Bell.
//#define Z_PROBE_SLED
//#define SLED_DOCKING_OFFSET 5 // The extra distance the X axis must travel to pickup the sled. 0 should be fine but you can push it further if you'd like.

// Z Probe to nozzle (X,Y) offset, relative to (0, 0).
// X and Y offsets must be integers.
//
// In the following example the X and Y offsets are both positive:
// #define X_PROBE_OFFSET_FROM_EXTRUDER 10
// #define Y_PROBE_OFFSET_FROM_EXTRUDER 10
//
//    +-- BACK ---+
//    |           |
//  L |    (+) P  | R <-- probe (20,20)
//  E |           | I
//  F | (-) N (+) | G <-- nozzle (10,10)
//  T |           | H
//    |    (-)    | T
//    |           |
//    O-- FRONT --+
//  (0,0)
#define X_PROBE_OFFSET_FROM_EXTRUDER -25.00  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -16.00  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -9.70   // Z offset: -below +above  [the nozzle]

// X and Y axis travel speed (mm/m) between probes
#define XY_PROBE_SPEED 4000
// Speed for the first approach when double-probing (with PROBE_DOUBLE_TOUCH)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
// Speed for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)
// Use double touch for probing
//#define PROBE_DOUBLE_TOUCH

//
// Allen Key Probe is defined in the Delta example configurations.
//

// Enable Z_MIN_PROBE_ENDSTOP to use _both_ a Z Probe and a Z-min-endstop on the same machine.
// With this option the Z_MIN_PROBE_PIN will only be used for probing, never for homing.
//
// *** PLEASE READ ALL INSTRUCTIONS BELOW FOR SAFETY! ***
//
[... 88 lines cut ...]

// @section homing

//#define Z_HOMING_HEIGHT 4  // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
                             // Be sure you have this distance over your Z_MAX_POS in case.

// ENDSTOP SETTINGS:
// Sets direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR -1

#define min_software_endstops true // If true, axis won't move to coordinates less than HOME_POS.
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.

// @section machine

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 200
#define Y_MAX_POS 180
#define Z_MAX_POS 200

kaliboe, 2016-11-13:
Found my problem, don't know just the reason but I change version of Marlin, I used RC7 now I installed RCBugFix instead and it works.

kaliboe, 2016-11-13:
Auto bed levelling is working now but I find that the nozzle is a little bit to high from the nozzle.
Where can I adjust this setting in the configuration.h file ?
Thanks in advance

Roxy-3D, 2016-11-13:
I typicall adjust Z_PROBE_OFFSET_FROM_EXTRUDER in Configuration.h.

Be sure to do a M502 and M500 if you change the number.  Otherwise, it will keep using the version stored in EEPROM.

kaliboe, 2016-11-14:
thanks for the answer Roxy-3D
is it also possible that the nozzle move first in the air to the starting point where it lower the nozzle and start to print then instead of now it lower first and move the nozzle over the bed to the starting point?

Roxy-3D, 2016-11-14:
Ah...   I'm confused!    Of course nozzle should always have clearance before it moves.  That behavior is always controlled by the 
```cpp
#define Z_HOMING_HEIGHT 15  // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
                             // Be sure you have this distance over your Z_MAX_POS in case.
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```

Those names have changed.   Do these items have reasonable numbers for your printer in them?

kaliboe, 2016-11-14:
I have those settings in my setup ...
let me ex plane it better, before I start a print job it runs first a G28 (homing) no problem here
then a G29 (auto-levelling) follows behind, here also no problem. The nozzle stop for a second and will start the print job, because the nozzle is to height from the bed it lowers first then it stops (on printing height) and will travel then to the middle of the bed to start printing on the object. Is there a way it will move first to the middle of the bed and lowers then the nozzle?

Roxy-3D, 2016-11-14:
It might be worth while to look at the GCode file for the print you are doing.   My guess is you will see it wait for the extruder to come up to temperature.  Then it will lower the nozzle to .2mm.  And then it will move to the center of the bed.

One work around would be to change your startup GCode in the slicer to move the nozzle to the center of the bed after it is done waiting for the nozzle to come up to temp (and before the slicer tells the nozzle to go to .2mm)

Blue-Marlin, 2016-11-15:
On the way from where G29 endet to the middle of the bed ABL is already active. If you are abele to actually touch the bed, your `Z_PROBE_OFFSET_FROM_EXTRUDER` is wrong. Try a value closer to zero.

closed 2017-11-01 by thinkyhead