MarlinFirmware/Marlin #5215: Start print move before unretraction is complete.

NimrodQuimbus, 2016-11-13:
With bowden tubes and geared extruder, blobs seem to be created because nozzle pressure builds much slower than direct extruders, so, print move should start accelerating a bit before the extruder gets done with unretraction,

Is this possible to do? 
or has it already been done?

Thanks!

Sebastianv650, 2016-11-13:
Want you mean is a nozzle pressure control system. Have a look at LIN_ADVANCE in the Configuration_adv.h file.

Sebastianv650, 2016-11-13:
I forgot to mention the most important thing, the documentation:
https://www.marlinfw.org/docs/features/lin_advance.html

NimrodQuimbus, 2016-11-13:
No, not an advance, but print movement while un-retracting.Because a blob is created while extruder is un-retracting while there is no hotend movement taking place.
Direct extruder retractions may be as small as 1mm, while Bowden retractions are as large as 5mm or even more, so, if there is a small amount of print move during the 5mm un-retraction, it may eliminate the start blob.

Thanks for the quick response.

Sebastianv650, 2016-11-13:
Think about why there is a blob at the start point during "un-retract" (=prime). If your last print move (which ended with a retract) would be finishing without an excess of nozzle pressure, there would be no blob during prime for the next print move. You restore the same length of filament during priming than you retract before - so there is the exact same pressure as at the end of the line.
I'm ignoring oozing for this example - it's a different problem for it's own. It's also negligible if pressure control is activ.

Why there is too much nozzle pressure at the end of a printed line? Because you need force to push the filament through the nozzle at print speed. This force compresses the free filament between nozzle and extruder. If the printer is decelerating at the end of the print move, the saved energy in this filament (like a spring) is driving more filament through your nozzle even if your extruder comes to a full stop. If your print move ends before this stored energy is released completely, you have nozzle pressure at the end of the print move, where there should be none. With retract, you bring that pressure to zero. But during retract, you are also restoring this unwanted pressure -> blob.

NimrodQuimbus, 2016-11-13:
I think you are making an assumption that retract and un-retract is happening instantaneously, which it does not, and is most noticeable with Bowden set ups. 
A start blob happens during pressure build-up in the nozzle at the start of a print move, so, therefore a small amount of movement while an un-retraction takes place would eliminate it.ADVANCING the filament would make a bigger blob.

Sebastianv650, 2016-11-13:
> I think you are making an assumption that retract and un-retract is happening instantaneously

No, but fast enough. Significant blobs don't form in small fractions of a second..

> A start blob happens during pressure build-up in the nozzle at the start of a print move, so, therefore a small amount of movement while an un-retraction takes place would eliminate it.ADVANCING the filament would make a bigger blob.

I guess you have not read the documentation linked above up to now. It's named LIN_ADVANCE because the term "advance" is already in use for such features (JKN advance K and the old ADVANCE in Marlin). Of course it's not only speeding up the extruder during acceleration, it also does the reverse thing during deceleration.

Believe it or not, I wrote this feature to compensate for all that effects (blobs, gabs between perimeters, stringy infill, hughe retract lengths, ..) and it's working. It's maybe not that perfect on a bowden system due to other effects, but it will be a hughe improvement compared to the actual state.
For example, with LIN_ADVANCE enabled I lowered by retraction length from 1.5mm to 0.5mm. This property of LIN_ADVANCE alone will reduce your problem by a nice factor.

NimrodQuimbus, 2016-11-13:
Sorry, apparently I was misunderstanding you, and the language. Thank you for your patiently understanding and correcting me.
There are security issues with the link that you sent me, and it will not open the page.
Is LIN_ADVANCE a feature add-on? how do I get it?

Sebastianv650, 2016-11-13:
The site is under construction and has no officialy signed vertificate up to now. You should be able to ignore the warning from your browser and continue to the site (if you trust us).
LIN_ADVANCE is integrated into the RCBugFix code and can be enabled in the Configuration_adv.h file. Just open it and search for "//LIN_ADVANCE", change it to "LIN_ADVANCE" will enable it. But without reading the (quite long) documentation, I'm afraid you will have trouble setting it up. It's not complicated, but you have to know what are the needed steps to proceed.

NimrodQuimbus, 2016-11-13:
It has bee a while since I have set up the my current version of Marlin that I am using, maybe some time next week I will put effort into the task to use this new version, as I will also have to change, and relearn quit a few slicer settings, but, it looks like it will be worth it.
Thanks for your contribution to the firmware, and your assistance in this matter, I am sure the upgrade will be well worth it.
My Bowden printer is printing perfectly except for the start blob, and my current retraction is 5mm, so, It will be fun to get it working better.......
Thanks again,
Rich

NimrodQuimbus, 2016-11-15:
Hello again,
Under which tab will I find LIN_ADVANCE ?

Thanks!
Rich
 

    On Sunday, November 13, 2016 3:37 PM, Nimrod Quimbus <richh001@yahoo.com> wrote:
 

 It has bee a while since I have set up the my current version of Marlin that I am using, maybe some time next week I will put effort into the task to use this new version, as I will also have to change, and relearn quit a few slicer settings, but, it looks like it will be worth it.
Thanks for your contribution to the firmware, and your assistance in this matter, I am sure the upgrade will be well worth it.
My Bowden printer is printing perfectly except for the start blob, and my current retraction is 5mm, so, It will be fun to get it working better.......
Thanks again,
Rich

Sebastianv650, 2016-11-15:
Configuration_adv.h
Around line 529.

NimrodQuimbus, 2016-11-16:
hello again,I think I have it working,Cubes look great, tiny blob at end of perimeter.

Do I need to use/enable ADVANCE ?
I currently have LIN_ADVANCE at 760and DEFAULT_ACCELERATION at 665 ( higher values starts making a blob at start/end point on the cube)
I have a bowden set up with  39:11 Wades extruder.
Thanks,
Rich

Sebastianv650, 2016-11-16:
Please note PR #5230, I think I forgot to mention you there. You can continue without this, but your K factor will be off as soon as it gets merged into RCBugFix. You can calculate the new k factor as stated in the PR based on your current one.

> Do I need to use/enable ADVANCE ?

No, you can only enable LIN_ADVANCE or ADVANCE, not both. ADVANCE was the earlier implementation of a pressure control in Marlin, but when I started developing LIN_ADVANCE, it was broken. I found the bug in it during the development, I think somebody fixed it in the meantime. Nevertheless, ADVANCE is not doing the right calculation from my point of view. So I did LIN_ADVANCE ;)

> I currently have LIN_ADVANCE at 760and DEFAULT_ACCELERATION at 665 ( higher values starts making a blob at start/end point on the cube)

Good to know, up to now I don't know much about the behaviour of LIN_ADVANCE on bowden systems. I will try to memorize that, maybe I can du further improvements when I get more responses from bowden users.

NimrodQuimbus, 2016-11-16:
" Good to know, up to now I don't know much about the behaviour of LIN_ADVANCE on bowden systems. I will try to memorize that, maybe I can du further improvements when I get more responses from bowden users. "
Ok, I am glad I had some useful info for you, If there is anything I can do to help you out let me know, as I really do appreciate what you are doing.
I have noticed that DEFAULT_ACCELERATION is a really important setting, and is apparent with just small changes around the "sweet spot" 

I would love to find a comprehensive explanation of all the Acceleration settings in Marlin.....
Thanks,Rich

landodragon141, 2016-11-17:
I haven't gotten around to writing expanded documentation for acceleration. It'll happen soon™.

thinkyhead, 2016-11-17:
Another option not mentioned is that some slicers can do advance-extrusion as an expert/advanced option. This is probably a common solution for Bowden setups where the machine firmware lacks advance-extrusion.

closed 2016-11-17 by thinkyhead

NimrodQuimbus, 2016-11-17:
My original issue is that a blob is created on the start of a a line, using a Wades extruder, the unretract is slow, so, the blob is being created while the pressure is building up at the start before movement takes place.

thinkyhead, 2016-11-17:
Well, one of the purposes of the "advance" technique is to make sure there's minimal pressure left at the end of the previous line so that there will be less (or no) ooze at the beginning of the next line. When doing continuous lines where segments "flow together" there's no pressure drop between segments, but when changing direction (e.g., speed) between segments, especially on sharp corners, the pressure should be minimal at that point.

Can you post a video demonstrating the place where the blob is forming?

NimrodQuimbus, 2016-11-17:
My last post/message was about before I even knew about the latest Marlin with LIN_ADVANCE, and I am still "tweaking" it.I am still working on getting a good print on this part, it is much more challenging that the test cubes, and I am still getting a bit of a stop blob, and thin extrusion at the beginning.http://www.thingiverse.com/thing:1276384

But, as I said, I still need more tweaking.

NimrodQuimbus, 2016-11-17:
I am still playing with LIN_ADVANCE and am getting pretty nice results.
Make sure DEFAULT_MAX_FEEDRATE (for the E term) does not exceed extruder max speed.
For me, If K is set too high, I start getting real messy lines at the start point, and the shorter the line, the messier the line, and the odd thing that if it is a long line, it have under-extrusion at the beginning.

As of now, my current numbers are
K=400
DEFAULT_MAX_FEEDRATE (for the E term) = 40
3.75mm retractions (Gcode, Simplify3d slicer)
Hardware:
3mm filament
26 inch long Bowden tubes (E3d V5 setup)
39:11 Wade geared extruder.

I needed to use 5-6mm retractions before LIN_ADVANCE.

Hopes this helps someone.