MarlinFirmware/Marlin #5217: QUICK_ACCESS_TO_Z_BABYSTEPPING

ruggb, 2016-11-14:
@Roxy-3D 
What build does this function appear in? I have fa6bf12 and I don't see it in configuration.h
I have rearranged the code in ultralcd.cpp to place the babystepping function at the top of the main menu.
This allows fast access at the start of a print and lets me test it without queuing up a print.

Roxy-3D, 2016-11-14:
Originally...  I had it so the user could select a pin to put a switch on for the Quick Access feature.   But the more I thought about it, I didn't think most users would wire up a switch.   So instead, I changed the code to be controlled by:
```cpp
#define QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN
```

If you want to select the pin that triggers that feature and wire up a switch...  Just grab the code base immediately before  QUICK_ACCESS_TO_Z_BABY_STEPPING_VIA_KILL_PIN showed up.  That will have the generic Quick Access feature and you have to pick a pin to trigger it.

ruggb, 2016-11-14:
The point was that I searched config and config.adv and did not find this definition. Rearranging the code every build is a PITA but it works just like I need it that way. There is no point of adding a switch. I don't understand why anyone would use this function during a print except at the start to adjust Z0 and with it buried in the menu and available only when printing it is useless to try to use it unless one is printing a LONGGGG brim. You must wait till the print starts to access it and to get there takes too many operations. You may want to consider changing it to the way I have mine and not defaulting to XY being enabled since I don't think anyone uses it for XY and it causes it not to work for CoreXY/H-bot configurations. I keep asking for this, and ppl with coreXY keep complaining that it doesn't work on their machine and others complain it takes tooooo long to access it, but y'all aren't listening.
It is a simple change - moving 2 sections of code and commenting out the XY enable code.
It may also make it clearer if you relocated that XY enable to where the main BS define is located as is the coreXY and coreXZ and coreYZ defines. I also increased the timeout in ultralcd.h:
  #define LCD_TIMEOUT_TO_STATUS 30000 //** default 15000 not enough time for BS operation

for your consideration here is my change in ultralcd.cpp which places the Babystep function at the top of the main menu. It takes 2 knob presses (and no menu searching) to get to the adjustment and you can get there b4 you start a print:
at approx line 572

`  #endif //SDSUPPORT

//** ========START=======moved from below===================

  #if ENABLED(BABYSTEPPING)

    long babysteps_done = 0;

    static void _lcd_babystep(const AxisEnum axis, const char* msg) {
      if (LCD_CLICKED) { lcd_goto_previous_menu(true); return; }
      ENCODER_DIRECTION_NORMAL();
      if (encoderPosition) {
        int babystep_increment = (int32_t)encoderPosition * BABYSTEP_MULTIPLICATOR;
        encoderPosition = 0;
        lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
        thermalManager.babystep_axis(axis, babystep_increment);
        babysteps_done += babystep_increment;
      }
      if (lcdDrawUpdate)
        lcd_implementation_drawedit(msg, ftostr43sign(
          ((1000 * babysteps_done) * planner.steps_to_mm[axis]) * 0.001f
        ));
    }

    #if ENABLED(BABYSTEP_XY)
      static void _lcd_babystep_x() { _lcd_babystep(X_AXIS, PSTR(MSG_BABYSTEPPING_X)); }
      static void _lcd_babystep_y() { _lcd_babystep(Y_AXIS, PSTR(MSG_BABYSTEPPING_Y)); }
      static void lcd_babystep_x() { babysteps_done = 0; lcd_goto_screen(_lcd_babystep_x); }
      static void lcd_babystep_y() { babysteps_done = 0; lcd_goto_screen(_lcd_babystep_y); }
    #endif
    static void _lcd_babystep_z() { _lcd_babystep(Z_AXIS, PSTR(MSG_BABYSTEPPING_Z)); }
    static void lcd_babystep_z() { babysteps_done = 0; lcd_goto_screen(_lcd_babystep_z); }

  #endif //BABYSTEPPING

//** =========END===moved from below==========================

  /**
   *
   * "Main" menu
   *
   */

  static void lcd_main_menu() {
    START_MENU();
    //** =================START MOVE from below========================

    //
    // Babystep X:
    // Babystep Y:
    // Babystep Z:
    //
    #if ENABLED(BABYSTEPPING)
      #if ENABLED(BABYSTEP_XY)
        MENU_ITEM(submenu, MSG_BABYSTEP_X, lcd_babystep_x);
        MENU_ITEM(submenu, MSG_BABYSTEP_Y, lcd_babystep_y);
      #endif //BABYSTEP_XY
      MENU_ITEM(submenu, MSG_BABYSTEP_Z, lcd_babystep_z);
    #endif

//** ===========END move from below===========================

    MENU_BACK(MSG_WATCH);

`

Roxy-3D, 2016-11-14:
> for your consideration here is my change in ultralcd.cpp which places the Babystep function at the top of the main menu. It takes 2 knob presses (and no menu searching) to get to the adjustment and you can get there b4 you start a print:
at approx line 572

This is a good idea.   I'll add it to the devel_ubl branch.    It may take a little while to show up there however.   I'm in the middle of trying to get my new printer up and running and I'm fighting a bunch of issues.   Plus, I'm adding a 'Debug Panel' with control switches and and LED Matrix to give me more feedback on things.    

But in a couple of weeks it will be in the devel_ubl branch.

ruggb, 2016-11-14:
Yeah, I'm sure there must be a variation of a printer that might use BSxy, but if it is disabled by default then someone who wants it will find it and corexy users won't be confused by it.
I think the last time is used it w/o disabling it, the xy appeared after the z, but I can't remember. That would make z default for those who enabled xy.
what is your new printer?

Roxy-3D, 2016-11-14:
I've got a gCreate gMax 1.5+   http://www.gcreate.com/gmax-15-2016-series

It has a bed that is slightly smaller than a Tennis Court.  It is the ideal machine to use as a development platform for the UBL code base.

cdedwards, 2016-11-14:
couldn't find a smaller printer?

ruggb, 2016-11-14:
That looks more like an aircraft carrier deck. Mine is a self-built corexy 12x12x12 with an 8x8 hotbed for ABS

Kaibob2, 2016-11-21:
@Roxy-3D 

> Soon... I'll have it so the feature can also be triggered by a "Press and Hold" of the Encoder wheel for 1.250 ms when in the status screen.

Do you have "Press and hold" already working or is it just an idea?

Roxy-3D, 2016-11-21:
I have it working...   I just don't have it buttoned up where I can upload it.  I have a number of other changes mixed in that are not quite ready to upload.   But in the mean time...  I suggest you give ruggb's approach up above a try:   https://github.com/MarlinFirmware/Marlin/issues/5217#issuecomment-260374526

That is a very quick and sure way to get into the Z-BabyStepping.

Kaibob2, 2016-11-21:
:( This was exactly what i didn't want. When i first started with Marlin i mix up the menu entrys as i thought it would be useful. This was only nice until i had to merge it with the next RCBugFix.
I will wait until press and hold becomes available.

ruggb, 2016-11-21:
It is a PITA changing it for every build but I couldn't get any acknowledgement about the issue - it did get easy after a few times since it is just 2 moves - Thankx for seeing the light @Roxy-3D

Roxy-3D, 2016-11-23:
First a big Thank You to @ruggb !    His idea is very elegant for doing what we want with the quick and easy access to the Z-BabyStepping!

I don't have the code uploaded yet.   I'll upload it to the devel_ubl branch tomorrow sometime.   But I have @ruggb 's idea implemented.  I ended up implementing it a little differently than he did it.   The way it works is if you are in the 'Status Screen' and you give it two quick clicks, you end up in Z-BabyStepping.

What I did was over ride the Main Menu's first item (which is a 'Go Back to Status Screen') and told the code to see how long it has been in the Main Menu prior to doing that.   If the Main Menu screen has been up less than 1.25 seconds and you click on the 'Go Back to Status Screen' you proceed immediately to the Z-BabyStepping.    It works pretty cleanly and should provide the easy access you need.

ruggb, 2016-11-23:
@Roxy-3D nice improvement. What you are doing is leaving the return to status at the top of the menu but if you dbl clk from the status menu it goes to Z babystepping. I like it. That is exactly how I do it now, quickly dbl clk. So getting there is exactly the same as mine, but the menu works the same as it did. If I did something else instead of babystepping, going back to status was a little awkward on that menu. But, in reality, most of the time I just left it and it kicked itself back home. I assume you also made it accessible any time - IE it doesn't have to be printing.

Roxy-3D, 2016-11-23:
It can be what ever people say they want it to be.   But "Yes..."   Right now, the algorithm just checks to see if the previous menu was the top level "Status Screen" and if the current menu is the "Main Menu" and if the next Encoder Wheel click showed up within 1.25 seconds.   If so...  It is off to the Z-BabyStepping.   

Ignoring the fact you need a few declarations to allow the 'Forward References'...   This is the entire code to do it:
```cpp
  /**
   * General function to go directly to a menu
   */

  static void lcd_goto_screen(screenFunc_t screen, const bool feedback = false, const uint32_t encoder = 0) {

  #if ENABLED(BABYSTEPPING)
    if (currentScreen==lcd_status_screen && screen==lcd_main_menu)	// We are in leaving the status screen to goto the main_menu 
      status_screen_click_time = millis();				// screen.  Mark the time so we know how quick the user is
									// pressing buttons.
    if (currentScreen==lcd_main_menu)  {
      if ( screen==lcd_status_screen && status_screen_click_time+1250>millis() ) {
        lcdDrawUpdate = LCDVIEW_CLEAR_CALL_REDRAW;
	status_screen_click_time = 0;
	lcd_babystep_z();
	return;
      }
    }
  #endif
```

ruggb, 2016-11-23:
when I do it the screen the just shows BABYSTEPPING Z: 0.000 - seems like that would be sufficient.
Will this get into RC7, or RC7 bug fix?

ruggb, 2016-11-23:
@Roxy-3D OH BTW, since you are calling the first screen the "STATUS" screen it might be good to rename that menu item from "INFO SCREEN" to "STATUS SCREEN". Just saying.......
Also, can you make it so that a clk will move it back off the babystepping function. Right now I must wait till it times out to get back to the menu.

Roxy-3D, 2016-11-23:
> when I do it the screen the just shows BABYSTEPPING Z: 0.000 - seems like that would be sufficient.

Remember...  devel_ubl forked from RCBugFix on July 4th.   'BabyStepping Z: 0.00' must of been added after that.   I can probably just cross those changes over.  I'll take a look at that before I upload the changes to the UBL branch.

> Will this get into RC7, or RC7 bug fix?

It certainly can.   Mostly it will be up to @thinkyhead .   But we do need to freeze features on RC-7 and start just doing nothing but bug fixes or it will never stabilize enough to be a 'Stable' release.

> OH BTW, since you are calling the first screen the "STATUS" screen it might be good to rename that menu item from "INFO SCREEN" to "STATUS SCREEN". Just saying.......

Well...  I just used the names that the functions are called in the code.   Yes, I do know the text of that first line says 'Info' and the function is actually called 'Status'.

> Also, can you make it so that a clk will move it back off the babystepping function. Right now I must wait till it times out to get back to the menu.

Yes...  When you are in the BabyStepping function, doing a Click will put back at the Main Menu.   That is probably the right place to go because if you clicked you are done with BabyStepping and may want to change something else like Temperature.    Even if the click was mistake, it is just one more click to get back to the Status screen and then 2 more clicks to be back in Z-BabyStepping.

ruggb, 2016-11-23:
Does the clk on my mod not get me back to the main menu because I missed something or did you have to fix that?
Yes, I agree, the MAIN menu is the logical place to backup to.

Roxy-3D, 2016-11-23:
The UBL BabyStep code is slightly different because it has to accommodate UBL taking over control of the LCD Panel.   If you delete the 'if' statement...  Probably you get the behavior you want:
```cpp
    static void _lcd_babystep(const AxisEnum axis, const char* msg) {
      if (lcd_clicked) { defer_return_to_status = false; return lcd_goto_previous_menu(); }
```
That 'if' statement is not in the UBL code.  And it behaves the way I described up above.

ruggb, 2016-11-24:
@Roxy-3D I added that if statement 
`      if ( /* !UBL_has_control_of_LCD_Panel && */ LCD_CLICKED ) {
	lcd_goto_previous_menu(true);
      }
to what I have, but it does not do anything - except I notice a single pixel on lower right goes on/off when I click it. It may have been doing that and I never noticed it.
I then changed the last line to
`lcd_goto_screen(lcd_main_menu)`
That didn't do it either.

Roxy-3D, 2016-11-24:
Are you running RCBugFix or devel_ubl ?    That code (without the commented out UBL_has_control_of_LCD_Panel) is what is in UBL and it works as described.  I would think it should work in RCBugFix also.  Although RCBugFix has the check up at the top of the function.

ruggb, 2016-11-24:
I am using bug fix - it may be that I moved those functions to get what I have that is the problem. The code that you posted is exactly what I have except for that last if. But that whole section is also what I moved.
If I don't figure it out it isn't a show stopper. Did you get to upload that today? I may try it for fun. What is the build number?

ruggb, 2016-11-24:
@Roxy-3D I dnld the ubl build and compiled it - I got bunches of errors most having to do with - "RAW_CURRENT_POSITION" - I generally do this on a virgin build just to make sure when I change things it isn't my fault but I have not received errors on other builds. Is there something wrong or did I miss an instruction?

Roxy-3D, 2016-11-24:
> Did you get to upload that today? I may try it for fun. What is the build number?

I don't know the build number.  But it happened at roughly lunch time.   You don't need any of that info.  Just select the devel_ubl branch and do a download.   It is at the end of the branch.

Roxy-3D, 2016-11-24:
> I dnld the ubl build and compiled it - I got bunches of errors most having to do with - "RAW_CURRENT_POSITION" - I generally do this on a virgin build just to make sure when I change things it isn't [...]

Please check out:  https://github.com/MarlinFirmware/Marlin/issues/4948#issuecomment-261824721

That exact same error showed up yesterday.  But when I pulled down the code and compiled it... It compiled clean.   And when I did the changes today, I pulled down the entire branch and compiled it.  (I honestly do not believe somebody trying to use my code should have to fight the compiler to make it go!!!)   It compiled clean.

May I ask you to pull down a fresh copy, and make no changes to it.  Just pull it down and compile it.  It is my belief it will compile clean.  And then be careful crossing your changes over to the Configuration.h file and maybe just do a few at a time and keep checking that it compiles clean.  

I don't know...  But I suspect we are fighting a problem getting your settings mapped over to the UBL Configuration.h file.  The thing that is weird is that same message keeps showing up.  If you figure out what is causing it please post and let us know.  And if you don't figure it out...  I'll read all your messages in the morning and dig in and try to get the problem fixed!!!

Roxy-3D, 2016-11-24:
> @Roxy-3D I dnld the ubl build and compiled it - I got bunches of errors most having to do with - "RAW_CURRENT_POSITION" - I generally do this on a virgin build just to make sure when I change things [...]

And one more thought...  You can't take an RCBugFix Configuration and drop it into a devel_ubl directory and expect it to compile.   Too many names and such have changed.   You have to cross your settings over to the Configuration.h file in the UBL directory.   Sorry...

ruggb, 2016-11-24:
@Roxy-3D same thing - I dnld another copy - same build - unzip and compile and here are the beginning of the errors I get, no changes - the list is MUCH longer and has other errors - using Arduino 1.6.12. all libraries installed appear to be up to date. There are other errors if you want me to post all of them.
`
In file included from sketch\Bed_Leveling.cpp:23:0:

sketch\Marlin.h:349:0: warning: "RAW_CURRENT_POSITION" redefined

 #define RAW_CURRENT_POSITION(AXIS)  RAW_POSITION(current_position[AXIS], AXIS)

 ^

In file included from sketch\Bed_Leveling.cpp:23:0:

sketch\Marlin.h:140:0: note: this is the location of the previous definition

 #define RAW_CURRENT_POSITION(AXIS) (RAW_POSITION(current_position[AXIS], AXIS))

 ^

In file included from sketch\G25.cpp:6:0:

sketch\Marlin.h:349:0: warning: "RAW_CURRENT_POSITION" redefined

 #define RAW_CURRENT_POSITION(AXIS)  RAW_POSITION(current_position[AXIS], AXIS)

 ^

In file included from sketch\G25.cpp:6:0:

sketch\Marlin.h:140:0: note: this is the location of the previous definition

 #define RAW_CURRENT_POSITION(AXIS) (RAW_POSITION(current_position[AXIS], AXIS))

 ^`

ruggb, 2016-11-24:
Oh BTW - happy Thanksgiving - I will be gone from 11-26 thru 12-9 but may be able to respond after 12/4. So I am not abandoning this.........

Roxy-3D, 2016-11-24:
@ruggb   I just pulled it down again and compiled it.  It compiled clean with Arduino 1.6.9    I do not have any libraries installed.    I'm wondering, are you using a Graphics board?   I seem to remember something bad about the latest U8glib v1.19.0 being bad.  I could be making that up.  But I also have in my head that U8glib v1.18.0 is OK.   I don't know if that is really true.
[image]

ruggb, 2016-11-25:
@Roxy-3D I have no problem compiling the last few bug fix builds. All the RAW... errors are gone in your last build, this is what is left:

`sketch\G26_Mesh_Validation_Tool.cpp: In function 'void gcode_G26()':

sketch\G26_Mesh_Validation_Tool.cpp:237:72: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   debug_current_and_destination("Starting G26 Mesh Validation Pattern.");

                                                                        ^

sketch\G26_Mesh_Validation_Tool.cpp:355:6: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 cptr = "01234567890ABCDEF????????";

      ^

sketch\G26_Mesh_Validation_Tool.cpp:370:69: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

        debug_current_and_destination("Looking for lines to connect.");

                                                                     ^

sketch\G26_Mesh_Validation_Tool.cpp:372:63: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

        debug_current_and_destination("Done with line connect.");

                                                               ^

sketch\G26_Mesh_Validation_Tool.cpp:375:63: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

      debug_current_and_destination("Done with current circle.");

                                                               ^

sketch\G26_Mesh_Validation_Tool.cpp:385:53: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 debug_current_and_destination("ready to do Z-Raise.");

                                                     ^

sketch\G26_Mesh_Validation_Tool.cpp:387:52: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 debug_current_and_destination("done doing Z-Raise.");

                                                    ^

sketch\G26_Mesh_Validation_Tool.cpp:394:53: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 debug_current_and_destination("done doing X/Y move.");

                                                     ^

sketch\G26_Mesh_Validation_Tool.cpp: In function 'void look_for_lines_to_connect()':

sketch\G26_Mesh_Validation_Tool.cpp:532:60: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 debug_current_and_destination("Connecting horizontal line.");

                                                            ^

sketch\G26_Mesh_Validation_Tool.cpp:572:58: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 debug_current_and_destination("Connecting vertical line.");

                                                          ^

sketch\G26_Mesh_Validation_Tool.cpp: In function 'void move_to(float, float, float, float)':

sketch\G26_Mesh_Validation_Tool.cpp:691:63: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

[... 130 lines cut ...]
                                                                        ^

sketch\planner.cpp: In static member function 'static void Planner::buffer_line(float, float, float, float, float, uint8_t)':

sketch\planner.cpp:549:64: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 debug_current_and_destination("Start of planner::buffer_line()");

                                                                ^

sketch\G29_Unified_Bed_Leveling.cpp: In function 'gcode_G29() [clone .part.11]':

sketch\G29_Unified_Bed_Leveling.cpp:353:27: warning: iteration 10 invokes undefined behavior [-Waggressive-loop-optimizations]

    z_values[i][i] += 10.0 ;     // cells thick that is raised

                           ^

sketch\G29_Unified_Bed_Leveling.cpp:352:10: note: containing loop

  case 1: for(i=0; i<MESH_NUM_X_POINTS; i++ ) {    // Create a diagonal line several Mesh

          ^

`

Roxy-3D, 2016-11-25:
OK!  As long as they are warnings...  Just ignore them.  I'll get those cleaned up so they compile without warnings.  Which version of the compiler generated these?   I'm going to switch to it so I can get everything neat and tidy.

ruggb, 2016-11-25:
i'm using 1.6.12 - I just checked and .13 is out - I dnld that and installed it and...........same thing.
So it is probably good to go to 1.6.13.

Roxy-3D, 2016-11-25:
I just tried v1.6.13 and it compiled clean.   We must have different settings or something else going on???

ruggb, 2016-11-25:
scared me there. I got the email all was well, then I logged in.
thanks

ruggb, 2017-02-08:
I now have RC8 9b55159 and there is no sign of babystepping when the printer is idle. Have the powers that were decided that I am incapable of controlling this or is there a switch that I must turn to make it happen. OR must I continue to modify ultraldc.cpp for every new build.

Roxy-3D, 2017-02-08:
Well...   I'm in the process of moving the UBL code base so it runs on top of RC-8's RCBugFix.   But that isn't done yet.   The Z-BabyStepping has to be turned on in Configuration_adv.h    It is towards the end of the file.      And the changes to ultralcd.cpp to make it so a double click from the status screen lands you in Z-BabyStepping are just a 4 or 5 lines of code.    It probably would be good to make a Pull Request so that can be added to the RCBugFix side of things, but I haven't done that.   If you make a pull request for the Double Click Z-BabyStepping...   I'll merge it...

ruggb, 2017-02-09:
I modified my utltraLCD.cpp file as I did in the past and now it no longer times out back to status screen.
Other screens seem to work. Something broke with the current build.

Roxy-3D, 2017-02-09:
Oh!   I'm seeing that behavior on the Graphics LCD Panel when I do the double click to get to the Z-BabyStepping.    This is pretty much my first time using the Graphics LCD and I thought that was normal behavior for that type of panel.

ruggb, 2017-02-09:
There appears to be some major changes to the ultralcd.cpp file comparing the latest BF with the UBL version. I think the missing timeout needs to be fixed b4 throwing another change on it.

ruggb, 2017-02-09:
I had also changed the timeout from 15000 to 30000 in the .h file b/c it was too short for me. So I know it did work.

closed 2017-10-31 by thinkyhead