MarlinFirmware/Marlin #5218: M203 Behavior?

darrelhunt, 2016-11-14:
Curious as to the designed behavior of M203

From what I can understand, we set max feedrates for the axis in Configuration.h
We can then override those with M203

It would seem to me that this is backwards.

Say a a slicer decides to issue an M203 Z7800 for some reason, say accidental mis-configuration by user, software bug, whatever.

It's then now trying to operating the machine at a speed that's dangerous to itself.

I'm suggesting a change that that would make it work in the opposite.
Essentially, always honor the max feedrates.  Allow them to be lowered by M203, but never exceeded.

thinkyhead, 2016-11-16:
Actually, you won't harm anything by using too high a feedrate. Your steppers will simply be unable to manage the required torque and skip steps.

darrelhunt, 2016-11-18:
There's a physical limit that it'll reach before it gets to that point.  For example I've got flexible aluminum couplings on my Z axis, I've actually had them distort.  If the move it feeds too quickly on is counter to the spiral cut, it actually can start to "unspiral" them.  Same with belts, there's a point where the belt can slip on the pulley and if done enough can wear the teeth.  Though those scenarios are both rare, if a person allows either to occur multiple times, that's their fault.

Say I turn EEPROM off?  would M203 still be able to affect the speeds I have set as MAX?

thinkyhead, 2016-11-19:
`M203` will still work. It just won't be preserved when restarting and host-connecting.

BagelOrb, 2016-11-29:
In Cura we issue a user warning when he puts the max feedrate above the max feedrate defined by the firmware. Cura has machine settings which record the max feedrate used by a specific machine. When a user overrides this feedrate, a warning is given if he overwrites it to a value higher than what's normal in the firmware.

thinkyhead, 2016-11-30:
@BagelOrb Cura can also, if it wishes, issue `M503 S0` and capture the current settings, without the user needing to enter them into the Machine Settings dialog. These will be reported even if there's no EEPROM enabled.

BagelOrb, 2016-11-30:
That wouldn't work for the Ultimaker 2 family. They are in no way connected
to the computer.

I don't think it's worth the effort of implementing such a feature if it's
just to make it easier to provide settings for a new machine.

closed 2017-10-31 by thinkyhead